meerk40t/meerk40t #179: In the navigation window, the rightmost square has no mouseover messages

dietervansteenwegen, 2020-06-06:
As the title states, there seem to be no mouseover messages for these buttons:
[image]

While they are very intriguing, I can't guess at what they do. Used for additional axis?

tatarize, 2020-06-07:
Matrix transform operations for only changing the affine matrix on the selected design. Thought I had those too, but they must not made it in the final and gotten lost in the design bits somewhere. The ones on the left move, just the laser head. In the middle they move both laserhead and design, and on the right they only move the design (relative to the laser head for some of those operations).

Fixed it.
https://github.com/meerk40t/meerk40t/pull/160/commits/232310c26dda9eabc921833a549352591e6ac239

tatarize, 2020-06-23:
This issue was corrected in 0.6.0. Which is now in full release.

closed 2020-06-23 by tatarize