meerk40t/meerk40t #181: Saving a project and reopening does not return the project to the saved state

dietervansteenwegen, 2020-06-06:
I have a "project" but need to close down the computer. I'd like to continue working on it in a couple of days and I have set up a number of operations. So I click File -> Save as.

When using File -> Open Project, previously created operations are lost and replaced by the default 150mm/s raster operation for all elements.

Since the Save function saves to a SVG file, I'm not even sure if that is supposed to save the project instead of just the saving the current area.

tatarize, 2020-06-07:
Yeah, the saving of operations isn't yet done. I may have to move that up. To 0.6.1 though I really just want the high usability changes there. SVG files can save a bunch of extra details without any issue. The file format allows a lot of unknown content and foreignObjects. To use these the program this should include a namespace for the custom objects, so they are known. The meerk40t namespace points here: https://github.com/meerk40t/meerk40t/wiki/Namespace it's a bit old though and I don't think most of that works anymore. 

When fixed it should include cuts/engrave/rasters/project-notes/? as objects within the meerk40t namespace then use the traditional svg stuff to make the objects and the meerk40t namespaced objects to explain the cuts and which svg-objects those operations use, the speed, directionality, etc. 

After 0.5.0 and the switch to operation first, the initial code there was broken except for making the bed size correct and the svg-objects output. Your saved values for raster and engrave and cut will persist between usages but the actual information in the file there doesn't save. However, it could save in an SVG and that is the current intent.

#170 #153

tatarize, 2020-07-16:
This is corrected in 0.6.1, now live.

closed 2020-07-16 by tatarize