meerk40t/meerk40t #182: Error while hiding paths

samgozman, 2020-06-14:
Steps to reproduce:
1) Insert any SVG file
2) Click "Hide paths"
3) Enjoy 😊 

```
MeerK40t crash log. Version: 0.5.2, Windows 10 x64
Traceback (most recent call last):
  File "site-packages\wx\core.py", line 3285, in <lambda>
  File "wxMeerK40t.py", line 700, in refresh_in_ui
  File "wxMeerK40t.py", line 716, in update_buffer_ui_thread
  File "wxMeerK40t.py", line 1168, in on_draw_scene
  File "LaserRender.py", line 46, in render
  File "LaserRender.py", line 46, in <listcomp>
  File "Kernel.py", line 702, in __getitem__
TypeError: ConfigBase.Read(): argument 1 has unexpected type 'int'

```

tatarize, 2020-06-14:
Corrected in 0.6.0

tatarize, 2020-06-23:
This issue was corrected in 0.6.0. Which is now in full release.

closed 2020-06-23 by tatarize