meerk40t/meerk40t #186: FAQ, Documentation

tatarize, 2020-06-26:
The idea here is a log of the questions asked about the software and answers given to keep a log of where the UI/UX does not imply the answers. These need to be integrated into some documentation, manual, or main video series about the software. Any good documentation should address these points.

tatarize, 2020-06-26:
* to unlock rail and manually adjust the laser select “automatically lock rail”?
    * You can also click the unlock button in navigator that looks like an unlocked lock. Or hit F12 and type "unlock". The automatically unlock just makes sure it tends to remain unlocked after jobs or jogging.
    * Since unlock is console command, you can also also bind it to a key. Though currently the binds aren't saving between sessions. But, "bind space unlock" in console or something like that in keymap would be nice if you need to do that.
    * But, really you don't need to unlock the rail, and move it around. That's more of a thing required in Whisperer. You can put the project at the position in your bed that you want, and run it from there. The main scene is your bed scene. So you can just locate it how and where you want it, then run it, and whereever your laser is, it will go to the start location and begin work.
* the power of the laser is still controlled by the panel?
    * The power is controlled by the panel. That does the amount of current to the laser and must remain within bounds. The power here is PPI pixels per inch. If you put it at 500 it will be half as strong of a line as it would be at 1000. But it does this modulating the pulses in software. The modulation of the current to the laser is typically PWM and the modulation here is PPI.
*  do you always leave PPI power at 1000? Why should it be changed and when?
    * You would change the PPI if you wanted to engrave something lighter or use less power. Like if you wanted to engrave a piece of tissue paper, you literally could not engrave that at *any* speed fast enough to not burn it, you'd use the PPI. Or if you want some lines to be half the depth of other lines. Or if you want to do a very light outlining of something. Sometimes if you're rastering a solid black thing adjusting the PPI is an easy fix to it being too dark.
*  the speed settings should be the same as K40 Whisperer?
    * The speed settings are the same as Whisperer. In fact, it runs a slightly updated version of the code in Whisperer which I also wrote as I spent a month or so properly reverse engineering the speedcodes for the M2 Nano.

celticht32, 2020-08-24:
Does Meerk40t support the following svg... it is a dashed line... k40w treats it as a solid line... what I am looking to do is have my laser cut the dashed lines I would use to stitch leather.. otherwise its me having to cut the dashed lines.

[dash lines .zip](https://github.com/meerk40t/meerk40t/files/5115240/dash.lines.zip)

tatarize, 2020-08-24:
No, MeerK40t doesn't process dashed lines. But, I specifically looked at dash arrays when I was coding the svg spec. But, I didn't think it was useful enough to include (given the effort it would take to implement), but more generalized dashed lines seemed like they would be useful. The loaded SVG files have the information but MeerK40t never does anything with it, though it has access to that information.

I did however make a PPI advanced setting that hasn't made it into production called Dot Length (it's in advanced, in operation properties and disabled). The core idea was to set a dot length for the PPI carry forward so you could basically make dashed lines yourself with that. If set very short, they would be let you avoid some potential stuttering. If made longer they could give a dashed effect.

I've been meaning to check whether this bunch of chained plotting generators could be turned into something more elegant. If it can be I'll look into, implementing a dashed line. Though I'm only a bit away from a version update. So if something elegant doesn't present itself, I likely wouldn't include it. Since I can't really risk getting too far into the weeds. But, there's some chance that there's a really elegant solution that'll let me implement dashed lines and restore grouped pulses being triggered on the fly. As well as these other settings, or adding other features to the pulse train without causing exponentially more complexity.

tatarize, 2020-08-25:
@celticht32 In inkscape, go `Path->Stroke to Path` Then `Path->Break Apart` -- Make sure they have a correct stroke as they'll be little closed shapes. But, that'll get you passed that road block and move you on to more work. It'll be a bit before I get any line dashing in MeerK40t. I'm too afraid of screwing up the next version to switch yet.

Sophist-UK, 2023-04-25:
The reality of documentation is that you need to have a few volunteers to focus on this. For this you need a user-friendly and supportive community and actively to encourage users to contribute.

closed 2024-10-04 by jpirnay