meerk40t/meerk40t #189: bug in zinglplotter "ValueError: dx() or dy() exceeds 1"

jaredly, 2020-07-02:
When I try to execute this "Cut" operation, I get an error:
```
Traceback (most recent call last):
  File "/Users/jared/clone/fork/meerk40t/Kernel.py", line 1579, in run
    job.process(*job.args)
  File "/Users/jared/clone/fork/meerk40t/Kernel.py", line 231, in process_spool
    self.execute()
  File "/Users/jared/clone/fork/meerk40t/LhystudiosDevice.py", line 229, in execute
    x, y, on = next(self.plot)
  File "/Users/jared/clone/fork/meerk40t/zinglplotter.py", line 661, in groups
    raise ValueError("dx(%d) or dy(%d) exceeds 1" % (dx, dy))
ValueError: dx(-7) or dy(-20) exceeds 1
```

Here's a reduced SVG file that demonstrates the bug:
https://gist.github.com/jaredly/96d5dff21b8ad45716d561a62c3c6522

tatarize, 2020-07-02:
Yeah, this also spiked a few extra bugs with the plotting not returning a proper integer.
```
  File "LhystudiosDevice.py", line 121, in lhymicro_distance
TypeError: list indices must be integers or slices, not float
```

 I fixed that and figured it would work, but I ran into the one you have there too.

It's a flaw in the plotting routine. The core Zingl algorithms are actually originally set to plot in a graphics content and not necessarily as I use them a pixel by pixel routine. I mostly fixed them up, but here I clearly missed a spot the routine checks for a horizontal cut at P4, and a vertical cut at p6 and then swaps the points based on some criteria. The problem here is if it swaps the point it takes the cubic bezier curve backwards which isn't a problem for the original code but is a death knell for how I use it. 

```python
             if abs((y0 - 2 * y1 + y2) / t * x) > abs(y):  # /* which first? */
                    x0 = x2
                    x2 = x + x1
                    y0 = y2
                    y2 = y + y1  # /* swap points */
                    # /* now horizontal cut at P4 comes first */
```

But, rather than storing that and playing it backwards as I did at other swaps, I just let it happen and that's a problem. I updated zinglplotter in the commit I posted. You could just grab that from the 0.6.1 branch and it'll work. I'll make a couple tests to be sure I didn't miss anything else.

jaredly, 2020-07-02:
nice! Thanks for the quick response!

tatarize, 2020-07-02:
Thanks for the easily repeatable bug report. Made tracking it down pretty easy. Checked the various other code, and made some better tests in the zinglplotter project. Seems like there's no other whammies out there.

tatarize, 2020-07-16:
This is corrected in 0.6.1, now live.

closed 2020-07-16 by tatarize