meerk40t/meerk40t #191: Feature request: have the laser head move faster between strokes, when the cut speed is slow.

jaredly, 2020-07-04:
To cut through my 3mm HDF I've got the speed set to 10mm/s, and meerk40t currently also uses that speed when moving *between* cuts (from the end of one cut to the start of another cut). This, combined with the fact that there's no cut path optimization yet (from what I can tell) adds a significant amount of time to the job.
I don't know exactly what k40w is doing, but it's certainly moving the head at a more reasonable speed when going between cuts.

Thanks for an awesome program!

tatarize, 2020-07-05:
There was a weird issue with what I believed to be a race condition in the controller for @joerlane's use. I think if you hop out with `@` and then try to move the laser head in rapid mode, before setting the speed there's a rare chance it flies off weirdly. Under a time crunch to get 0.6.0 (it had been months since a version was published) since I didn't bother to run the needed tests to prove that theory, I just sort of did the moves within the same speed routine and didn't risk it.

Basically whisperer is jumping out to rapid mode, and performing the move. I could also switch the speed and do a move that looks like a rapid. Or rearrange the order of events so the rapid move happens where they happen in LaserDRW and Whisperer. Which is perhaps a safer way to do that. I could also just perform a finish (`F`) and get back out to rapid mode.  I'll likely add an option for that. Though the next version hyper-focused on usability might be spoken for already.

--

The lack of cut path optimizations is likely a much larger speed difference there. You can get to about a pareto-optimal travelling salesman approximation with two-opt in a really short amount of time and drop the travelling times by a factor of about 10. Especially if you permit for the reversing of a cut directions. There's a inner-first path optimization but that's mostly because you need to cut things inside first or it can ruin the job. 

Though if you compare that with the highly impressive Lightburn implemented cut planning routines, it's pretty minor, and highlights a couple other major gaps. Like planning tends to change the route and people need to have access to the that information so they know what to expect. This tends to make a simulation view a requirement, so you know how it will cut and how those cut planning options affect how it cuts. None of which is that hard. I've implemented two-opt for a very similar application before. The interactions for that with the requirements to "cut inside first" isn't something I know off the top of my head. It would likely just establish some ordering requirement and then you apply the other options as weighted heuristics applied to that path. #3 #114. 

Also, if I knew all the timing it might be the case that jumping out to rapid would take a certain amount of time. And that might be longer than just moving at the slower speed if the distance being travelled is below some threshold.

Might be worth it to add something more basic and cursory, to distance travel optimization. It's part of the reason I have a console after-all. I can implement a weird feature that does some cool stuff without needing to add all the UI stuff or working out the UX.

tatarize, 2020-07-06:
If you want a slight speed fix there the latest commit (https://github.com/meerk40t/meerk40t/commit/e4bcb8d6c9475ba030a514d9bae11a62e32d0c1b) in the 0.6.1 development branch will actually let you do TSP optimization.

In console type:

* `element * merge`

This will invoke the element command, `*` which does select all and `merge` which merges paths together. This will make everything into a single element. Which is what I coded up the optimization routine to use. Or just select the stuff you need and do `element merge`.

* `optimize travel`

This will optimize the current selected elements for the paths and if they are large compound objects like the merged thing the first command makes, it will optimize them. So a number of smaller parts can get put together the fastest way possible. Though if you do this for a cut, it will re-optimize it for inner first which means it will likely mess it up unless nothing is inside of anything else. Really it needs to do both of these together, but just threw together a quicky function. You can break this large merged object apart with `element subpath` which is the console command for "break subpath" etc. But, that's generally unneeded. 

The elements themselves can be remixed with in the operations so the operations themselves tend to need to cut planning done there rather than be done on the elements themselves. But, it's just a little console command laying some groundwork for stuff I have planned later.

jaredly, 2020-07-06:
very cool, thanks! So I guess for the "cut" scenario (where inside out needs to be preserved) speeding up the between-cut movement is still relevant?

tatarize, 2020-07-07:
It's relevant in both cases, and I plan to add an option for this. 

It used to run like that before Joe reported some runaway stuff so I basically edited out everywhere that could happen until I get around to making a good test to verify my theory and add them back in. With cuts, if it rearranges the layers it will likely still be a bit faster than it would have before, but running across the field to do one item and then all the way back to where you started, can easily be longer than a well optimized sequence all by itself.

I'd actually need to do both at the same time. Figure out which subpaths are inside what other subpaths. Flag them in a way to indicate that one subpath absolutely must occur before another. Then perform the two-opt travel optimization with that constraint. Such that, if the operation would violate the constraint it may not be performed. The rest of the seen and useful cut planning features are largely heuristics where you'd assign a value to continuing in the same direction and another value to the shortness of the travels and you'd do two-opt to do that combined path value.

There's a bunch of other considerations like you could cut-plan with different operations at different speed. The only functional difference with a cut is that it requires the constraint that if you must cut inner nested paths first (since functionally they fall out of the medium and then you can't do more operations on them). But, if you have two somewhat distance cuts and they are basically connected very closely with an engrave. You really could do one cut, then switch to engraving at a different speed, then do the other cut. Since that would shorten the overall time and the engraved line would need to be done at some point anyway.

I might scribble out the math to see how constraints and heuristics would play together and make something out of that. I don't think it would be too hard to combine them. But, then there's a bunch of other stuff like a simulation view that would be needed to flesh that work out more completely. And I still haven't really figured out what I'm doing with these colors-operations things. And that's supposed to be the focus currently. Since that'll make things much easier for people.

tatarize, 2020-09-20:
My code tends to go @FNSE<Move>V<speedcode>S1E... This may cause a race condition which is why I avoided rapid moves here.

```WinsealXP(R) Seal System BnDzzzzzz110@Lzzz055Tzzzzzz122NSEV2252523G004RdNRBS1ElDzzzzzz110```

Gotta fix my code here. @ Jump position, NSE VSpeed Jump NRBS1E<moves> is how the chinese software did that. This needs to be properly checked but there shouldn't be any code falling between NSE and the V of the next segment. This might be able to fall after the @ but before the NSE or after the V code info. There seemed to be a race condition elsewhere.

``K40 Whisperer ReTe@NSECV2112442051000031NRBS1ELeBeNLzzz|vTzzzzzzzzzzzzzzzzzzzzzz180SEFNSE```

Uses @NSECV and then jumps back into mode and runs N <Move> SE FNSE. This isn't a well understood code item.

```WinsealXP(R) Seal System IV2221554G000G003BcNBRS1EiDzzzzzz111```
Here, the Bc move occurs before the N but after the V.

tatarize, 2020-10-24:
This is fixed in 0.6.8, now live.

closed 2020-10-24 by tatarize