meerk40t/meerk40t #195: No Beep

kshenneman, 2020-07-11:
I have the settings setup to beep at the end of the process, but it does not beep.  I do have sound on my computer and I even turned my speakers all the way up, still no sound from the app.  I can however hear music or sound from YouTube or other sources.  It would be nice if you provided a default "beep", but allowed the end user to specify their own sound file.

tatarize, 2020-07-12:
There are some differences between OSes. The beep used is sort of a cop out. It prints the beep value. Which I think works for Mac and Linux but sucks for Windows. I could likely check the OS and send a winsound command as needed for windows.

```python 
if os.name == 'nt':
	import winsound
	winsound.Beep(900,500)
```

Or something of that sort.

tatarize, 2020-07-12:
https://github.com/meerk40t/meerk40t/commit/22d1ad0c3d0e1a4cabe02faa1c3c48f5d6363e52

Added a specific flag for that in windows (I'm assuming you're using windows).

I went with:

import winsound
winsound.Beep(900, 500)

If you'd like a have python you can try `import winsound` and try a few beeps, if there's one you'd prefer instead then feel free to ask. Just guesstimated a good beep.

kshenneman, 2020-07-12:
I am using Windows.  Any beep would work.  I tried what you came up with. Sure, whatever you think.

Maybe even something like this...  Something a little higher seems to get your attention more. I'm not saying you shouldn't be near the laser, but in my case the desk is a few feet from the laser, so I may be sitting there working on something else while its running.

import winsound
for x in range(5):
	winsound.Beep(2000,100)

Thanks,
Kris

tatarize, 2020-07-12:
Okay, that's less ear attacky, for a beep sequence.

tatarize, 2020-07-16:
This is corrected in 0.6.1, now live.

closed 2020-07-16 by tatarize