meerk40t/meerk40t #196: Right Click Rotate Does Not Work

kshenneman, 2020-07-12:
In version 0.6.0 on Windows, I open an SVG file, then try to rotate it using any of the right click rotate options. They all rotate it about the same distance, just a few degrees. They do not rotate it as the degrees list on the screen.  For instance, if I select "Rotate turn/2, 180", it will rotate it about 5 degrees clockwise.  If I select "Rotate turn/2 -180", it will rotate the image counter clockwise the same distance and return it to its original position.

tatarize, 2020-07-12:
Yeah, I noticed that and corrected it in 0.6.1. I switched the code from the normal rotate to pushing those commands through console. At issue though I used radians in the command values and just fed that into the console. But, since the console runs the rotation as through the same CSS Angle code I have it rotates in degrees. So 90 degrees becomes 1/4tau which is like 1.57 degrees. If you hit F12 and type rotate 90, or rotate 0.5 turns, that'll work correctly.

It's fixed already in the next version.

tatarize, 2020-07-16:
This is corrected in 0.6.1, now live.

closed 2020-07-16 by tatarize