meerk40t/meerk40t #199: Power and PPI Settings Not Being Saved

kshenneman, 2020-07-20:
Installed 0.6.2.  When changing the speed and ppi settings for a particular file, then selecting File > Save or Save As, the power and ppi settings are not saved for a file.  Then, if you open another file and enter different settings, then open the original file, the settings for the second file are displayed, instead of the settings that should have been saved for the first file.

tatarize, 2020-07-20:
I tested that procedure. I don't get the same results.

Does the file internally save that data? You can open it up with a text editor, it's just svg xml so it should show the operations. I am not finding any failure to save these items.

kshenneman, 2020-07-20:
OK, So I retested this morning. I think I may have accidentally saved over my first file yesterday, at some point and didn't notice. So, I think you are right. The save does seem to be working.  However during my testing, I did notice that if I have a raster and a cut, and I disable the cut and save the file, the disabled setting for the cut doesn't save.  I have a cut layer on there that I don't normally use, but I do use it to occasionally check alignment.

tatarize, 2020-07-21:
Yeah, I'm getting other similar reports so I'm not sure. It should be harder to mess up, or it's saving at some point it maybe shouldn't. Or the way it loads files without operations makes things confusing or something. It's might actually be something.

tatarize, 2020-07-24:
The saving might be improved with it treating MeerK40t SVG files different. Loaded MeerK40t SVG files can just be saved rather than save-as. Also, it's a bit easier to open recent files with recents.

closed 2020-07-24 by tatarize