meerk40t/meerk40t #200: Crash Log FYI

Daparat, 2020-07-22:
MeerK40t crash log. Version: 0.6.2
Please report to: https://github.com/meerk40t/meerk40t/issues

Traceback (most recent call last):
  File "PyInstaller\loader\pyiboot01_bootstrap.py", line 167, in __init__
  File "ctypes\__init__.py", line 364, in __init__
OSError: [WinError 126] The specified module could not be found

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "Controller.py", line 347, in on_button_connect
  File "Kernel.py", line 1773, in execute
  File "LhystudiosDevice.py", line 1057, in open
  File "LhystudiosDevice.py", line 1209, in detect_driver_and_open
  File "CH341WindllDriver.py", line 20, in __init__
  File "ctypes\__init__.py", line 442, in LoadLibrary
  File "PyInstaller\loader\pyiboot01_bootstrap.py", line 169, in __init__
PyInstallerImportError: Failed to load dynlib/dll 'CH341DLL.dll'. Most probably this dynlib/dll was not found when the application was frozen.

tatarize, 2020-07-22:
Hm. It usually always finds the dll since it installs when the K40 is plugged into the PC. That or people have Whisperer and it'll use the libusb driver there. I guess either install the CH341DLL manually, or grab a copy of Whisperer and it install the libusb driver. I note it's a pyinstaller error so it's part of the compiled version. Maybe I need to include the CH341DLL.dll into the compiled exe to be safe.

Any background? Did run MeerK40t with never having plugged in the laser? -- I guess the easiest method here would be catch the OS error. And run some tests with absolutely no driver at all. It tries the Ch341DLL after it fails finding the LibUSB backend so it really would have to have absolutely no driver at all to fail like that.

It is certainly something I did not test against though.

Daparat, 2020-07-23:
HI,

Whisperer had the driver "locked", releasing the driver from whisperer allowed Merk40t to work, alternatively a simple reboot and not starting whisperer also resolves.

tatarize, 2020-07-23:
Ah, that would explain why it went to the windows driver fallback and couldn't find the windows driver. I'll try to make sure it fails more gracefully in the future though.

tatarize, 2020-07-24:
This will send out a connection error rather than a crash if you do this in the future. Fixed 0.6.3

closed 2020-07-24 by tatarize