meerk40t/meerk40t #203: Crash on First Close for rPi and OSX.

tatarize, 2020-08-03:
File "meerk40t/Kernel.py", line 1484, in run
File "meerk40t/Kernel.py", line 1682, in run
File "meerk40t/Kernel.py", line 2085, in shutdown
File "meerk40t/Kernel.py", line 2102, in save_keymap_alias
wx._core.wxAssertionError: C++ assertion "szValue.empty()" failed at /Users/.../projects/bb2/dist-osx-py38/build/ext/wxWidgets/src/common/fileconf.cpp(920) in DoWriteString(): can't set value of a group!

First close can't set the name.

tatarize, 2020-08-05:
This should be fixed circa 0.6.4. The error was the key "numpad\" which was registered as that key but the `\` looks like a path notation and created an issue there.

joerlane, 2020-08-05:
Looks good on both Mac and rPi. now.

closed 2020-08-05 by joerlane