meerk40t/meerk40t #204: PyUsb and Ubuntu 20.04

pkarza, 2020-08-14:
I can successfully install the executable for Linux on Ubuntu 20.04 - but I can't access the USB port - it tries and fails - saying that PyUsb is not installed.  20.04 uses Python3 - and I have installed pip3 install python3-usb ... but this does not work.

tatarize, 2020-08-14:
It would be `pip install pyusb` though if it's the executable that should already be included since it should package like that. @joerlane, did you not install pyusb in the linux build?

joerlane, 2020-08-14:
The MeerK40t_Linux.tar binary has been updated to include pyusb. If you download the new version (which is about 20 meg larger than before) it should work. My build log from last time does indicate that it failed to hook pyusb during the packaging process.

pkarza, 2020-08-15:
Yes !  it works.  6.3 is now confirmed to be working in Ubuntu 20.04.

I tried the executable in 18.04 and that does not work - but that is ok - 20.04 is new .

Tatarize - 20.04 no longer allows the use of pip or python - ONLY python 3 is supported.  Frustrating....

Thank you - now to figure this out.  Is there a wiki??!!  ( I will look)

joerlane, 2020-08-15:
I added two new builds to the 0.6.3 release. There had been some issues getting wxPython to update on some of my linux build systems; so not everything was at version parity before... That's fixed. 

I uploaded a release compiled on that Ubuntu 18.04; and one compiled on centOS. The centOS might be universal if it didn't have a UI glitch in the tree (missing drop down triangles); but at list this one gives some indication of "where" they should be on screen. That's a slight improvement from the wxPython 4.0.7 behavior where it was pure white left of the objects and operations in the tree. 

The one labeled Ubuntu LTS might be the reigning champion for a universal linux binary build at this point. I've tested it on all my other systems and it's working fine. It should work on both Ubuntu 18 and 20; probably many other things as well.

tatarize, 2020-08-18:
Jan 1st, 2020 was Python 2 End-Of-Life. A lot of people dropped python 2 support. I'm not even sure meerk40t is compatible with python2 anymore. I dropped support at that time for all my python packages too. There is a wiki, but it might need updating for 20.04.

https://github.com/meerk40t/meerk40t/wiki/Installing-MeerK40t-on-Ubuntu-18.04.3

Joe would likely have a better idea, since that wiki's like 8 months old and I dunno where my linux stick is currently. And I checked and the instructions are a bit meh.

joerlane, 2020-08-19:
I started trying to edit that 18.04.3 instruction wiki a few days ago. I recognized that it mirrored much of my experience doing it myself later on (more recently; and with other distros). It was mostly a guess and check affair, seeing what wasn't installed; over and over again. I have quite a few notes on versions/flavors with similar "if-not-this"-than do install that; type instructions. 

Honestly not sure the best path forward. Other than; it might be time to work towards some kind of legitimate installation paradigm. Not sure what other programs install more easily; any references would be helpful. I do this often enough that I can easily ignore the substantial hurdles others might find... Scripting might be a cure all around?

pkarza, 2020-08-19:
I think you might be on to something.  I would be willing to help in any
way possible.

Honestly - before finding meerk40t, I was considering making a grbl
controller or buying a cohesion board - I don't need to, now.

I will be sending pictures of my pi, tomorrow (in my 3d printed 2013 mac
pro case) and include some other documentation.  Just waiting for my usb 3
hub... should be here tomorrow!
Phil

tatarize, 2020-08-19:
"Honestly - before finding meerk40t, I was considering making a grbl controller or buying a cohesion board - I don't need to, now." --- Thank you. This is literally a goal Joe and I have discussed at length.

joerlane, 2020-08-20:
I feel the same way. When I bought my K40; I had fully planned to change the controller; as so many people recommended it as an "essential upgrade". I even had a spare controller on hand at the time (and still do). 

Originally I read that as: "3d printed 2003 Mac pro case"; and I was so confused. It took me a few more reads to add the decade, to make "2013"; and suspect that might be the "trash can"? That's really a great design!

pkarza, 2020-08-21:
Yep... 2013... can't type!

closed 2020-09-30 by joerlane