meerk40t/meerk40t #205: raspberry pi

pkarza, 2020-08-16:
I am really enjoying the program  - much to learn - and I thought it would be great to use with a raspberry pi.  I loaded ubuntu LTS 18.04 server with xubuntu desktop, and neither the LTS, Ubuntu, or pi tarball worked.

Are there any preconditions that need to be satisfied?

Thanks

Phil

joerlane, 2020-08-16:
I've never tried ubuntu on the Pi. That binary is compiled in raspbian. 

I noticed there are both 32 and 64 bit version of Ubuntu for the Pi. What Pi version (2,3,4) do you have? And did you opt for 32 bit or 64 bit software? I'll have to pick up an SD card and whip up another build to see what's up with that.

pkarza, 2020-08-16:
I picked up the 64 bit raspberry pi 4 version.  It was the Ubuntu server
18.o4 lts on a SanDisk ultra 32 gb with xubuntu desktop.  Thank you for
your great work

pkarza, 2020-08-16:
I will also try raspbian but currently it is 32 bit

joerlane, 2020-08-16:
I picked up another microSD card; and I'm looking for a spare raspberry pi at the moment. I would hedge a bet that the current Pi binary is too new for 32 bit Ubuntu 18. But I'll give that a try when I get up and running. I suspect the main issue here is going to be the need for a 64 bit Pi binary, as well as the existing 32 bit. 

I'll update once I've had a chance to run a few of those combinations, and post up a 64 bit Pi binary as soon as I get it built.

pkarza, 2020-08-16:
Here is an update:  I flashed the new 32bit raspberry pi OS Raspberry Pi OS (32-bit) with desktop
Image with desktop based on Debian Buster
Version: May 2020
Release date: 2020-05-27
Kernel version: 4.19
Size: 1128 MB

I was first able to install k40whisperer using python 3 - and included all libraries... 

Next, I grabbed the copy of Raspberry pi 6.3 and installed it.  I clicked to execute... and waited... thought it would not work and then ... it did!

I am now updating the firmware to boot from and external SSD.  I am happy that it WORKS but it is 32 bit.   Would I notice that much of a difference if I used 64 bit?

joerlane, 2020-08-16:
I presume what you would notice is that the 32 bit binary version doesn't run on 64 bit OS (just my guess; and certainly not going to work other way around)? I haven't gotten quite that far yet on my end. I've got two images downloaded for Arm/Pi based Ubuntu; but my test machine is running another install at the moment. Once I get those 64 bit Ubuntu images burned to flash drives for install, and loaded, I'll have more information on that. I suspect releasing a 64 bit Pi binary along side the existing 32 bit will be enough; but we'll see how that works out.

I agree; there is no reason to run a 32 bit OS when there are 64 bit options available. You are simply the first to ask. I spent quite a long time trying to get the original Raspberry Pi B+ running with MeerK40t but never quite got it functional in a universal binary build.

Yeah; the 32 bit build is a bit slow on the Pi. But the Pi is a bit slow to begin with. Once launched, it seems to be pretty responsive. I use a Pi connected to my K40 pretty much exclusively; so I can vouch for its overall functionality. The only problem I've run into is it seems to run out of memory loading EXTREMELY large dxf files; but even complex .svg files have never been a problem for me. It is nice to have a dedicated machine on the K40; my Pi with a touch screen and MeerK40t is right next to the mouse on my main computer.

pkarza, 2020-08-16:
I'm looking forward to the updates from you.

An update for you - the load time from an external ssd is under 5
seconds... probably under 4.  I have the Pi 4 GB and haven't had a problem,
yet.  Running with SD is abysmal for launch - but works.  The SSD makes it
completely different.

joerlane, 2020-08-17:
How is this external SSD connected; USB 3.0 I presume? I've never tried USB booting myself. Would be curious about the speed over USB 2.0... You might be able to swap the ports (depending on your hardware)?

I'm currently trying to load a multi-boot system onto the internal microSD. This first net-install is taking quite a long time. The USB 2.0 flash drive image installs should be a lot faster once I get to that point. Still bound to a local keyboard and mouse for the initial setup. First time I tried BerryBoot, but it seemed to fit the bill for what I need verify MeerK40t binaries.

I still don't own a Pi 4; but you have me intrigued. It wouldn't take much of a push for me to finally upgrade; then I can shuffle some other hardware down the line. The official 7" touch screen is a near perfect fit in the standard k40 controls cutout. I've been thinking about doing that for a long time. A super fast Pi 4 installed in the K40; and served over VNC, might just be the ticket for me.

pkarza, 2020-08-17:
Hey Joseph -

The SSD is connected through a USB3 port to an external hub- powered by 5V
2A.  I imagine there would be quite a slowdown going to usb2 from 3 - I
will use one of the USB 2 ports for a wireless keyboard dongle and the
other to the K40.

The Pi 4 is a great advancement.  I use a pi 3 on my CR10 with octoprint -
and the touchUI is much slower on it - as is the entire system - but as it
is only sending gcode to the printer, the speed is more than adequate.

I have the 4gb pi4, and in reality, I don't need any more than that.  I
won't be multitasking, and it flies.  Tom's hardware has a great article on
how to update the firmware and clone the sd card to the SSD.  The SSD will
power off the pi - but the 4 is power hungry and it is best to put it on an
external hub. I didn't try it, but the article said it would work, but
other peripherals would suffer.

After I get it all cleaned up (it is a mess) I will send you a pic of my
setup.

my email is pkarza@gmail.com - I am in Canada

tatarize, 2020-08-17:
It should be entirely reasonable to just run it from the python code directly, on any OS hopefully with helpful instructions, especially if it's a more obscure OS, or an obscure OS flavor. Mostly the pre-compiles for Windows and Mac users who likely would have trouble running from the python and would have a much better experience with it, and thus are solid marketing. Telling some Mac users their problems are all solved out of the gate by running this thing, is great. And Raspberry Pi, since that setup should be be a good baseline for a quinicential slow computer and typically smaller screens. And it's worth avoiding spinners and other graphics elements that suck on the rPi to achieve that end.

But, this is way more Joe than it is me. If he wants to cook up a 32 bit BeOS version, more power to him. I only really need the Mac and Pi versions and I cook up the Windows precomp. If the pure python fails for some reason, that's an issue for me.

Though I would recommend labeling the tar for 0.6.4 as Raspian Pi. I'll just watch this from afar.

pkarza, 2020-08-17:
I've noticed that sometimes the python code doesn't seem to work on the pi
- the raspberry os sometimes loses something in the translation as python
works... sometimes - I have had no problem with most python code, however.

joerlane, 2020-08-20:
I've not had any specific problems with python MeerK40t on the Pi. In fact, it's almost exactly 10x faster to launch in python than the compiled binary on the Pi I use regularly. As a means of testing I run the Pi binary most of the time; I do sometimes find issues with the binary, where it works fine in python. So far I've not seen anything the other way around though (python doesn't work but binary does). Are you maybe referring to the errors that show up in the CLI; regarding gtk errors, or pixmap *** BUGS ***? I sure would be interested if there are any functional issues with the software/operation that are reproducible. 

Spent some time trying to dual boot with 64 bit OSs on the test Pi; but wasn't successful. I gather that BerryBoot on a Pi3 may be a limiting factor out of the box, and already ordered a Pi 4 to make sure everything works out to compile 64 bit builds for the arm. The more recent Pi 4 seems to be a bit better supported for what I was trying to do in this case.

At this point, it's kind of a race to see if the Pi 4 arrives first; or I figure out how to get a Pi 3 doing what I need. In the mean time; I've saved all the OS images I downloaded (mostly for Pi3). It seems likely that I'll be able to load at least some of those individually; (without dual-boot or multi-boot on the same SD card).  I've seen some reported issues related to RAM being too small for first boot on the Pi3s, but I doubt that's been a problem for me so far. I probably should have bought a stack of microSD cards the other day; but I wasn't expecting this outcome at the time... I also saw a mention that 64 gig cards be a problem in some specific situations; but I'm not at all convinced that's been my problem so far either. 

Dragged the 7" touch screen down here, and have it mostly laid out for a new control panel setup. Just now realizing I need to think more about wiring since I have a camera setup already, and a few other things ready to connect.

pkarza, 2020-08-21:
> Did you just untar the linux version?

... I guess I need a little help - I can launch the program from the
compiled binary - but I can't seem to get it to launch directly in python -

Could you tell me how you did it?  I can't even get wxPython to install

joerlane, 2020-08-21:
Ah yes. I'm actually working on that right now. Finally got a Pi 3 to boot arm64...

Still working on installing the GUI but I can detail the process in a short while. What OS are you trying to install wxPython in?

joerlane, 2020-08-21:
So; here's a quick rundown of my 64 bit Pi install process on Ubuntu. The **() are comments; otherwise commands to run. Should be very similar on other Pi OSs. But I'd be happy to walk you through any specific one if you want. 

python3 -V **(check python3 version)
sudo apt install python3-pip
sudo apt install zlib1g-dev
sudo apt install libjpeg-dev
sudo apt install libgtk-3-dev

which git **(check to see if git is installed; mine was already)
cd /to/whatever/dir/you/want **(you'll need to supply the MeerK40t install path you prefer here)
git clone https://github.com/meerk40t/meerk40t.git

pip3 install wxPython **(This takes a REALLY long time (on an original raspberry Pi model B it takes 96 hours; four days...); this install also collects Pillow, six, and numpy)
**(If that install did not fail; then MeerK40t will run. But you need a few more things to use the laser over USB, open DXF files, and use the camera features)
pip3 install pyusb ezdxf 
pip3 install opencv-python-headless==3.4.8.29 **(this doesn't NEED to be 3.4.8.29 unless you want to build binaries - that specific version may not even be available depending on the distro/version)

cd meerk40t
python3 MeerK40t.py

One final note. libusb was already installed for me; but it too is a requirement for pyusb. It's not installed by default on Mac OS X for example.

joerlane, 2020-08-24:
If that's true; you could build a wxPython wheel. What version of Python are you running there? What OS? I just now powered up my Pi4 and still don't have a display; or handle on what's next. 

My Pi3 64 bit failed again (perhaps the 5th time?); but I'm looking into my options. I'm suspicious of the python version in this case.

joerlane, 2020-09-07:
Took me a while... But I was able to build a binary for the 64 bit OS. For compatibility sake that has to be done on the oldest OS possible; so that was "fun"... Took a few weeks on and off trying to solve various problems with the build build prerequisites. It is finally uploaded to the release section for the latest version (0.6.4). I'll continue with these 64 bit builds going forward.

One thing I noticed recently was that all Linux binary builds are considerably slower to launch than running in python. But those binaries can also be built with loose components; and those builds run at full speed. In that case the main app is living inside a folder full of necessary support files. Once launched, both python and the binary builds appear to operate at the same speed. Manipulating objects, and loading windows, all happen at the same speed in either case. 

I presume that the single file binaries are compressed in some way; and need to be unpacked on the fly during the launch process. Otherwise; that's how fast the Pi is. It's certainly not desktop speed. Is there anything else the Raspberry Pi is fast or responsive with? Just opening a Terminal window is painfully slow. With SVG files that are already set up (proper colors, speeds, and power settings); it still only takes a few moments to load a file, and start the job. 

I'd be curious to know how that 64 bit setup is working; or if the 32 bit turns out to be better for some other reason? I only just finished this build a few minutes ago; so I haven't had a chance to compare in depth.

joerlane, 2020-09-30:
Binary is being compiled regularly now that I have a working 64 bit Pi OS running for that purpose (ubuntu).

closed 2020-09-30 by joerlane