meerk40t/meerk40t #208: Real Progress Bar for Jobs

tatarize, 2020-08-24:
Estimate the time remaining on the job. With proper updates with the known information about where we currently are, the distances left to travel, and how long that will take to finish.

The idea is to help business minded folks who need to know how long they have left on a project before they can start the new project. This likely should be built into the preview-progress view dialog info.

tatarize, 2020-11-22:
Alternatively a minor progress information can be set pretty easily. It would basically say when the last job was started and when the last job finished. And give you the duration in the console.

tatarize, 2021-08-09:
"There is/was plans to do this. You can kind of guess how long the job has been running for and how much of the job is left to do. Also, it would be possible to adjust the time estimate as the job progresses. That was at least the plan. There'd be a couple ways of doing that. Either indicate it in scene or make a gui pane to track the estimate. With updates for the spooled amount in the writer."

jaredly, 2022-05-10:
k40whisperer clearly displays "% of commands sent to the device" which is hugely useful. By far the feature I miss the most when using meerk40t.

closed 2023-04-04 by jpirnay