meerk40t/meerk40t #210: Feature Request: dashed lines

celticht32, 2020-08-25:
Would like to have the ability to create dashed lines and have the software cut a dashed line.

tatarize, 2020-08-25:
For now, in inkscape. `Paths->Stroke To Path` ... `Paths->Break Apart` on a dashed array is a workable workaround. It does create filled objects rather than stroked. And the paths themselves are actually closed paths there rather rather than small line segments so it's a bit jerky but it would work.

I added in most of PlotPlanner which gives a sort of central position to wrap and modify the pulse train being sent out to the laser. It seems to work, and adding dashes to that would be pretty easy. Likely I could add a full on dasharray. Though weirdly this might have to occur before the application of the PPI. Since the whole dot length math isn't my friend. And I don't want a repeat of the 1500 ppi bug. But, just straight dasharray would be fine.

littlemandave, 2020-11-12:
I've just run into this too: a dashed line in Inkscape imports as a solid one. Nice that I can use the workaround above, thanks! This time I'll just use whisperer for that job. Probably because it uses Inkscape for some of its work, the dashes are respected.

tatarize, 2021-01-30:
0.7.0 has some line dashing with dot length. It's unknown if that should settle this issue. But, it would likely create the holes needed in leather or whatever people would use a dashed line for.

closed 2021-05-20 by tatarize