meerk40t/meerk40t #211: Rotary Y-Axis Return to Home

kshenneman, 2020-08-25:
When using a rotary, you plug it into the Y axis plug on the K40 type lasers when using the stock M2 Nano board.  Therefore, when you turn on the machine or press the home button, the rotary will continually spin until you manually press the stopper with your finger.  That makes absolute sense.  With Meerk40t, when a job is complete, and you have the "Home after job complete" setting checked, the X axis will return to home and the rotary will just spin forever until you press the stopper with your finger.  However, when using K40Whisperer and a rotary, when the job completes, the rotary only spins backward as far as it originally moved during the job, basically returning home and stops.  Could you please alter the functionality of Meerk40t to be more like K40Whisperer in this regard.  So basically, count the number of Y axis steps during the job, and reverse that number of steps at the end of the job and stop.  Otherwise, if you have something else you need to do to the object, you will never get it lined up properly.

tatarize, 2020-08-25:
Hm. Sounds like the home after job complete should just be turned off if we're using the rotary. It sends an issue for a homing command to the board. K40 Whisperer doesn't store the known position of the head so the only way it knows where it is, is by always returning to where it started so it generally just know it should be about where you put it. It does however lose a mil or two in those calculations.

MeerK40t actually knows exactly where it is at all times. If you selected the object when you finished and told navigation to go to the upper left hand corner it would return to that position (assume that's where you started from since it doesn't much matter, where you started). 

https://www.youtube.com/watch?v=LvZLu4_GgNM

What you want isn't home after job, since that actually homes the laser (goes up and left until it hits both endstops). Or even return to 0,0 which is actually what I had to change the Visicut driver to do. Since if you unlocked the rail and moved the head, and homed between each of them. The first would be where you put it, but the second would be relative to the real home position on the laser.

It already counts every movement the laser makes. So I'm guessing you want either "return to 0,0" or "return to start position". Or maybe go to the defined home point without homing the laser. Return to 0,0. Would make sense, I could add that to the jobs.

kshenneman, 2020-08-25:
Interesting. When I was using the rotary, I had the home X location set to the equivalent of 80mm, I forget the exact mils, something like 3000 and some. The Y was 0 because I moved the rail manually. So, I would want it to return back to where it started. If I understand, next time I will turn off the home checkbox.

tatarize, 2020-08-26:
Yeah, but for a rotary you'd have to move the position yourself. You don't have a real y-position, at least not really. I'm turning the Home functions off if Rotary is on. At least the automatic ones. And adding in a return to origin command. Which should return to 0,0. That you could use instead.

Also, it might be useful to mark your start position. If you press control+1 (works for numbers 1-5). And you can go to that position when you press 1. So when you're done with whatever, you can go to that position. Even though it's technically not an origin position or a home position.

tatarize, 2020-08-26:
https://youtu.be/bfUSbh--yyg

closed 2020-09-22 by tatarize