meerk40t/meerk40t #212: Save multiple different home points

ViolentCrumble, 2020-08-27:
Just had a thought that when i am engraving different size jars, I plan to CNC a wood block with holes for the jar lids to be placed inside of. It would be super useful to position the head in the top left corner of each size and save it as a home position.

Then easily click different home positions depending on what you are engraving?

As far as i saw this is not a current feature, Sure i could write down the positions but this would be a nice to have!

tatarize, 2020-08-27:
On the keyboard use control+1 control+2 control+3... up to control+5. Then on the keyboard 1, 2, 3, 4, 5 will bind to the location. You can have it auto-move your head to that position. They aren't quite different home positions, but they do start move there. So long as the object is in the right position in the field it will always go to that location. Regardless what you consider home to be. 

https://www.youtube.com/watch?v=LvZLu4_GgNM

ViolentCrumble, 2020-08-27:
hell yes that works! thank you!

tatarize, 2020-08-27:
One of the other admins is pretty adamant that you're wanting to set your jigged position and use the position as home. While that will jump to that location. The location in the field is not the location it will end up running it. If the item is located at 0,0 and you're head is located at say 5000,2000 your head will go -5000,-2000 and *start* the item from 0,0 regardless where you head is located when you tell it to start.

tatarize, 2020-08-27:
There is an alternative home location in settings where if you have things jigged to that location when you home. It will go to 0,0. Then perform a stealth move that it ignores. And that point becomes the actual new 0,0 location.

There's also the option of taking everything your working with and moving it such that your current position is basically the same as it *would* be if you were at 0,0.

tatarize, 2020-08-27:
In console, if you type:
alias jig0 set home_adjust_x 3000;set home_adjust_y 2000;home
bind 6 jig0

Then the key 6, would set you to a jigged 3000 2000 location (3in, 2in) which would work exactly like that location is your home location.

ViolentCrumble, 2020-08-27:
basically i am cutting a jig from wood that has a 10cm hole and an 11cm hole. For 2 different size jar lids that i sell and engrave.

At the moment in inkscape i have a template saved and all i need to do is import the artwork and position it then save as.
when i load it into k40 whisperer i know i hit right once and down 4 times and the laser will be exactly at the top left of my jar lid and it will engraved centered.

When i want to do is have 2 holes so i can place in a jar lid in my template. click "home2" which i know is 10cm jars and the laser moves to the top left of that circle so my artwork will laser exactly center on the jar lid.

At the moment i just have a bit of wood that i vector engraved circles on. and then hand place the jade lid inside that circle, but due to tiny variations in human error it is not always perfect center.

being able to save as many homes as we want and naming them would be ideal! but i can also right down the positions of the head when i get it perfect and save it in a notepad file. Provided i can move head to x and then start the job from there.

tatarize, 2020-08-27:
Yeah, you'd have to do the designs within the inkscape jig. You can then put them correctly. Otherwise you'd want to jig the location with the adjust_home values. there. The difference here is that whisperer never tracks the location. So when you move the head you are moving the start location. In MeerK40t when you move the head you are just moving the head. The start location is where-ever the stuff is in the field. So if you want it to start at 10cm 10cm you would put your work there. And whereever the laser head is it will go to 10cm 10cm and start the job.

The adjust home position, will cause it to home but then move to a jig location. You can bind keys to set these values and call home pretty quickly. So if you know where you're jigging to you can set that as a home position by changing the value of the adjusted home. 

But, the main thing is 0,0 if that's where your artwork is, your laserhead position doesn't matter because it will go to where the artwork is. So either have your artwork done with an expected jigging or add in a couple adjustment for the home location as above.

ViolentCrumble, 2020-08-27:
ah roger ok. so probably just easier to keep using k40 for these lids then. I am just trying to get rid of the minor human error. I cant place jar at 0,0 so that is not an option.

A different feature suggestion is probably start job from current position then.

tatarize, 2020-08-27:
It either needs to use the adjust home position routine. The commands I gave there would set a jig position. Or it need to have the actual designs where they are intended to be positioned. You can set that in navigator. Just lock to the corner and move it to the right position with the middle buttons in navigator.

tatarize, 2021-02-10:
This work is completed in 0.6.19-beta2 with the addition of home taking optional arguments. Namely it takes x and y which override the default. So `home 2in 2in` in console will do a home and override that offset value set in Preferences. You can bind these keys to your various home locations.

https://github.com/meerk40t/meerk40t/releases/tag/0.6.19-beta2

closed 2021-02-11 by tatarize