meerk40t/meerk40t #221: Tooltip typo in rotary scale factor

johnsonm, 2020-09-15:
Some things feel too small for a PR... Cut and paste X/Y typo:

```diff --git a/RotarySettings.py b/RotarySettings.py
index 2a4ab62..159de1b 100644
--- a/RotarySettings.py
+++ b/RotarySettings.py
@@ -98,7 +98,7 @@ class RotarySettings(wx.Frame, Module):
             wx.Font(12, wx.FONTFAMILY_DEFAULT, wx.FONTSTYLE_NORMAL, wx.FONTWEIGHT_NORMAL, 0, "Segoe UI"))
         self.checkbox_rotary.SetToolTip(_("Use Rotary Settings"))
         self.spin_rotary_scaley.SetMinSize((80, 23))
-        self.spin_rotary_scaley.SetToolTip(_("Rotary Scale Factor X"))
+        self.spin_rotary_scaley.SetToolTip(_("Rotary Scale Factor Y"))
         self.spin_rotary_scaley.Enable(False)
         self.spin_rotary_scaley.SetIncrement(0.01)
         self.spin_rotary_scalex.SetMinSize((80, 23))
```

tatarize, 2020-09-15:
https://github.com/meerk40t/meerk40t/commit/89b3044177ce040657abeda05744ba7b1fd2511b

tatarize, 2020-09-23:
This was corrected with 0.6.5.

closed 2020-09-23 by tatarize