meerk40t/meerk40t #223: M2 Nano - USB power

slalum01, 2020-09-16:
I know this is not a meerk40t issue but i'm desparate to find a solution to my problem. Having problems getting a M2 Nano to work after replacing a Moshi MS10105 in a K40. I initially tried to connect outside the K40 with only a USB cable to see if computer would detect it: no USB device found by Windows 10, even using usbdeview utility, tested on three different PCs. Tried Zadig driver installer as per K40 Whisperer instructions, no go. Connected power lines from the K40 power supply, still no detection. Latest 2019 v.9 version of the board has missing components from previous 2018 v.9 next to USB connector but seller on eBay says it is normal with this "improved" version. Can the board run "standalone" from USB to install drivers or does it need power from the K40 (5 volts?)? Am I missing anything?
Any thoughts?
Thanks!

joerlane, 2020-09-17:
No; it can't run standalone on 5 volts from the USB port. I have a few extra M2 Nano boards here in my desk, and they don't communicate at all without the full 24v machine power.

slalum01, 2020-09-17:
Great, thank you for the reply. I'll reconnect everything and start troubleshooting again. Power connections to the Nano is pretty simple so I don't think I screwed up that part. It'll test it without any other connections.

joerlane, 2020-09-30:
Tested this one last time to be sure. USB power doesn't present power on the 5V power connector. 

Feeding 5V into the 5V power connector input; and connecting the USB port still doesn't wake up the controllers. It really does need that 24V power as well, to communicate. As soon as it boots it will self home; so having end-stops connected is also reasonably important at that stage of testing.

closed 2020-09-30 by joerlane