meerk40t/meerk40t #224: .dfx files with curves are not working

gyan007, 2020-09-20:
MeerK40t crash log. Version: 0.6.4
Please report to: https://github.com/meerk40t/meerk40t/issues

Traceback (most recent call last):
  File "D:\meerk40t-master\wxMeerK40t.py", line 1334, in on_click_open
    self.load(pathname)
  File "D:\meerk40t-master\wxMeerK40t.py", line 923, in load
    results = self.device.load(pathname, channel=self.device.channel_open('load'))
  File "D:\meerk40t-master\:Kernel.py", line 2358, in load
    return self.device_root.load(pathname, **kwargs)
  File "D:\meerk40t-master\Kernel.py", line 1299, in load
    results = loader.load(self.device, pathname, **kwargs)
  File "D:\meerk40t-master\DefaultModules.py", line 240, in load
    import ezdxf
ModuleNotFoundError: No module named 'ezdxf'

gyan007, 2020-09-20:
[https://www.freepatternsarea.com/wp-content/uploads/template-files/projects/elephant-phone-pencil-holder-laser-cutting-product-sample.dxf](url)

tatarize, 2020-09-20:
If you are running a precompiled version, please tell us which version. If you're running it from python directly type ```pip install ezdxf``` it's an optional requirements since not everybody will mess with dxf files. But, here the error is just saying the reader doesn't exist and needs to be installed. The latest windows version works for that so it's pure python or one of @joerlane compiles.

tatarize, 2020-09-20:
MeerK40t crash log. Version: 0.6.4
Please report to: https://github.com/meerk40t/meerk40t/issues

Traceback (most recent call last):
  File "wxMeerK40t.py", line 946, in on_drop_file
  File "wxMeerK40t.py", line 923, in load
  File "Kernel.py", line 2358, in load
  File "Kernel.py", line 1299, in load
  File "DefaultModules.py", line 382, in load
  File "ezdxf\entities\dxfentity.py", line 121, in __getattr__
ezdxf.lldxf.const.DXFAttributeError: Invalid DXF attribute "text" for entity MTEXT

You posted the wrong log. Yeah, that doesn't load.

tatarize, 2020-09-20:
[image]

The MText entry is wrong. Fixing ```element = SVGText(x=insert[0], y=insert[1], text=entity.dxf.text)``` to be ```element = SVGText(x=insert[0], y=insert[1], text=entity.text)``` will make it load.

tatarize, 2020-09-20:
Apparently the MText also needs parsing:

'{\\fTimes New Roman|b0|i0|c162|p18;freepatternsarea.com}' This got interpreted as being a particular size and font in my other dxf application.

tatarize, 2020-09-20:
https://ezdxf.readthedocs.io/en/stable/tutorials/mtext.html

```MText supports inline formatting by special codes: MText Inline Codes

mtext.text = "{\\C1red text} - {\\C3green text} - {\\C5blue text}"
```

https://github.com/meerk40t/meerk40t/commit/5ff8656512fa0d1a50fa445eda3af78b39a0214b

Has a correction for the big issue. Dunno about the lesser parsing one but I guess I should get around to fixing that parsing defect.

gyan007, 2020-09-21:
I am using it directly form python. I used the dfx file after deleting the text. 
I just pip installed ezdxf but new errors show up: 

MeerK40t crash log. Version: 0.6.4
Please report to: https://github.com/meerk40t/meerk40t/issues

Traceback (most recent call last):
  File "D:\Alula Technologies\meerk40t-master\Kernel.py", line 2019, in run
    job.process(*job.args)
  File "D:\Alula Technologies\meerk40t-master\Kernel.py", line 268, in process_spool
    self.execute()
  File "D:\Alula Technologies\meerk40t-master\LhystudiosDevice.py", line 231, in execute
    x, y, on = next(self.plot)
  File "D:\Alula Technologies\meerk40t-master\zinglplotter.py", line 674, in groups
    raise ValueError("dx(%d) or dy(%d) exceeds 1" % (dx, dy))
ValueError: dx(457) or dy(-1173) exceeds 1

gyan007, 2020-09-21:
[elephant-phone-pencil-holder-laser-cutting-product-sample.zip](https://github.com/meerk40t/meerk40t/files/5253442/elephant-phone-pencil-holder-laser-cutting-product-sample.zip)

tatarize, 2020-09-21:
Unlike previously, I can't replicate that error. I saw another report of something similar, so I'm sure it happened but, that bug isn't something that happens routinely enough. I loaded up 0.6.4 and ran that code and it just started normally. Maybe it classed different. If you try it again, it'll often work. I'm really not sure how that bug works. But, I'd fix it in 0.6.5 and pump out out a new version if I could solve it. (rather than 0.7.0 which is the version I'm currently working on).

tatarize, 2020-09-21:
Hm. Actually, wait I did get a crash on it. Lemme check.

tatarize, 2020-09-21:
Yay, just found it. At least I'm pretty sure. It doesn't correctly revalidate after changing the order for cutting. So you can have move commands that start from the wrong position and during the weird moves I have that means sometimes insisting it started somewhere it did not actually start. Which does a line from the start to the move destination, but the start is wrong so it crashes.

Awesome. I saw this bug elsewhere but had no method of tracking it down without any ability to repeat things.

tatarize, 2020-09-21:
Okay, actually found it. Some the DXF elements were building items without moves to start so the created items were lists of path segments but were not valid path segments. So they would sometimes go Move, Arc, Arc, Arc, Arc, Close, Line, Line, Close. Which messed things up. Since the close originally went to the first item in the new list created by the element, namely the `Line`. But, `Close` commands actually go to the last `Move`. Which is where it properly went. So it started at the first `Line` and ended at the original `Move` which was not contiguous with the entire since the path before that was something else.

tatarize, 2020-09-22:
This is a critical error. And is fixed in 0.6.5 waiting on @joerlane  to approve the branch and publish the version. I also ported in most of the easily portable features.

tatarize, 2020-09-23:
This was corrected with 0.6.5.

closed 2020-09-23 by tatarize