meerk40t/meerk40t #226: CLI/Batch/Console Workflow

tatarize, 2020-09-23:
Testing and adding potential workflows.  The CLI is useful but not extensively tested. And some commands are missing from the CLI . The Batch and Console have some strict limits on their functionality. The CLI is forced to apply some events in a set order that might be done in other orders. A lot of the functionality of the CLI could be put into batch files which should apply the console commands. Or a command should be added do a string of console commands much like a batch file. Likely with some `;` commands between them.

These should be worked on as part of 0.7.0 which will include some optimizations/jobs routines. And to fill in gaps where there are operations that are simply not possible. Or places where oversights are made.

---

* Without wxPython installed.
* Home device.
* Move to 20 20
* Quit when finished.

You can home, and maybe move. But, you can't do both since it hits a busy error and won't let you queue the move correct. And setting quit will quit after the home and prevent the move from happening.

---

* Without wxPython installed.
* Run meerk40t to load an svg file with blue and red lines and have them run at desired speed.
* The speeds should be set separately.
* The cuts should be optimize for cut inner first.
* It should exit when finished.

You can load the file with the CLI, but you cannot set independent speeds. No optimizations are done. With the -za flags it will finish and quit.

---

* Without wxPython installed.
* Home Job before starting
* Create a 1 inch square.
* Create a grid of squares. 6x5 with 1.1in spacing.
* Execute cut on these squares. 
* Shutdown on completion.

You cannot home a job before starting. The loading in cli is locked as before the batch or console commands happen. You can create a 1 inch square and create a grid. This however requires console commands and and the current job cannot be started from console. Shutdown on completion is currently set with `set quit True` and could easily work wrongly.

---

* Without wxPython installed.
* Home Job before starting.
* Create 1 inch square.
* Create grid of squares 6x5 with 1in spacing.
* Optimize for not repeating the cut on an already cut line.
* Execute cut on these squares.
* Shutdown on completion.

Same objections as before. But, there is no way to apply optimizations to a job in either CLI or Console. And there is no optimization to remove extra cut lines. There's no way to start the job, or easy shutdown on completion.

---

* Without wxPython installed.
* Load a png.
* Set the size of the png.
* Apply rasterwizard script to the png.
* Save the png file.
* Shutdown on completion.

This might actually work with a batch file. You can't load the png with the batch file. But, you could load it with a cli and save it with `image save <filename>`.

---

* Without wxPython installed.
* Load a png.
* Set the size of the png.
* Apply rasterwizard script to the png.
* Raster the png.
* Shutdown on completion.

You could load the png, set the size, but you couldn't start the raster from the console. Or easily shutdown on completion. You can set the quit value which would likely do some of that.

tiger12506, 2020-12-27:
Squeaky wheel gets the grease, yes?
I'm eagerly awaiting a raspi build that can do this and #271.

I don't know how to "Start Job" -> "Execute Commands" -> "Start Job" from the console. The -a CLI option is not working for me for attached SVG. It seems to try, but quickly closes program with no visible crash.

I can manually open GUI, click elements, reclassify, and then Start Job process works correctly.

I can skip the GUI _almost_ if I use batch file containing
```
element *
classify
```

but I don't know how to start the job without -a (which doesn't work).
[snowflake.svg.zip](https://github.com/meerk40t/meerk40t/files/5744354/snowflake.svg.zip)

tatarize, 2020-12-27:
The start the job doesn't work from the console. Yet. In 0.7.0 it does. However I haven't worked through all of that code yet. Basically gives full planning steps and this might actually already work in 0.7.0 but there's a lot of criticalish bugs. I think some of the rastering is making basically gibberish code. Here it adds in a few commands which I plotted out as the different states but it's not totally finished.

* plan0 copy
> Copied Operations.
*  plan0 preprocess
*  plan0 validate
*  plan0 blob
*  plan0 preopt
*  plan0 optimize
*  plan0 spool
> Spooled Plan.
*  plan0 clear

I guess on the principle of squeaky wheel, I might actually be able to push something into the stable branch. It really just needs a start command. While it wouldn't do the execution of the commands, or maybe it could just do the default ones. It would be like a CLI command for -a or something. I might hook something up.

tatarize, 2020-12-27:
It seems like -a is bugged. it's classified a default -p path as a raster item.  It might be doing the same for loaded files.

tatarize, 2020-12-28:
Not bugged, seemed to work. Not quite sure. Still needs a CLI command.

tatarize, 2020-12-28:
Make sure you're using the latest version. I think there might have been some CLI issue on an earlier version. Both my attempts to use -a were perfectly successful `-zap M1000,1000h1000v1000h-1000z` worked fin, so did loading a file from the command like with `-za <file>`

tatarize, 2020-12-28:
@tiger12506
> Squeaky wheel gets the grease, yes?

Okay, pursuant to squeaky wheel protocol, I released beta-2 version of 0.6.15

This adds in plan commands with three entries to the console. These should be easy enough to add to 0.7.0 and maintain compatibility. 

`plan start`
Start the current operations feeding them into the spooler.

`plan fulfill`
Start the current operations feeding them into the spooler AND tell Meerk40t to shutdown when we're finished. This should work regardless whether we used the -z flag or not. You can launch the gui, watch it do its stuff and watch it quit out.

`plan default`
Set the operations to the Other/Red/Blue default values.

tiger12506, 2020-12-29:
That doesn't seem to change the situation on my end. Now, the plan start/fulfill suffers from the same problem that -a did for me before. Now that I've tested it a bunch more, I think what is happening is that for automatic operation, it is skipping the raster entirely. This may be due to it not performing the make_image step (?). I base this on the idea that I have to manually click Execute Commands as a separate operation when I do it manually.

Also, there is a new bug where red paths are getting classified as Engrave at 35mm/s (and creating a new operation to support it) rather than Cut at 10mm/s. Yes, I am clearing the operations and restarting each test. This happens for both automatic and manual classification. This doesn't happen on master branch, only for the new beta2 tag.

Is this the best place to discuss? I'd like to help however I can. Suggesting I wait for 0.7.0 release is quite acceptable if you would prefer I stop being squeaky. :)

tatarize, 2020-12-29:
If you want to hash it out in detail irc.freenode.net #meerk40t is sort of a live chat hang-out currently.

I need some folks trying to CLI stuff. I don't know exactly where the missing parts are, so there's clearly going to be some issues that need fixing and I need to know where those would be to make it a really useful tool for people.

---

Yeah, there's some clear chance the rasters aren't processing because they don't automatically apply that stuff. Because they honestly don't. The rasters will filled with shapes and there's no native method or turning that into raster.

There's two methods kicking around. First Inkscape lets you make that conversion to the image with a command line operation. Secondly the MeerK40t code can do it, with wxPython and Pillow installed. But, I don't think it's easily exposed. At least it wasn't easy circa 0.6.x. I think it's perfectly fine in 0.7.0. Which has gone into being *almost* usable. 

The -a flag and others there don't automatically apply those command preprocessing steps within the plan. But, I could probably tweak them to do so. But, I'd think it would be a bit non-trivial since I wrote 0.7.0 in part to make that easier. I'd have to check the stable branch to see if I can just make fulfill apply all those commands too. I don't think it has the job planning and preprocessing except really tightly coupled to the GUI Jobs dialog.

closed 2021-08-12 by tatarize