meerk40t/meerk40t #227: Cutting from the command line

egasz, 2020-09-23:
So this is following up on a Reddit post [Reddit post](https://www.reddit.com/r/ChineseLaserCutters/comments/iv7tb2/k40whisperer_on_cli/).

Just as a summary:
I'm currently using a K40 chinese laser cutter (M2nano board) connected through USB to a Raspberry Pi 3 running Raspbian w/ GUI.
My goal is to get the k40 to cut using just the Terminal.
I have downloaded the pre-compiled MeerK40t pi executable, and also the 0.6.5 source folder.

When I open the executable, this takes a long time to display the MeerK40t GUI frontend which is probably not an issue, as it's a pretty big file, so that's expected.

When I issue the commands:
python3 MeerK40t.py -zav <filename>
or
sudo python3 MeerK40t.py -zav <filename>
or
python3 MeerK40t.py -zavS 10 <filename>
or
sudo python3 MeerK40t.py -zavS 10 <filename>

it moves the head on the XX and YY axis but only slightly, however it doesn't follow the path of the outline in the file, nor does it activate the laser.

I've attached the output of the commands above as well as the .svg file I'm trying to cut, basically it's a 20x20mm square outline.

[logs.zip](https://github.com/meerk40t/meerk40t/files/5269090/logs.zip)

tatarize, 2020-09-23:
If it finishes too soon it might be a bug we're currently working like at this point.

We're pumping out a 0.6.5 patch, so we figured we'd make sure to get your stuff correctly done. 

We're pretty sure the problem is/was that the shutdown killed the controller before it sent the data it had. So it now, joins the thread and adds a quit command to the controller buffer. So it will flush the entire buffer before moving on to the next item.

At least that's the current hope.

tatarize, 2020-09-23:
We're testing if we have it fixed now. A tiny square that size might all get eaten by the premature shutdown. The very latest version on the patch branch might have it. We're testing it now, to make sure we have something that'll work for your stuff before publishing 0.6.5

tatarize, 2020-09-23:
Testing seemed to go well. Especially with the operation in the SVG you won't even need an -S command for that.

egasz, 2020-09-23:
I wouldn't say that it's finishing too soon as it's not starting.
Like I said before, if I open the file on the GUI it cuts everything fine.
What happens is that on the CLI is that it moves the head to specific point in space (this point is relative to the size of the drawing but it's not in the path of any line)

Note: attached a svg file output from CorelDraw
[3colour_test.zip](https://github.com/meerk40t/meerk40t/files/5269382/3colour_test.zip)

tatarize, 2020-09-23:
It's corrected in the main branch. And yeah, it was quitting too soon. Since your file was so fast, it finished nearly before it started. But, if you ran a longer one it would cut off a nice clip of commands at the end. It's why it twitched a little and then died. At the end it shutdown but the shutdown could beat the remaining buffer in the controller send. Now it only ends after the controller send buffer is finished.

egasz, 2020-09-23:
Thought that the issue might be related to CorelDraw so I used Inkscape to create a similar file.
100x100mm page with two square outlines, one black and one red.
Issued the command:
sudo python3 MeerK40t.py -zavS 10 /<path>/test_inkscape.svg 
behaves exactly like with the Corel file

Note: I noticed that the master branch was updated a few minutes ago, so that's the version I'm using.

[test_inkscape.zip](https://github.com/meerk40t/meerk40t/files/5269484/test_inkscape.zip)

egasz, 2020-09-23:
Downloaded the latest version (apparently I somehow didn't download the latest version earlier) and it's working!!!
Thank you so much guys!
Awesome job!!!

tatarize, 2020-09-23:
This was corrected with 0.6.5.

Homing is being dealt with in 0.6.6

closed 2020-09-23 by tatarize