meerk40t/meerk40t #228: Head returning to initial coordinates

egasz, 2020-09-23:
Great job you guys did on the CLI commands. Works exactly as expected, but I have one minor issue.
Like I said I'm using the CLI to cut simple shapes, the thing is that once I issue a command, the head remains where it finishes the job.
Is there a way for the head to go to the initial coordinates before exiting the job?
Quick example:
Let's say that I have my head parked at x20,y20 and I have a 100x100 square cut... if I issue a command to cut it, once the job is finished the head will start the next job at x120,y120 and when finished it will be at x220,y220... if I issue another command the head will crash against the bed edges.
If possible I would like to know how to issue a command (preferably an argument added in the terminal command) that informs Meerk40t to send the head to the coordinates where it was before starting the job.

tatarize, 2020-09-23:
Hm. Yeah, that sounds like a proper issue. I'd guess a -H command to do home would fix that. I mean properly the console thing lets you type `home` but @joerlane was having issues with it too. It was hard to do something like preset a start location. You couldn't really adjust your home location correctly, since it would forget between runs and would be better as a single run.

tatarize, 2020-09-23:
https://github.com/meerk40t/meerk40t/pull/230

This seems to have the right functionality. The code for set was changed and moved a bit to make sure you could set specific adjusted home locations.

```MeerK40t.py -zHas home_adjust_x=60 -s home_adjust_y=530 <filename>```

The -H does the home. It should work stand alone or in it's own stuff. The -s is set, which now takes exactly one pair, and they must be connected with an = sign. That'll set the adjusted hom position and start the file from that new location. So home, move and start as if that new home spot is 0,0 within the file.

tatarize, 2020-09-23:
Sadly added a crash bug on Actualize Pixels so 0.6.6 timeline gets pushed up. So might as well make sure this tweak works for you, or if you figure out something else you'd need to get CLI working for your uses. It'll probably be sooner rather than later.

tatarize, 2020-09-24:
Hm. That would likely be the easiest method of implementing that. In the GUI section I did recently add exactly that. `After - Origin` basically return to 0,0 or where MeerK40t thinks 0,0 should be. It becomes a little bit mission critical when you consider rotary jobs. You can't home the device with a rotary attached, it has the y spin a cylinder that doesn't stop spinning or hits the endstop. So the absolute best thing you can do is return back to 0,0. This also was an issue in the Visicut driver that I also wrote, namely it didn't have jog control so I would tell it to home, but if you adjusted the home location you were basically resetting that adjustment which would cause some obvious issues. 

It turns out that returning to where you started is a much more versatile command than homing. Because home is sometimes impossible such as in a rotary and sometimes loses manual positioning of the laser head. Or requires some odd jigging. 

It seems like it's certainly a feature request. But, it absolutely fits the standard need I have towards making the CLI more functional. And though I'm pushing some items until I get 0.7.0 sorted, it's not a hard addition here. Even if there's a chance it might not reappear in 0.7.0.

tatarize, 2020-09-24:
I added -O to the CLI for 0.6.6. My mock tests seemed to work.

egasz, 2020-09-25:
Great!!! the -H option works fine homing on every command, but if I manually move the head to another initial position then all goes to hell :) Can't wait for the 0.6.6 release with the -O option (I'm assuming this option takes the head back to the initial position right?)

joerlane, 2020-09-25:
Tested the -O option and it works as intended. It is also useful in conjunction with the -H and -s commands. 

I currently use this command on my machine since I have a stock smoke vent and a honeycomb bed. I have to move 60 mil in the X to hit the left corner of my honeycomb, and then 530 in the Y to clear the smoke vent and find the top of my honeycomb. That allows me to align objects to the upper right of my bed, and the head returns to that position when finished.
python3 MeerK40t.py -zaHOs home_adjust_x=60 -s home_adjust_y=530 <filename>

Alternatively; you can actually use negative numbers in the -s commands to set X and Y home adjustments. When the machine hardware homes, it moves just enough to un-trigger the end-stops after finding them. The home_adjust commands can then be used to put the head back into that extreme upper left corner. Just be careful not to go too far or you'll skip steps and lose consistency and likely crash the head after a few runs. Very conservatively; this works just fine on my machine to expand the usable area of my bed:
python3 MeerK40t.py -zaHOs home_adjust_x=-10 -s home_adjust_y=-10 <filename>

tatarize, 2020-09-25:
Down to what seems like final check of 0.6.6 before publishing.  Might still be a bit but no major changes so testing should be more reasonable. Unlike the testing 0.7.0 is going to need.

closed 2020-09-30 by tatarize