meerk40t/meerk40t #232: Version .6.6 unable to download on Windows 10 - Virus detected

WildMany, 2020-09-27:
Just wanted to try new version, but after trying to download .exe file in Google Chrome - it says "Failed - virus detected".
Will try .zip version, but just to notify you that something is wrong there.
milan
---

WildMany, 2020-09-27:
Now tried to download it by using Mozila, and after running .exe file got system message: "Opreation did not complete successfully because the file contains a virus or potentially unwanted software"

tatarize, 2020-09-27:
Hrmf. I specifically recompile the boot loaders to make sure that that stuff doesn't happen. I might have to check what virus detection that stuff uses because for the most part I check. I upload the file to virustotal.com and make sure no false positive triggers an issue.  I make sure Microsoft, AVG, and all the major ones don't get this stuff.

tatarize, 2020-09-27:
I tried locally and I don't think I turned anything off but it downloaded and ran correctly, with my latest version of Chrome.

Initial research says that Chrome and Firefox use Microsoft's local virus scan and I passed that one. Maybe an older version of the malware detection tool is throwing an issue.

I make sure to check and that only weirdo never heard of engines get false positives. If not I recompile the bootloaders some more. Here's the signature for 0.6.6 microsoft is a hard pass for me.

https://www.virustotal.com/gui/file/df9bb986bf11832c73b2155bbb219c8e18394735941fb05d8842defa0a16084c/detection

tatarize, 2020-09-27:
I'll try updating the Windows Malicious Software Removal Tool 5.83 and Security Intelligence Update for Windows Defender Antivirus. There was a point a few days ago when I ran a file through, and microsoft was missing and another time they had like the only false positive of the major virus detection players.

tatarize, 2020-09-27:
I still need something to give me a pass/fail on this false positive so I can try different bootloaders until something compiles differently that it doesn't trigger this. And for this I can't seem to do it. Can you (without losing the information of what version of windows defender you have) see if there's an update of windows defender that maybe doesn't do this.

I'm sure people have some older versions here and there, and maybe virustotal is using the latest and it only false positives in an older version. Though I guess that's just a much harder fix that'll still require some recompiles.

tatarize, 2020-09-27:
https://www.virustotal.com/gui/file/4fdea4647456f612ace6851eb8b03ea10da9418400236f6074ab0a7be6ea64b3/detection

I was compiling the 64 bit bootloaders and packaging with the 32 bit bootloaders. So I wasn't getting the proper false-positive protection. Now has zero.

closed 2020-10-01 by tatarize