meerk40t/meerk40t #233: Paths Importing Empty

kwilliams147, 2020-09-28:
When importing an SVG that is black fill v0.6.6 imports with no fill color. After doing some checking when an all black image is exported the SVG data has no color tags and should default to black. I tested this with v0.6.5 and it imports as expected with black fill. I'm not very good with python but I did notice "[commit bbc7c2ad879c1ca7a7a0e1628d0f2b57c7788af7](https://github.com/meerk40t/meerk40t/commit/bbc7c2ad879c1ca7a7a0e1628d0f2b57c7788af7)" refers to changing the default fill color to none from the default of black.

This is the SVG I tested both versions with.

`<svg version="1.1" id="Layer_1" xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" x="0px" y="0px"
	 width="42.52px" height="42.52px" viewBox="0 0 42.52 42.52" enable-background="new 0 0 42.52 42.52" xml:space="preserve">
<rect x="7.087" y="7.087" width="28.346" height="28.346"/>
</svg>`

tatarize, 2020-09-28:
Yeah, part of the issue is that when I added the default colors in there. Images were getting a black stroke and classifying wrongly. I seem to recall there was something about it in the SVG/CSS specs but I couldn't find it. When you do this with something like inkscape it gives you a color, but says the `stroke` and `fill` is unset.  I went back and forth with regard to that default a few times. And I'm inclined to side with squeaky wheel (in that you're complaining so changing is preferred). But, I'm not exactly sure if that fits the spec correctly.

Yes, it changed. But, no. That svg does not have a black fill.  I'll look a bit more at the spec to decide on this, but that's not really an svg with a fill at all. And `none` might actually be the correct.

tatarize, 2020-09-28:
Ah, SVG 13.4.1 does have an initial for fill as `black` and 13.5.1 has initial for stroke as `none`. I'll correct this to fit with the standard. While the fill merely existing matters for some objects, the svgelements code is a bit flawed as it sets everything to currentColor by default anyway which is also wrong.

kwilliams147, 2020-09-29:
Thanks for the info. I agree that the SVG code has no color data but looking at your findings it makes since... Fill defaults to black if it is not specified in the SVG itself. Also note I tried exporting that block in Illustrator and Inkscape both with the same results. Thanks for the speedy fix and looking for 0.6.7 when it drops for now I am using 0.6.5.

Thanks

closed 2020-09-29 by kwilliams147

tatarize, 2020-09-29:
@kwilliams147 This is fixed in 0.6.7 which is now live.