meerk40t/meerk40t #237: Linux Release Build Fail

kwilliams147, 2020-10-05:
Looks like the linux releases for v0.6.7 didn't build correctly. Its only the source code instead of a built python file like v0.6.6

joerlane, 2020-10-05:
I'm not sure what you mean by "releases" (plural). There is only one Linux release for 0.6.7. I just unpacked it; it's a single file executable, and it runs for me. The source for MeerK40t is also considerably smaller than 135 Meg. How big is the file you downloaded? What did you find inside when you unpacked it?

There is a change to the Pi release that makes it launch 10x faster. Since the onefile binaries are compressed; it takes the raspberry pi a significant amount of time to unpack those contents into a temp directory on every launch. As such; a Pi_slow version was released for people who really enjoy waiting 60-70 seconds for the program to launch. The new Pi release does indeed unpack inside of a folder full of support files. Those files would otherwise have been compressed inside the onefile binary (which was very slow). It can still be launched the same way though; by executing the MeerK40t binary that is found inside the folder. In that case; it needs to remain inside the folder full of support files; but it will launch in just a few seconds on a Pi3.

kwilliams147, 2020-10-05:
Ok I was unaware of that and it was not in the release notes so I was comparing the two Pi releases from v0.6.6 to v0.6.7.

joerlane, 2020-10-05:
The difference there is that a Pi_fast binary build was available in 0.6.6, and the default Pi build was still the slow onefile package. In version 0.6.7 the old-style onefile build is now rightfully named Pi_slow instead. A user had complained about the abysmal speed when launching MeerK40t on the Pi, and this was the only way to resolve that (outside of running pure python and installing the requirements). 

It didn't make the release notes because it's not technically a code change. Both versions are built with pyinstaller; both with and without the --onefile option. Certainly wondered about the reaction to this change; but figured that the roughly 10x speed improvement would win most users over very quickly.

joerlane, 2020-10-05:
Release notes for version 0.6.7 modified to include a note about the change to default Raspberry Pi build.

closed 2021-01-30 by tatarize