meerk40t/meerk40t #242: MeerK40t crash log. Version: 0.6.7

tomislav17, 2020-10-10:
MeerK40t crash log. Version: 0.6.7
Please report to: https://github.com/meerk40t/meerk40t/issues

Traceback (most recent call last):
  File "MeerK40t.py", line 242, in <module>
  File "Kernel.py", line 2280, in open
  File "wxMeerK40t.py", line 185, in __init__
  File "wx\lib\embeddedimage.py", line 49, in GetBitmap
  File "wx\lib\embeddedimage.py", line 64, in GetImage
wx._core.wxAssertionError: C++ assertion "strcmp(setlocale(0, 0), "C") == 0" failed at ..\..\src\common\intl.cpp(1699) in wxLocale::GetInfo(): You probably called setlocale() directly instead of using wxLocale and now there is a mismatch between C/C++ and Windows locale.
Things are going to break, please only change locale by creating wxLocale objects to avoid this!

tatarize, 2020-10-10:
Hm. I had somebody else report this. I tried a few different things to fix it and failed. I tried getting rid of the languages and nuking the translation routines. I might have misunderstood the bug somehow.

`home = RB.RibbonPage(self._ribbon, wx.ID_ANY, _("Home"), icons8_opened_folder_50.GetBitmap(), )
`

I thought it was first instance using _(""") to translate which made a lot of sense. Because I used that there. Though apparently it's in getbitmap, getimage.

tatarize, 2020-10-10:
https://github.com/wxWidgets/Phoenix/issues/1804

I think it's a wxWidgets error.

tatarize, 2020-10-10:
https://wxpython.org/Phoenix/docs/html/internationalization.html

The code I use in: https://github.com/meerk40t/meerk40t/blob/55c70ad2478e880b1b9d7eafea9159607b3834e3/wxMeerK40t.py#L2972

Is the same as you find in the documentation:

```python
    def update_language(self, lang):
        """
        Update language to the requested language.
        """
        device = self.device
        try:
            language_code, language_name, language_index = supported_languages[lang]
        except (IndexError, ValueError):
            return
        device.language = lang

        if self.locale:
            assert sys.getrefcount(self.locale) <= 2
            del self.locale
        self.locale = wx.Locale(language_index)
        if self.locale.IsOk():
            self.locale.AddCatalog('meerk40t')
        else:
            self.locale = None
        device.signal('language', (lang, language_code, language_name, language_index))
```

vs.

```
    def updateLanguage(self, lang):
        """
        Update the language to the requested one.

        Make *sure* any existing locale is deleted before the new
        one is created.  The old C++ object needs to be deleted
        before the new one is created, and if we just assign a new
        instance to the old Python variable, the old C++ locale will
        not be destroyed soon enough, likely causing a crash.

        :param string `lang`: one of the supported language codes

        """
        # if an unsupported language is requested default to English
        if lang in appC.supLang:
            selLang = appC.supLang[lang]
        else:
            selLang = wx.LANGUAGE_ENGLISH

        if self.locale:
            assert sys.getrefcount(self.locale) <= 2
            del self.locale

        # create a locale object for this language
        self.locale = wx.Locale(selLang)
        if self.locale.IsOk():
            self.locale.AddCatalog(appC.langDomain)
        else:
            self.locale = None
```

While mine is a bit changed I hit all the main points still. It seems like if I messed that one thing up everything would crash. But, these crashes are really rare.

tatarize, 2020-10-10:
But, I nuked all the translation stuff and that bit still crashed.
https://drive.google.com/file/d/1qCRJy9_kt8Ovdr6qWGkYdwSiBTPSjkCz/view?usp=sharing

I wish I could duplicate this on my own machine. It would make tracking this down so much easier. Or at least finding a work around.

tomislav17, 2020-10-10:
Problem and solution found. the problem was in the language before I had Croatian and the program worked, yesterday I switched to English and the program no longer works. I returned to Croatian and now it works

tatarize, 2020-10-10:
That's a solid workaround, but the program *should* still work regardless what language you put it in. And it crashes for other people. I'll still need to get it to work even if the windows language is set to English, with windows installed in Croatian.

MitjaHenner, 2020-10-19:
Getting this crash on every start, Im on an en_GB Windows install.

tatarize, 2020-10-20:
@MitjaHenner 
Can you try the 0.6.8 beta exe. I tried to move back to Python 3.7 and wxPython 4.0.7 to see if it was stable. I had another person with it who never got back to me yet.

https://github.com/meerk40t/meerk40t/releases/tag/0.6.8-beta1

These were recommended to me at the wxPython stuff and haven't had that attempt validated or invalidated yet.

MitjaHenner, 2020-10-20:
Thanks, that version works.

tatarize, 2020-10-20:
@MitjaHenner Thank you. Good to have confirmation that that compile issue is fixable with more stable versions.

tatarize, 2020-10-26:
This seems to work if compiled in more stable versions of python and wxPython. I'll likely stick a bit more to wxPython 4.0.7 and Python 3.7 for a while longer until I can avoid this with confidence.

tatarize, 2020-12-06:
Reopening issue, for non-compiled version users.

This error is in wxPython which is failing in these cases if you suffer from this problem with a python version, revert your version of wxPython.

`pip install wxPython==4.0.7`

I tried and tried. I couldn't get it to work with the people who had it failing, it just wouldn't. The bug appears to be in wxPython and I was told 4.1.x is generally less stable and they have even and odd versionsing for the stable and less stable versions. And downgrading corrected the issue. And I haven't needed anything from 4.1.x so I removed that from the build and used the lower version to avoid that error. I can't repeat the error myself so working around it within the 4.1.x branch proves nearly impossible.

MitjaHenner, 2021-01-25:
And its back in the latest release.

tatarize, 2021-01-26:
*sigh* okay. I was hoping it would work and the latest could be used I'll re-correct it and load a secondary exe with other wxPython.

tatarize, 2021-01-26:
@MitjaHenner Okay, released one with wx 4.0 in the latest. Though I'll likely just release any other ones with just the wxPython 4.0 release. I was only hoping 4.1.1 fixed it.

tatarize, 2021-01-26:
https://github.com/meerk40t/meerk40t/releases/tag/0.6.17

MitjaHenner, 2021-01-26:
Sadly the same error:
MeerK40t crash log. Version: 0.6.17
Please report to: https://github.com/meerk40t/meerk40t/issues

Traceback (most recent call last):
  File "MeerK40t.py", line 257, in <module>
  File "Kernel.py", line 2164, in open
  File "wxMeerK40t.py", line 185, in __init__
  File "wx\lib\embeddedimage.py", line 49, in GetBitmap
  File "wx\lib\embeddedimage.py", line 64, in GetImage
wx._core.wxAssertionError: C++ assertion "strcmp(setlocale(LC_ALL, NULL), "C") == 0" failed at ..\..\src\common\intl.cpp(1579) in wxLocale::GetInfo(): You probably called setlocale() directly instead of using wxLocale and now there is a mismatch between C/C++ and Windows locale.
Things are going to break, please only change locale by creating wxLocale objects to avoid this!

tatarize, 2021-01-26:
Ah, I recall I should also use the more stable python version... lemme make another.

tatarize, 2021-01-26:
Okay. I updated again. Should say Rebuild2 in the download description.

Python 3.7 and wxPython 4.0.7-post2

tatarize, 2021-09-26:
https://github.com/wxWidgets/Phoenix/commit/040c59fd991cd08174b5acee7de9418c23c9de33

This commit may fix the bug. And a 4.1.2 snapshot may solve things.

tigert, 2022-01-25:
You probably dont want to hear this, but 0.7.5-beta3 and the one before that also crash with the same error on Windows 10 set into English in the GUI, but I guess this is Finnish version.

Meerk40t runs fine if I set my desktop language to Finnish.

tatarize, 2022-01-25:
They all do. You need to run wx40 version to make it not do that. I could never track it down, but with the compatibility code stuff it can be made to work but that's why there's two versions there.

tatarize, 2022-01-25:
I've been pretty sure 4.1.2 fixes it but no release there has been forthcoming, and I might need to try the daily to test whether it gets rid of that stuff.

Sophist-UK, 2022-03-28:
See https://discuss.wxpython.org/t/wxpython4-1-1-python3-8-locale-wxassertionerror/35168 for a possible workaround.

tatarize, 2022-05-30:
People have been using the checked code for a few months now and there are no instances of this crash. I'm going to consider it corrected and discontinue the wx40 version.