meerk40t/meerk40t #244: Speed when not cutting

Karijn, 2020-10-16:
The speed is set per color now, when moving to a new position without cutting or engraving, the speed should be faster

tatarize, 2020-10-19:
See #191 

I am getting this one a lot and might just put it in the minor development version. Though I think it would bit of actual research at the machine to get the rapid moves correct. I was hoping this could wait until 0.7.0 but I'm thinking people notice this really clearly. And are annoyed with it. When really things like travel reduction optimization will have a bigger impact. I tried adding some cheap work around into 0.6.8 but it apparently doesn't work. I guess I'll duplicate the option I made for 0.7.0 and try hacking it in there. Even though a change to the code production will require some more rigorous testing than I usually give minor versions.

tatarize, 2020-10-20:
I've moved this up. 0.6.8 will have this set as a flag. Currently undergoing testing.

tatarize, 2020-10-24:
This is fixed in 0.6.8, now live.

closed 2020-10-24 by tatarize