meerk40t/meerk40t #245: failed to execute script pyi_rth_usb

stevebottrell, 2020-10-18:
Shows error box on double clicking Meerk40t.exe and displays this error and quits

failed to execute script pyi_rth_usb

Windows 7 32bit

tatarize, 2020-10-19:
Hm. I run Windows 7 64 bit. It likely is a compile error there with compiling MeerK40t in 64bit.

Do the older version work?
Does it work to run it from python?

It's quite likely a compile issue. I didn't much care about 32 bit OSes. Though I did compile the bootloaders for 32 bit. I guess I'll have to install some 32 bit windows to check that. Though maybe I could figure out a good emulation.

If you run it in pure python it should work. Install python run the `install_run.bat` batch file as a work around.

stevebottrell, 2020-10-20:
Latest Compiled exe version to work on windows 7 32bit is 0.6.4

Running meerk40t.py gives this error :

 File "meerk40t.py", line 219
   device.add_watcher('console', print)
                                                         ^

my version of python is 2.7.18 if that makes any difference.

for me its not a problem as i can run the latest version on windows 10 64bit on my laptop.

tatarize, 2020-10-21:
Yeah, use Python 3. Python 2 was dropped at end of life. January 1st of this year.

closed 2021-01-30 by tatarize