meerk40t/meerk40t #248: Failed Install on Mac

stan916, 2020-10-19:
I have installed Meerk40t on my Mac (running Catalina 10.15.6) using the latest version 0.6.7 and the program will not run correctly. I can't open any files and the bed grid is missing from the display. Also the grey area around the bid seems smaller than on my Windows install.

[image]

When I try to open a file nothing happens but I can't then close down Meerk40t as it just hangs so I have to force quit. All the dialogue boxes for settings, controller, etc seem to work ok. I have not tried to connect it to my K40 yet.

joerlane, 2020-10-19:
Could be helpful to see if anything changes on the screen when trying to load an SVG file, or an image. Also; I'm somewhat concerned about the double title bar on the main MeerK40t window. Never seen that before, and it may be indicative of something else messing with the UI in a strange way. Does the white box zoom in an out, draw selection boxes, or scroll around; or is it frozen?

Working to get a similar version of OS X installed and configured to test this myself. In the short term, you could try to create a new user on your computer; and run MeerK40t from that new account. Might be surprised...

stan916, 2020-10-19:
Nothing happens when trying to load a file. The left hand column of
operations, etc, is selectable and the dialogues associated all work
correctly. The white box also does nothing.

Will give it a go with a new user.

stan916, 2020-10-19:
UPDATE! I have just installed 0.6.5 and this seems to work! I'm doing this
remotely to my Mac at home from work so I'm a bit limited with what I want
to do, but so far a file has opened correctly at least and the grid is
there, etc, etc, so it looks good so far.

tatarize, 2020-10-19:
That means whatever the issue is, it's actually going to somewhat minor.

stan916, 2020-10-19:
I can now open a file but there is something weird going on with the naming
of the engrave paths, see attached screenshot.

joerlane, 2020-10-19:
Ok. That's good news. I don't think I've made any changes at all on the Mac build system since MeerK40t version 0.6.5.
There aren't a lot of major changes to MeerK40t since 0.6.5 either; so it should work alright for you in the short term. I'll certainly be looking in to this; interested to see what 0.6.7 does with a clean install of OS X 10.15.6. I can always compile on that OS if it's absolutely necessary.

joerlane, 2020-10-19:
> I can now open a file but there is something weird going on with the naming of the engrave paths, see attached screenshot.

Screenshot didn't attach for some reason. The path descriptions had some long names in older versions (like 0.6.5) describing the SVG path. There should be a tiny icon to the left showing it if it's not terribly large. You can also click on the path in the tree, and arrow up and down to see which ones they are in the scene.

stan916, 2020-10-19:
Thx. I’ll try 0.6.6 as well just out of curiosity.

stan916, 2020-10-19:
0.6.6 no good, just a blank screen where the bed and grid should be,
screenshot Capture2. Sorry forgot to attach the previous one for 0.6.5, now
attached also

tatarize, 2020-10-19:
I don't see a screenshot. Might not attach from an email response or something.

stan916, 2020-10-19:
[image]
[image]

stan916, 2020-10-19:
I was using gmail to reply, just attached them on github.

stan916, 2020-10-20:
Further update. The 0.6.5 install on the Mac runs fine except for the element naming shown in the screenshot. However there is another issue which affects BOTH Windows and MacOS. No matter what buffer size limiting figure is used the laser pauses briefly for a couple of seconds every now and then (maybe every 45 secs - 1 minute) during a raster if the PPI is set at anything less than 1000 (i.e. full power). 
I can't be 100% certain but it looks like the rastering intensity between each pause changes slightly as there are visible changes in the rastering on the job which seem to coincide with when the pauses occur, but I need to do some more tests on that tonight to be 100% sure.

tatarize, 2020-10-20:
The element naming was one of the minor things changed. People insisting that the gibberishy names weren't helpful so they were changed.

tatarize, 2020-10-20:
@stan916 will have information on stuttering and why it happens.
https://www.youtube.com/watch?v=blhbbq22sGU

Since this is a issues forum there's a couple things being tried to get around this. In the plot planning I'm adding an option to do grouping where the actual pulse train will be moved by 1 pixel here and there to make sure it doesn't stutter. As well as some dot length patterning which may certainly help. The video explains why it happens, and I can likely do some stuff like remove a double check for OK issue that may actually crank out a few extra ms timewise.

joerlane, 2020-10-21:
Initial issue was verified to be a problem on another 10.15.6 Mac. On launch it shows a light grey background in the large pane of the main window. Resizing the window draws the dark grey background and main drawing scene in white. The rulers and reticle never show up, even after "refresh" command in the MeerK40t Terminal.

Loading a file does not update the tree or main window. Unsure if starting a job will queue anything after loading the file. Or issuing the refresh command again and again.

After loading a file; clicking the Start Job button would bring up the Jobs window, and it would be known if the file loaded if those components were in the queue.

joerlane, 2020-10-24:
The recently uploaded 0.6.8 release is built using a different version of wxPython. This needs to be tested on OS X 10.15.6 and similar.

tatarize, 2020-10-26:
Might have fixed some internal -psn_?_???? crash thing. @stan916 can you confirm 0.6.9 works for you?

joerlane, 2020-10-28:
The MeerK40t 0.6.9 updates has been confirmed to work on a Mac running OS X 10.5.7. 

There is some reason to suspect that the previously non-functional versions (0.6.6-0.6.8) may function if launched from the command line as well as either a file to open, or the -v switch. Some testing indicated that process may actual fix the app so that double clicking works afterward. Not terribly important information now that MeerK40t 0.6.9 is available...

joerlane, 2020-11-02:
MeerK40t 0.6.10 was released. Still awaiting user verification to close this issue. Versions 0.6.10, and 0.6.9 should both work in Mac OS X 10.15.x.

stan916, 2020-12-16:
All fixed on 0.6.10

closed 2020-12-16 by stan916