meerk40t/meerk40t #249: With Rotary On, Location Indicator is Incorrect

kshenneman, 2020-10-21:
When the rotary is turned on, and the Y scale is set, the red circle location indicator moves at the Y scale too. Therefore it does not provide an accurate indication of where the job actually is. So for instance, if the rotary Y scale is 2.0, the location indicator will also move that distance. Therefore, if the image you are engraving is 15mm high, by the time the engraving is complete, the indicator will be 30mm from the top of the image, instead of 15.

tatarize, 2020-10-22:
Yeah, the reason for this is the circle isn't scaled. It's at the x,y coords that it knows about, but when a rotary job is fed to the spooler it stretches it to the right size for job. But, this stretched job isn't something meerk40t sees. I suppose if you wanted you could stretch the job yourself by hand and that would make them sync up.

To do that you can press F9 and type: `scale(1,2,$x,$y)` in your example. This will apply a scale relative to your current laserhead position, stretching the y by a factor of 2. Consequently when you job (with rotary turned off, you did the stretch manually) it would track the correct position. Since you preprocessed your job for the rotary.

tatarize, 2020-10-22:
There's another issue here with rotary of a raster-wizard image. If you want to do an image with a rotary, you'll hit a problem when the stretch, which is applied last during the preprocessing stage of the job planning there forces the image to be resampled which is absolutely not what you want.  But, if you pre-rotary stretch it, you'd be able to apply the raster wizard to the stretched image without issue.

tatarize, 2020-10-22:
There does need to be something done with this, it terms of scene and functionality. It also seems like since the stretched image will be taller that the guides and grids are functionally wrong in this view.  Since you're looking at interpreted space at that point. Now I could stretch the view itself, by the inverse of the matrix. The idea here would be that anything loaded would get stretched and the view itself would get smushed, so they would look like flat objects, but they'd be stretched objects. The view stretching wouldn't be any harder than applying a zoom to the scene. And you'd likely be able to toggle it on and off. It would just be a couple lines of code. It's clear that the 11th hour stretch at the last second has some issues, and there'd be a lot of features that would be useful for a rotary, especially things like stretching the objects in the scene and the view. Though I don't have a rotary myself, so it's a bit guess and check as to what people would want.  I've had some high fallooting ideas like making the view itself topologically donut shaped with the top of the view able to flip back to the bottom so you'd see the flattened outer shape of the object being used with an infinite surface stretching along the rotary spin. But, it's not quite clear what would be that useful here and what wouldn't.

As is everything is done with a 1:1 field, and then the preprocessing applies the rotary stretch at the last second. So the view on screen doesn't match the thing you're actually rotarying. Since the project itself got preprocessed and that's why it doesn't match. The best way to fix this would be to process it in the scene itself before hand and give you tools to mess that in useful ways. 

This requires me understanding what useful ways might be without actually having a rotary myself.

tatarize, 2020-10-22:
I added in RotaryView and RotaryScale, they should exist next version. Currently I'm in the middle of solving jogging which isn't fully mapped out, and I don't fully understand the routine apparently. But, next version should have couple options for rotary that aren't use the readymade rotary stuff.

They will be stored in console under rotaryview and rotaryscale but will use `control+f3` and `alt+f3` as hotkeys. Though you might need to hit `control+alt+shift+escape` to reset your binds and get those hotkeys. And I don't love the setting Rotary scale value and then turning the rotary off, UI.

tatarize, 2020-10-24:
RotaryView and RotaryScale are live. 

https://youtu.be/L6pRTvsdKvA

The end bit goes over them. Hopefully they are kinda working. It's worth a try. If that's more your liking for Rotary, I can go down that path as the way rotary works. I just don't own one and it's not really clear what people would want in this regard. Clearly there's some issues with applying the stretch as a post-processing operation.

closed 2022-07-20 by tatarize