meerk40t/meerk40t #252: WiFi Camera Support

kshenneman, 2020-10-27:
Previously, you had WiFi camera support in the application, but it would crash the application. Is this still in the works?

joerlane, 2020-10-27:
What operating system was causing a crash in the application when using a WiFi camera? What kind of crash was it; did the application fail, or did the video stop updating? I don't see any issues open in regards to that problem. Haven't personally had any crashes on Mac, or Linux with the IP camera support which I use regularly. I just tested the latest version of MeerK40t on Windows 10 and had no issues connecting to an IP camera I use often. From my recollection; I believe opencv was updated to resolve that issue.

Having tested a few different IP cameras; I found that using alternative URLs may help with cameras that connect, and display video only briefly. In once case, I have a camera that will only work for a few moments with the RTSP URL, but works normally if the HTTP camera URL is chosen instead. This may in fact be an issue with the video format provided by each (but I've been unable to verify that); in that particular case the HTTP URL (the one that works) outputs an MJPEG stream.

tatarize, 2020-10-27:
The Wifi/Ip camera support compiles and works again. It had issues in the compiled version for windows. But, it works fine in the latest version. I tested it and yeah, I can get ip webcams that work. Updating OpenCV did the trick.

kshenneman, 2020-10-27:
Running on Windows 10. I was previously running version 0.6.4. Had been working with Dave. I was able to get a little further with the Python code version than the compiled version. The camera I am using does not have http access, only rtsp. With rtsp, it will only run for a few seconds (like 10) before it dies and says "No Webcam Found". I downloaded 0.6.9. When you add the camera, it defaults to 4 frames a second. It's not dropping frames so as I'm seeing the time displayed on my camera, the seconds keep clicking off slower and slower. Because it's not dropping frames, i think it eventually runs out of memory or buffer, and kills the connection.

tatarize, 2020-10-27:
Yeah. I had done some of this in an initial upgrade in 0.6.4 but reverted to 0.6.3 and it's still been stuck there. There's some threading that helps and I had some aspects working. Mostly the camera would lag behind and build up a buffer. It turns out that it doesn't drop frames if you slow it down. It grabs each frame in order which caused it to fill up a buffer and crash. I made some code changes to keep it grabbing frames constantly and just deliver the last one that was loaded when queried. 

But, I wanted good multi-camera support and some of the code changes I made in 0.7.0 were great for that. The API changes I made had some clear advantages there, where rather than bound to some particular device I could just have many cameras with their own settings without needing to define some specific device for those settings to exist. The kernel would simply allow me to derive some sub parameters. Then I could launch cameras independent of devices because the kernel is loaded once and different contexts provide different aspects of the kernel. It was one of the reasons I made the changes in 0.7.0 that I did.

However 0.7.0 is biting off more than I can chew. And while I do like the cut planning upgrades and the kernel upgrades. I might have to scale some of the stuff back and space it out for 0.7.+, perhaps a more intermediate jobs window that doesn't expect to do full previews. The camera upgrade is third in the list on 0.7.0 todo list: #214. I got slogged down a lot of 0.6.5 - 0.6.10 changes. I did get through most of cut planning console commands. Then I need to do the sort of blob/cutcode upgrade stuff and get everything back working. Revert the project foresight changes for a bit. 

The camera upgrade stuff wasn't lost, but it was made kinda contingent on 0.7.0 since if you want to run 3 different cameras with different settings, it sort of does need the ability to let them each have their own settings. And that was something I was moving along with the camera upgrade. The problem largely inspired the 0.7.0 upgrades as I tweaked things to use contexts rather than devices. I'd have to revert project foresight. (https://github.com/meerk40t/meerk40t/milestone/8), finish the cutcode upgrade. Maybe gloss over some of the great improvements cutcode will bring, even though they are low hanging fruit. And finish up the camera upgrade. And port in changes from 0.6.7+ into 0.7.0 which is itself built off the first attempt at 0.6.5 which was intended to finish up the camera stuff but then changed the core API (hence the 0.6.5 branch being for 0.7.0). Most of the core API stuff is done. It's getting some secondary changes functional and the new cutcode stuff.

Paring down the version to core elements and finishing up the needed stuff is the current game plan. This should also include the camera upgrade depending on how much work it had left to be done. I think I had much of the core crashing problem solved with some threading, and was setting up a sort of URI set of sources you could choose from. So it wasn't that far off.

joerlane, 2020-10-28:
If the camera in question supports a different method of video encoding over the RTSP protocol; it is worth trying. It may be a setting in the camera to change the video encoding (typically along with video resolution, etc.), or separate URL to support it. 

I have a different camera here that works fine on RTSP; so it's not specifically an issue with that RTSP vs. HTTP.

tatarize, 2021-01-21:
This is effectively done in 0.7.0 thought it's still pre-beta. The core code is functional for these purposes.

tatarize, 2021-01-31:
@kshenneman There's a couple buggy betas of 0.7.0 you could try this feature in. The Multi-camera stuff works fine but also the camera cycles correctly and lets you reconnect as needed.

closed 2021-08-12 by tatarize