meerk40t/meerk40t #253: X/Y Scaling for Non-Rotary.

tatarize, 2020-11-01:
> ""X/Y scale factor" as k40 Whisperer have? The machine that I own have a different pulley/teeths count so I used to change it on Whisperer in order to get the correct size of the engraved picture."
> 
> "it's good to have a calibration feature like that. Some k40 after some time get the belts get stretched so need to used it."

I did not know this was a thing. Apparently not just rotary but apparently there's some need to have normal topological x-y scaling effects.  This should work with #249 to allow Rotary and scaling or scaling simply without the rotary. The rotaryview/rotaryscale functions seem like the best way forward for this. Or something more comprehensive that also corrects rotary and non-linear belt issues.

tatarize, 2020-11-01:
This could probably best be done by having different matrices. Access give to the view matrix. manipulations so that all x and y scale can be sent. And allow a scene matrix that is used to apply to any paths adding to the scene as a pre-processing thing.

So if you have your view matrix the inverse of the pre-process matrix an image dragged onto the scene will end up wider or more narrow, but the view matrix can either counteract that or allow it to be shown in it a newer kind of fatter view. This would allow for a variety of pre-processing affine transformations and view manipulations. These matrices could also be available to the console in 0.7.0. This would basically emulate how rotaryview works but then make rotaryscale a single time matrix value as new objects are added to the scene.

tatarize, 2020-11-01:
Though if done at insertion to the scene, it would actually make the objects fatter on load, so if you rotated them they would be tweaked wrong.

makegeneve, 2020-12-04:
Just discovered this software.  Looks great!  Have to say that this feature (scaling in X&Y) is a go/nogo for our fablab.  K40 Whisperer saved us when our machine decided it was no longer in calibration after changing pulleys.  But this software looks even better.

makegeneve, 2020-12-07:
Should just add, happy to help test this.

tatarize, 2020-12-08:
In the recent versions of 0.6.x you can try control+F3 and alt+F3. You need to set the scale_x in rotary. Though go ahead and turn the rotary off. The feature just uses the rotary value there. This should trigger rotaryview and rotaryscale which will kind of adjust things to correct them for your weird size and correct your view to see them the same.

https://youtu.be/viCVItFcXi4?t=388

I think I went over it there. There might be some other UI changes that would make this better or more official or reasonable, if you have them post them here. There's some efforts to make this rotary stuff more built in.

tatarize, 2021-02-11:
Keeping #281 as this issue's home.

closed 2021-02-11 by tatarize