meerk40t/meerk40t #255: Mac OS X Specific Issues

joerlane, 2020-11-02:
Text in About box does not word-wrap.

The cursor changes to interact with objects in the main view don't show up. On rare occasions they will blink, but the default cursor takes over and makes it very difficult to estimate which action will happen when clicking and dragging. 

The grid pattern is a very thin line, and only visible when zoomed to about 1mm per ruler division. It is also a very light grey and still rather difficult to see. This "seems" like a stroke weight type issue, where it's too thin to even draw. It doesn't look anything like this in the Windows screens I've seen.

Multiple-selections have been non-function in the tree for quite a number of versions.

MeerK40t main window doesn't open in upper left where it would be expected. Other windows (Terminal, Preferences, Controller, etc.) all open upper left of main window which would make sense if the main window was in top left.

MeerK40t main window doesn't always transition from a second display when it is turned off (the title bar hangs off the top and cannot be grabbed).

littlemandave, 2020-11-03:
All of the above, plus: on launch, I always get 2 windows opened automatically.

joerlane, 2020-11-03:
I suspect you have a second device in the Devices list (Device Manager). Also; do the windows have a different number in the title bar? Like Lhystudios:1 and Lhystudios:2 

If so; you should be able to delete one. You may have to try this a few times to get it to stick; but deleting the second entry, and closing the windows to quit MeerK40t should resolve that.

tatarize, 2020-11-03:
Yeah, that's 2 devices in Device Manager. Right-click and turn one of them off.

littlemandave, 2020-11-04:
Thanks! Yes, exactly, Lhystudios:1 and Lhystudios:2. I've been just closing
the extra window, but I'll kill it in the device manager too.

rsre, 2020-11-18:
Also, adding 
`<key>NSHighResolutionCapable</key>
<true/>`
in the info.plist might be useful for those using high definition displays.

andyjedell, 2020-12-11:
hi when trying to connect to device it fails to connect in the log it is below, any help would be apreciated

Using LibUSB to connect.
Attempting connection to USB.
Attempting connection to USB.
Using LibUSB to connect.
Finding devices.
K40 device detected:
DEVICE ID 1a86:5512 on Bus 006 Address 005 =================
	 bLength                :   0x12 (18 bytes)
	 bDescriptorType        :    0x1 Device
	 bcdUSB                 :  0x110 USB 1.1
	 bDeviceClass           :   0xff Vendor-specific
	 bDeviceSubClass        :    0x0
	 bDeviceProtocol        :    0x2
	 bMaxPacketSize0        :    0x8 (8 bytes)
	 idVendor               : 0x1a86
	 idProduct              : 0x5512
	 bcdDevice              :  0x304 Device 3.04
	 iManufacturer          :    0x0 
	 iProduct               :    0x0 
	 iSerialNumber          :    0x0 
	 bNumConfigurations     :    0x1
	  CONFIGURATION 1: 100 mA ==================================
	   bLength              :    0x9 (9 bytes)
	   bDescriptorType      :    0x2 Configuration
	   wTotalLength         :   0x27 (39 bytes)
	   bNumInterfaces       :    0x1
	   bConfigurationValue  :    0x1
	   iConfiguration       :    0x0 
	   bmAttributes         :   0x80 Bus Powered
	   bMaxPower            :   0x32 (100 mA)
	    INTERFACE 0: Vendor Specific ===========================
	     bLength            :    0x9 (9 bytes)
	     bDescriptorType    :    0x4 Interface
	     bInterfaceNumber   :    0x0
	     bAlternateSetting  :    0x0
	     bNumEndpoints      :    0x3
	     bInterfaceClass    :   0xff Vendor Specific
	     bInterfaceSubClass :    0x1
	     bInterfaceProtocol :    0x2
	     iInterface         :    0x0 
	      ENDPOINT 0x82: Bulk IN ===============================
	       bLength          :    0x7 (7 bytes)
	       bDescriptorType  :    0x5 Endpoint
	       bEndpointAddress :   0x82 IN
	       bmAttributes     :    0x2 Bulk
	       wMaxPacketSize   :   0x20 (32 bytes)
	       bInterval        :    0x0
	      ENDPOINT 0x2: Bulk OUT ===============================
	       bLength          :    0x7 (7 bytes)
	       bDescriptorType  :    0x5 Endpoint
	       bEndpointAddress :    0x2 OUT
	       bmAttributes     :    0x2 Bulk
	       wMaxPacketSize   :   0x20 (32 bytes)
	       bInterval        :    0x0
	      ENDPOINT 0x81: Interrupt IN ==========================
	       bLength          :    0x7 (7 bytes)
	       bDescriptorType  :    0x5 Endpoint
	       bEndpointAddress :   0x81 IN
	       bmAttributes     :    0x3 Interrupt
	       wMaxPacketSize   :    0x8 (8 bytes)
	       bInterval        :    0x1
Config Set
Config Set: Success
Setting Active Config
Active Config: Success.
Attempting to claim interface.
Interface claim: Failed. (Interface is in use.)
Attempting to release interface.
Interface released.
Attempting to claim interface.
Interface claim: Failed. (Interface is in use.)
Connection to USB failed.

Connection to USB failed.

tatarize, 2020-12-12:
@andyjedell Doesn't seem very mac os specific. But, it seems like the usb interface is in use. Do you have another program using that USB device at the same time? Like Whisperer up and initialized or something? It's an issue with the USB, it found the device fine though. Just the part where it tells the OS that it would like to own that device get rebuffed. Either it's owned by something else or some permission thing isn't there. I have heard some things wanting sudo or root to access that stuff but I've never personally seen it.

tatarize, 2020-12-12:
@rsre as for that high resolution thing that should be in the latest builds. Good catch and suggestion.

andyjedell, 2020-12-12:
@tatarize Its a Fresh install of OSX don't have whisperer or any other software installed

rsre, 2020-12-12:
> @rsre as for that high resolution thing that should be in the latest builds. Good catch and suggestion.

Perfect, I just tested and seems to be working in macOS 10.14.6. Thank you.

tatarize, 2020-12-13:
> @tatarize Its a Fresh install of OSX don't have whisperer or any other software installed

I'd try making sure things have access, its the OS rejecting the program's attempt to own that particular usb device.

closed 2021-08-12 by tatarize