meerk40t/meerk40t #256: Cutplanning within a closed shape.

tatarize, 2020-11-03:
@camrbuss 

From what is stated above, it makes it easy enough to put any of your cuts/engraves that you want to happen before the final cut in an operation the precedes the final cut. I usually do this just by setting a RGB color in inkscape. The issue that I run into is when I am cutting out a rectangle with curved edges. The curved edges and lines are represented by individual path elements with M or m attributes for d. It could be left up to the user to get these elements in the correct order so the laser cuts the top line, the right line, the bottom line, and the finally the left line, but often the path is parsed so it does the top line from right the left, then it does the right line from bottom to top. This causes an incredible amount of wasted time even with rapid moves and affects cut quality with interruptions.

I believe that it would be possible to determine if a series of paths are in a "closed" loop by guessing if the endpoints of svgelements::Arc or svgelements::Line are within a certain tolerance of each other. Once you determine if a series of paths is closed, then proceed in re-arranging the path elements in the right order to allow for one contiguous cut around a closed loop. Maybe the Inkscpae Gcode tools extension has something that could be borrowed?

I am happy to help contribute, but am still finding my way around the code base. Would this live in svgelements or in something after Classify so that it can only be applied to Cut operations?

Would an issue be better suited on the svgelements repository?

tatarize, 2020-11-03:
The various cut planning steps can be rearranged. But, this would certainly not fall under `svgelements` except for something like cycling the shape as part of the loop. There's actually some code there supporting an operation like that. When you reverse a shape it messes with that order. Changing the starting PathSegment is not too far out of the way. 

Cut planning here would jump out of the various cut processes and do those items and be able to determine a different order that would be better by being able to cycle that order orders. However, it would be more useful would be to make it decide the order by the optimization routine itself.

If you want you might be able to transform a rectangle and move the starting corner, like toss on a rotate and it might move that thing. The reason it's actually set the way that it is set is because SVG 2.0 Rect Shape decomposition says to do it that way (https://www.w3.org/TR/SVG2/shapes.html#RectElement) since `svgelements` defers to the standard whenever there is a decision to be made like what corner a rectangle starts with that's not going to be changed there.

I have a working implementation of 2-opt for optimizations but currently only for merged paths. But, it'll be used for the proper cutplanner in 0.7.0.

tatarize, 2020-11-03:
The current 0.7.0 code for the cuts divides the pathsegments into CutCode which do not necessarily connect. This will be where most of the cutplanning will end up. So the closed shape property will matter when one closed shape is within another closed shape for a Cut operation which would provide a constraint that that shape must occur before the other shape. But, the individual sequence of the pathsegments do not need to occur in the same order of the shape. So if I minimize the travel for the jumps between the paths. It is the case that it doesn't need to do every part every shape in the order laid out. I could rather treat each CutObject which is basically a pathsegment with laser properties as it's own thing. The shortest path between two shapes may include jumping around for the shapes and does.

One notable Youtube MeerK40t channels  ( https://youtu.be/5X2x7g99CCI?t=142 ) shows generally what this would look like. For some reason all of his designs get chopped into single pathsegments like they are all `move + line` and `move + curve` so the sort of hidden/advanced `optimize travel` feature actually doesn't care what corner a rect element starts in. It literally has the ability to jump within a shape to another another shape and back again if it finds that path to be shorter. This results in a strange but kind of amazing pathing (See the video at the starting time). In theory there are clear benefits to running each CutCode object in a sequence like this. Or having it do all the shapes in the order you set. Though it would be hard to tell without a preview which is part of the reason for project foresight being in 0.7.0.

But, divide each cut into its own object with its own settings, then optimize the movements between them with the 2-opt algorithm for solid semi-optimal TSP solution. This would currently not require paths be linked since it would break each individual segment into its own CutObject and take those in whatever order is better given the heuristic (typically shorter jumps). These issues would be settled with proper cut planning. Which is currently a work in progress in 0.7.0.

tatarize, 2020-11-03:
> It could be left up to the user to get these elements in the correct order so the laser cuts the top line, the right line, the bottom line, and the finally the left line, but often the path is [...]

This is not standard for a rectangle. Can you give an example of when it would do this? Because actual rectangle objects don't do this and the correction in plot-planner would fix it but I've never had rectangle sides flipped and they could only really ever go in the correct order if they are a single svg path. I'd need to see the SVG/DXF or whatever since this isn't really the way it should ever be doing that. I'm not even sure it's possible without breaking it akin to the way Milan's are done, which actually does subdivide the rectangle before hand into individual shapes that really are at time backwards. Which is why I told him to use the `optimize travel` hidden feature since it would reverse those lines and take the path in a plausible order. ( https://www.youtube.com/watch?v=9DQpuke8j_w ).

tatarize, 2020-11-03:
> Maybe the Inkscpae Gcode tools extension 

If you know the algorithm or what it's doing I could certainly do that. But, the code there is GPL so I can't use any of it for my MIT Licensed project. 

> I believe that it would be possible to determine if a series of paths are in a "closed" loop by guessing if the endpoints of svgelements::Arc or svgelements::Line are within a certain tolerance of each other.

The closedness of a path is determined by whether the path ends with a Z command. It can merely be checked. If the method you're using is just generating pathsegments that are each moved into position. The `optimize travel` would take care of your issue, but the determination that 4 cuts that look separated are actually a single shape that should be treated as closed is a different and harder problem. This would make the inner first cut optimization harder, though some of the current methodology for determining the insideness of a particular object could work since I use a fairly interesting algorithm there to determine if a point is inside a particular set of shapes. Depending on the context this gets a little bit harder to figure out from the geometry. It might be best to have the shapes declared contiguous somehow in the program if they are weirdly broken into pieces with particular line elements backwards. This is actually what Milan has in his youtube videos which is why I had him optimize travel and would ask you to do the same. I don't know why the segments are backwards in his or your case, but it seems as if it's the program generating the SVG. The SVG doesn't care, but when the laser cutting uses the SVG order it starts to matter greatly.

But, the fault would maybe lie with the program you're using to generate the SVG unless I got something significantly wrong about my analysis. But, all of this would be fixed with robust cut planning. And would be *greatly* helped by the less robust cut planning I currently have. 

If you want to lean into `optimize travel` make sure you have 0.6.10+ since it corrected a rarish bug with that that Milan found.

tatarize, 2020-11-03:
> I am happy to help contribute, but am still finding my way around the code base. Would this live in svgelements or in something after Classify so that it can only be applied to Cut operations?

@camrbuss The typical inner first stuff is done in jobs. It's one of the pre-process commands called optimize_cuts which applies the inner first optimization. Currently the only way to do the `optimize travel` is by rearranging your paths as they currently exist in the scene. Eventually and in 0.7.0 the plan is for the staging to be in several parts:

* Copy - Makes a copy of the operations.
* Preprocess - Determines required commands to make these operations usable.
* Validate -  Run the command operations.
* Blob - Convert the operations into a cutcode blob.
* Preopt - Determine the selected and needed pre-optimization commands.
* Optimize - Run the optimizations.

Basically 0.6.x is based overly inside the actual Jobs gui dialog. So there's not really a set workflow for these things. But, following this plan it should make a copy of the operations, run some stuff, covert those to cutcode, run some stuff. Then let you execute the cutcode. In the 0.6.x series this step is just the first command added in the jobs dialog is to make sure we do cut_optimization and then there's nothing for the other optimizations. But, those optimizations work on the actual SVG paths themselves which has some weird knock-on effects.

In the 0.7.x series this should be properly staged and optimizations would be performed purely on cutcode blobs which is rather clearly a better way to do these things. So if you wanted things like removing a duplicated cut (say two rects are actually touching each other, it'll cut nothing without removing the doubled cut or doing it manually), or optimizing the travel to flip some cuts over, to make a single shape would have a proper home.

If something is confusing in the code or you feel like there'd be benefits from explaining it in greater detail go ahead and ask. I don't get many people actually looking through the code so if something confusing there or named weirdly please do mention it. I am absolutely planning on doing a couple videos on the code itself, to explain the inner workings and hopefully help people to hack on the code in helpful ways. But since 0.7.0 ( https://github.com/meerk40t/meerk40t/tree/0.6.5 weirdly named but that's 0.7.0 series) changed a number of things in the Kernel and some core parts of the API and developed out the cut planning, I held off for the 0.6.x series of code. Since I figured it would be different after 0.7.x. There's also some interesting big upgrades like cutcode and primary uses of channels based in the kernel which is going to do wonders for things like sending code over TCP and UDP and integration of various processes within MeerK40t.

Most of the documentation on how it works is from like 0.4.x and 0.5.x so it's heavily wrong and the code itself is generally the only thing you can read for documentation there.

camrbuss, 2020-11-03:
@tatarize Thanks for all the writeup! In general I do find the Wiki very useful and as I learn more I may put some PR to update it. Thanks for all the awesome work!

To clarify my main concern is with a "rectangle" that is made up of 8 paths (4 lines, 4 arcs).

When FreeCAD exports a View to SVG it uses a series of paths that look valid in Inkscape object viewer or in MeerK40t, but I get the following exception.

```
MeerK40t crash log. Version: 0.6.10
Please report to: https://github.com/meerk40t/meerk40t/issues

Traceback (most recent call last):
  File "/home/camrbuss/Documents/meerk40t/JobInfo.py", line 264, in on_button_start_job
    self.preprocessor.execute()
  File "/home/camrbuss/Documents/meerk40t/OperationPreprocessor.py", line 30, in execute
    cmd()
  File "/home/camrbuss/Documents/meerk40t/OperationPreprocessor.py", line 149, in optimize_cuts
    op_cuts = CutPlanner.optimize_cut_inside(op)
  File "/home/camrbuss/Documents/meerk40t/CutPlanner.py", line 84, in optimize_cut_inside
    if CutPlanner.is_inside(subpaths[k], subpaths[j]):
  File "/home/camrbuss/Documents/meerk40t/CutPlanner.py", line 48, in is_inside
    if outer_path.bounding_box[0] > inner_path.bounding_box[0]:
TypeError: 'NoneType' object is not subscriptable
```

[image]

``` xml
<?xml version='1.0' encoding='UTF-8' standalone='no'?>
<svg width="297mm" viewBox="0 0 2970 2100" baseProfile="tiny" xmlns:cc="http://creativecommons.org/ns#" height="210mm" version="1.2" xmlns:xlink="http://www.w3.org/1999/xlink" xmlns:freecad="http://www.freecadweb.org/wiki/index.php?title=Svg_Namespace" xmlns:dc="http://purl.org/dc/elements/1.1/" xmlns="http://www.w3.org/2000/svg" xmlns:svg="http://www.w3.org/2000/svg" xmlns:rdf="http://www.w3.org/1999/02/22-rdf-syntax-ns#">
 <title>FreeCAD SVG Export</title>
 <desc>Drawing page: Page exported from FreeCAD document: Unnamed</desc>
 <defs/>
 <g id="Page">
  <g stroke-linecap="square" fill="none" id="DrawingContent" stroke="black" fill-rule="evenodd" stroke-width="1" stroke-linejoin="bevel">
   <g transform="matrix(1,0,0,1,1485,1050)" fill="#ffffff" fill-opacity="1" stroke="none" font-weight="400" font-style="normal" font-family="Sans Serif" font-size="31.75">
    <path fill-rule="evenodd" d="M1000,-300 L1000,300 C1000,410.457 910.457,500 800,500 L-800,500 C-910.457,500 -1000,410.457 -1000,300 L-1000,-300 C-1000,-410.457 -910.457,-500 -800,-500 L800,-500 C910.457,-500 1000,-410.457 1000,-300 " vector-effect="non-scaling-stroke"/>
   </g>
   <g transform="matrix(1,0,0,1,1485,1050)" stroke-linecap="square" fill="none" stroke="#000000" font-weight="400" font-style="normal" font-family="Sans Serif" stroke-opacity="1" stroke-width="1" stroke-linejoin="bevel" font-size="31.75">
    <rect width="0" y="0" x="0" height="0"/>
   </g>
   <g transform="matrix(1,0,0,1,1485,1050)" stroke-linecap="round" fill="none" stroke="#000000" font-weight="400" font-style="normal" font-family="Sans Serif" stroke-opacity="1" stroke-width="7" stroke-linejoin="bevel" font-size="31.75">
    <path fill-rule="evenodd" d="M-1000,300 L-1000,-300" vector-effect="none"/>
   </g>
   <g transform="matrix(1,0,0,1,1485,1050)" stroke-linecap="round" fill="none" stroke="#000000" font-weight="400" font-style="normal" font-family="Sans Serif" stroke-opacity="1" stroke-width="7" stroke-linejoin="bevel" font-size="31.75">
    <path fill-rule="evenodd" d="M-800,-500 C-910.457,-500 -1000,-410.457 -1000,-300 " vector-effect="none"/>
   </g>
   <g transform="matrix(1,0,0,1,1485,1050)" stroke-linecap="round" fill="none" stroke="#000000" font-weight="400" font-style="normal" font-family="Sans Serif" stroke-opacity="1" stroke-width="7" stroke-linejoin="bevel" font-size="31.75">
    <path fill-rule="evenodd" d="M-800,-500 L800,-500" vector-effect="none"/>
   </g>
   <g transform="matrix(1,0,0,1,1485,1050)" stroke-linecap="round" fill="none" stroke="#000000" font-weight="400" font-style="normal" font-family="Sans Serif" stroke-opacity="1" stroke-width="7" stroke-linejoin="bevel" font-size="31.75">
    <path fill-rule="evenodd" d="M1000,-300 C1000,-410.457 910.457,-500 800,-500 " vector-effect="none"/>
   </g>
   <g transform="matrix(1,0,0,1,1485,1050)" stroke-linecap="round" fill="none" stroke="#000000" font-weight="400" font-style="normal" font-family="Sans Serif" stroke-opacity="1" stroke-width="7" stroke-linejoin="bevel" font-size="31.75">
    <path fill-rule="evenodd" d="M1000,-300 L1000,300" vector-effect="none"/>
   </g>
   <g transform="matrix(1,0,0,1,1485,1050)" stroke-linecap="round" fill="none" stroke="#000000" font-weight="400" font-style="normal" font-family="Sans Serif" stroke-opacity="1" stroke-width="7" stroke-linejoin="bevel" font-size="31.75">
    <path fill-rule="evenodd" d="M800,500 C910.457,500 1000,410.457 1000,300 " vector-effect="none"/>
   </g>
   <g transform="matrix(1,0,0,1,1485,1050)" stroke-linecap="round" fill="none" stroke="#000000" font-weight="400" font-style="normal" font-family="Sans Serif" stroke-opacity="1" stroke-width="7" stroke-linejoin="bevel" font-size="31.75">
    <path fill-rule="evenodd" d="M800,500 L-800,500" vector-effect="none"/>
   </g>
   <g transform="matrix(1,0,0,1,1485,1050)" stroke-linecap="round" fill="none" stroke="#000000" font-weight="400" font-style="normal" font-family="Sans Serif" stroke-opacity="1" stroke-width="7" stroke-linejoin="bevel" font-size="31.75">
    <path fill-rule="evenodd" d="M-1000,300 C-1000,410.457 -910.457,500 -800,500 " vector-effect="none"/>
   </g>
  </g>
 </g>
</svg>
```

This leads me to do a DXF export from FreeCAD, DXF import in Inkscape, recolor of paths in Inkscape, then import into MeerK40t. After it gets into meerk40t, the segments are out of order, so I manually drag them around in the operations pane, but yet in 0.6.10 I still get movements that look like this:

[image]

camrbuss, 2020-11-08:
I think this may need to be opened as a new issue, but I believe one of the problems that I am encountering is svg rect's that have 0 width and height. This causes the `TypeError: 'NoneType' object is not subscriptable` in multiple areas of meerk40t.

From the SVG above, this section is causing the issue
``` xml
   <g transform="matrix(1,0,0,1,1485,1050)" stroke-linecap="square" fill="none" stroke="#000000" font-weight="400" font-style="normal" font-family="Sans Serif" stroke-opacity="1" stroke-width="1" stroke-linejoin="bevel" font-size="31.75">
    <rect width="0" y="0" x="0" height="0"/>
   </g>
```

Error when selecting
```
Traceback (most recent call last):
  File "/home/camrbuss/Documents/meerk40t/wxMeerK40t.py", line 1128, in on_left_mouse_up
    self.widget_scene.event(event.GetPosition(), 'leftup')
  File "/home/camrbuss/Documents/meerk40t/Widget.py", line 187, in event
    response = current_widget.event(window_pos, space_pos, event_type)
  File "/home/camrbuss/Documents/meerk40t/Widget.py", line 898, in event
    xmin = q[0]
TypeError: 'NoneType' object is not subscriptable
```

My original intent of this issue was to improve the order that paths get cut. If you can determine that the end points of the path are at the same point, than it should be possible to reorder the direction that the line is cut. The following GIF demonstrates the problem I would like to solve. 

[image]

tatarize, 2020-11-10:
Sorry for the delay in responding. My last week was consumed with other matters. Yes. 0 length/width rectangles crash.  At issue is that rectangles at 0 length and 0 width remove their 0 length segments which become a path consisting of "" which is technically correct. But, these paths do not have bounding boxes and those turn into None and crash. Strictly because SVG says if length or width is 0 it should be non-rendered. I've added code in the 0.6.11 and 0.7.0 versions to strip these non-rendered paths.

https://github.com/meerk40t/meerk40t/commit/702fa542c40b1243b3208ad055cb10d4e7cfbaad

Now the quirky bit with that stuff rendering the wrong way is a problem that I can't see getting fixed without proper cut planning. Outside of the `optimize travel` feature.

camrbuss, 2020-12-06:
I know this is link is GPL and this project is MIT, but I found running this extension on my SVG or DXF from inkscape solved so many of my issues. Maybe instead of implementing this in the meerk40t code, I could just add documentation to show how to use it effectively. https://github.com/fablabnbg/inkscape-chain-paths

Thoughts?

tatarize, 2020-12-07:
It seems like a pretty great app. In theory it should be covered correctly if cut planning in installed. While merging in a chain isn't hard, without detecting the nearest endpoint in each case and reversing paths as needed a naive version of this could connect a segment backwards resulting in traversing it three times. Currently `element merge` and `optimize travel` in console will do this, sort of. It would figure out that segment is backwards and reverse it. 

In any event currently MeerK40t doesn't have things that would qualify as even rudimentary designing capabilities. If designing things in inkscape helps, have at it. The need to chain these things together correctly wouldn't matter as much with proper cut planning, and I know fablabnbg runs lasercutter stuff, he's known a bit in the ruida programming scene. And I could certainly see  adding this sort of feature to MeerK40t if proper cut planning exists.

closed 2022-07-20 by tatarize