meerk40t/meerk40t #257: problem with fillet in .dfx again..

gyan007, 2020-11-05:
some fillet not showing properly in meerk40t_v_0.6.10
[PlantPot.zip](https://github.com/meerk40t/meerk40t/files/5494305/PlantPot.zip)

tatarize, 2020-11-12:
Yeah, bottom right hand corners in the dfx are clearly swept the wrong direction. 

[image]

tatarize, 2020-11-12:
This is fixed in the dev branch and will be final corrected when 0.6.11 goes live.

tatarize, 2020-11-12:
This was done improperly for dxf: "ARC goes always from dxf.start_angle to dxf.end_angle in counter clockwise orientation around the dxf.extrusion vector, which is (0, 0, 1) by default and the usual case for 2D arcs."

tatarize, 2020-11-12:
https://github.com/meerk40t/meerk40t/releases/tag/0.6.11-beta-1

This should have the code corrected. I'll be a bit for @joerlane to review and approve, but it basically is just a couple very minor bug fixes.

gyan007, 2020-11-15:
@tatarize Thanks a lot!

closed 2020-11-15 by tatarize