meerk40t/meerk40t #263: Operations

RuKeBo, 2020-11-25:
I am curious about how the operations work.  I watched a video but still have some questions.  I brought in an SVG I made in Inkscape with 16 rectangles in a 4 x 4 grid, each with a different color.  When I opened the SVG in meerk40t, it put all the colored rectangles under one Engrave Operation, if it matters, the color was blue.  I wanted to try making a different engrave operation for each rectangle with a different ppi ranging from a low of say 10 all the way up to 1000 ppi.  So I would wind up with a total of 16 Engrave Operations, with a rectangle under each. I added an Engrave operation, changed the color to what I thought was the color of the rectangle I made in Inkscape and adjusted the ppi value.  I then dragged the rectangle of the same (similar) color to that Operation.  I repeated this a few times for some of the other colors, then saved the file.  When I say that I tried to match the color, I didn't make sure they were exact.  I suppose I could have gone in and matched the R,G,B values for each color that I made them in Inkscape.  I am pretty sure I had all the 6 primary colors (red, green, blue, yellow, magenta, cyan) except black.  I also had an engrave operation with a red color and yellow and green among others.

I wanted to see if this would stick, would it open back up as it was since this was becoming a bit time-consuming. I didn't want to go through all this trouble if it wasn't going to work.   It wasn't too bad but the tree would refresh each time and for some reason go to the bottom and I was working at the top.  Anyway, when I made a New file, then opened the one I saved, all the colored rectangles went back to be under one Operation, the blue one, the one it made when I opened the original Inkscape SVG.  The other Operations I added were still there, just nothing under them.

My question is, how is this supposed to work?  Does the color of the Operation need to match the color of the object exactly (R,G,B)?  I don't think it does because there are many different color rectangles under a Blue Operation.  Does it Reclassify them when you open the file?  If so, why did the red one leave the Red Engrave Operation?

Depending on the above answers, what is the point of the color of the Operation?  Is it simply for Classifying objects?  Is this so you can make operations with colors that match objects and then classify them automatically?  Aside from that, or maybe some organizational purpose, it would seem the color can be ignored if you want.

I am using version 0.6.12

Thanks,

Russell

tatarize, 2020-11-25:
The operations are mostly info about what you want to do with a particular path or set of paths. This information isn't provided by the nature of vector graphics themselves so they need to be provided by the operations. Since these need stuff to work on there is a classification of these particular objects. If all the rectangles are under the same engrave operation they all get the same properties of the engrave. If you want a different engrave operation for each rectangle you'd need to put them each in their own operation. The vector elements themselves don't have anything other than the instructions for making the path.

So let's say I want to do these operations entirely in console. I want little 1 inch rectangles in a 4x4 grid. So I type in console:
```
rect 1in 1in 1in 1in
grid 4 4 1.2in 1.2in
element *
engrave
````

Now I have all these elements in a grid but they are all in the same engrave operation. Regardless the setting it'll just do them all as with the same setting for that operation.  So let's say I select the first element and press `alt+e` on my keyboard which shortcuts the `engrave` operation. And I go about and select each of these engrave rectangles and give them an engrave operation. Now I could give them each their own setting. Because they are manually added to an engrave operation. However, in each of these the color of the rectangle is black. If I reclassified them, they wouldn't end up in these same operations I made.

Now if I did what you did and gave them all their own color. You would need to go to operations and hit reclassify. In theory if the color isn't seen elsewhere it will just get it's own engrave operation. So each item having its own color could have worked as such. But, without the exact color the operation doesn't work. Now it occurs to me that there's no actual reasonable way to make 16 different engrave operations with their own color, except by reclassifying the whole colored rectangle thing to generate the engrave operation to hold that color.

The color of the operation needs to match the color of the stroke of the object exactly during classification. If there is no match it will create its own engrave operation, with the color of the object it contains. Going to the root Operations and doing "reclassify" would create these operations. In theory if your operations existed but didn't match exactly they wouldn't be used. I am unsure why they would all end up in blue of the rectangles were stroked with those colors. If they were given a fill of that color, they would classify as a raster and if there was no raster within the right colored stroke operations.

The color of the operation is merely for classification purpose. Matching stroke color gives grounds to classify it. I developed a more open process rather than 1:1 for operation element types to operations but it was confusing. So I sort of layered this operations/colors things on top of that to give it the ability to do things like Whisperer did, or some of the stuff RDWorks and Lightburn tend to have. So you could sort of ignore the system and say blue means engrave, red means cut. Which is the point behind Other/Red/Blue. 

Color can be ignored for some things but if you classify things it becomes wildly important. The easiest method would be to give all the rectangles a color like you gave them. Reclassify them so they hopefully add to their own engrave operations then modify those engrave operations.

The file opening doesn't save the actual linked objects current mapping. Rather it saves all the operations and all the paths, and then reclassifies them based on their colors. #209 

You're clearly right that some of this is weirdly confusing, and if you have a better methodology for this I would be happy to employ it. 

The important bullet points are that operations basically are the things the laser is doing, the elements are the bits of data the elements use in doing those things. The colors are a thinly veiled method of making "classification" easier, and the colors selected need to match exactly. This matches the color of the operation to the stroke color of the path and puts it in there. Some of the other elements classify differently so a raster might get objects based on them having a `fill` color, and they tend to be first come first serve, it filters from the topmost operation and keeps going down trying to find a home for the element.

---

To make things a bit more weird. Version 0.7.0 is introducing some cutcode/blob objects which will be a combination of path-segments and a bunch of different properties for how to laser those segments. There's a variety of reasons for this but basically this is what the operations with a bunch of elements compiles into. And there's some precompiled laser code that already basically has that like lhymicro-gl and ruida, both of which would lose a lot of information to break them down into specifically paths or settings, when they are both. And this is needed for things like `ruidaemulation->m2-nano`

You're right, it's weird. If you have any idea how to make it less weird that would be great.

RuKeBo, 2020-11-25:
Part of the problem is that I am using the wrong operations.  The rectangles I made all had a stroke of blue color and fill of differing colors..  The reason it put them all under a blue engrave is because of the blue stroke, according to what you said.  This makes sense but the stroke is not really what I am after, it just outlines the rectangles.  I am interested in the fill.  As a test, I changed the stroke of each rectangle in Inkscape to the fill color and removed the fill, so they only have a stroke.  I brought this into meerk40t and it worked as expected, it made an Engrave operation for each color.  I expect from my previous description, this is what you thought I was doing originally.

I then went back to the original file in Inkscape with filled in rectangles, removed the stroke so that now I have 16 rectangles each with a different fill color and no stroke.  I brought this into meerk40t and no Operations were made automatically.  I assumed it would make a raster Operation for each color rectangle. I added a Raster Operation with color black and Reclassified, all the rectangles went under that Operation.  I then made a red raster operation, refreshed the classification, they all went under that operation as well.  Each time I make a raster operation, any color, and classify, all the rectangles go under that color.

I can manually add a raster operation then drag a rectangle to it, color does not seem to matter here as they don't classify to it.  I did this for 3-4 rectangles then saved the file.  When I opened this back up, each operation had all the rectangles, not just the one I dragged to it.

Kind of a side note but is there a way to remove multiple objects from an operation without removing them from everywhere else?  For example, if I do what I did above, make a raster operation and refresh classification and it will put everything under that operation. If I want to remove everything except just one rectangle, I can shift-select them but it only gives me the option to remove the last one selected or all, and all removes them from any other operations they are in and out of the Elements, I don't know how to put things back in Elements, if that is possible.

RuKeBo, 2020-11-25:
As I look more at this I found the following which might be the intended way it is supposed to work, it just is not how I would immediately think it should work.  As I think I mentioned above, it does not classify filled objects when a file is opened, it will classify stroked objects.  My assumption would be that it would do both, stroke as engrave and fill as raster, but maybe that is too simplistic or not desired. I had an SVG file with 16 rectangles with only a stroke, all different colors.  I then filled in 4 of the 16 rectangles with red, green, cyan, and yellow and brought it into meerk40t.  It classified the stroked objects with an engrave for each color, as expected, it did not make any raster operations for the fills, as I thought it would have.  I added a raster operation, changed the color to red and refreshed the classification.  It put all the filled rectangles in the raster operation along with the red stroked rectangle that had no fill.  I added another raster operation with color cyan.  Reclassified and it put all the filled rectangles in this operation along with a rectangle that has a cyan stroke, no fill.

So it is classifying the stroked object in both an engrave operation automatically on open and raster operation if it exists.  It classifies ALL filled objects in any raster operation that exists, regardless of the color of the operation.  I can see the desire to classify a stroked object in whatever operation has the same color.  Who is to say whether a stroke is to be strictly engraved or rastered? This is flexible and not limiting, not like K40Whisperer where everything has to be a certain color for an operation. If I happen to want to distinguish an engrave or raster, I just make sure they are different colors.  In my case I had a red stroked rectangle and a separate red filled rectangle so the stroked one got classified as both engrave and raster, I would simply make them different colors to separate them  

I guess since stroke works this way, shouldn't fill as well since I would think this is more specific?  Is "Engrave" a vector or raster engrave?  Does it matter?  If it is vector, then a fill could not be engraved. It would seem that any fill, should be classified as a raster operation only, but maybe it could be image as well.  Regardless, it would still make sense that it can be classified similar to a stroked object or am I missing something?

tatarize, 2020-11-25:
Fills don't classify things except as rasters and then they fall into *all* raster operations.

> I can manually add a raster operation then drag a rectangle to it.

Select the object and hit alt+r on your keyboard. Saves you the draggin.

> Kind of a side note but is there a way to remove multiple objects from an operation without removing them from everywhere else?

There is a clear all when selected from the operation itself.

> If I want to remove everything except just one rectangle, I can shift-select them but it only gives me the option to remove the last one selected or all, and all removes them from any other [...]

Yeah, that's needlessly weird and cumbersome and doesn't work. You also can't drag multiple objects into the same operation. I think this has been planned since 0.6.2 but still not fixed.

>it just is not how I would immediately think it should work.

Fills of different colors are not used entirely just for classifications. They actually actively have that color. If you color something gray it will actually be gray. If you run a #888 grey it will process almost exactly as a black running at 500 ppi would process. If you had a character made entirely of vector objects and you rastered it, the colors of the fill would be important to how the result comes out. If they are put into different raster operations, they would each raster one after another including their overlap.

> My assumption would be that it would do both, stroke as engrave and fill as raster, but maybe that is too simplistic or not desired.

I'm biting a few bullets here. Understand that classification is *not* awesome or perfect, sometimes it's not even good. It's a sort of hybrid system where my original idea. Let's make operations which are the primary thing the laser runs. Then we can add any number of elements to these operations and have them process correctly. So you create some operations and then specify the elements you want that operation to have. And you could reuse elements in different operations, and you can use them as you need them.

That setup, from a programming standpoint seemed really nice. You give maximum power to people. They can say exactly what they want to do with these processes, and then do them. Well, it kinda sucked and confused everybody. They were used to whisperer's workflow where blue stroke meant vector engrave, red stroke meant cut, and everything else meant raster. In my system there wasn't any need for specific colors at all. You could configure them in every way possible. But, again this was confusing. So I slap-dash added colors to operations. These colors are only the color of the stroke of the vectors added to that operation. And I set an easily resettable Other/Blue/Red setting in operations so they could run things like Whisperer did them and not lose any of their work flow.

You're not wrong that this is weird and the colors don't classify things in a fully consistent manner. It is sort of a thinly veiled methodology placed on top of the system I already had that lets you sort of duplicate how Whisperer, Lightburn, and RDWorks already work. It's a more powerful system but that system is harder to use so I gave the people what they wanted: color groups.

tatarize, 2020-11-25:
> It classified the stroked objects with an engrave for each color, as expected, it did not make any raster operations for the fills, as I thought it would have. 

Only stroke color gets a default engrave operation created for it.

> I added a raster operation, changed the color to red and refreshed the classification. ... Reclassified and it put all the filled rectangles in this operation along with a rectangle that has a cyan stroke, no fill.

Also accurate. Even as a raster operation will take everything with a fill, it will also take anything with a stroke that matches it's stated color.

> This is flexible and not limiting, not like K40Whisperer where everything has to be a certain color for an operation. If I happen to want to distinguish an engrave or raster, I just make sure they are different colors.

Which is the intent. I actually made sure my default methods would take any whisperer file properly set and make it do the same operation. This code here isn't perfect and the system isn't optimal but the goal here was to let people reuse their Whisperer files and have a workflow they knew. It is entirely the case that you the *real* system that is under that system is you can put any element in any operation you want, give them properties and run them. But, that system is too hard. And the biggest issue there is that the current objects aren't saved into those operations, when you save the file. They should reference the object and avoid default classification since if you arranged that yourself, you're losing that information. And the saves are lossy.

> I would simply make them different colors to separate them

And the color can be #FFFFFE and it'll get a different classification. Part of the goal was to correct for Whisperer's annoying suite of bugs where somebody has 1 stroke color off by a single digit and so it shows up as a raster without any redemption. In MeerK40t you'll get a different #FFFFFE engrave and could just drag it to where it should go instead.

>I guess since stroke works this way, shouldn't fill as well since I would think this is more specific?

It could work that way. But, I'm not sure people want a raster operation. And it seems like it risked making a dozen fills if they dragged in a vector file with fills of slightly different shades to those fills. If they intended to have them all end up in the *same* vs different raster operation. How could I tell that? And this wasn't how Whisperer worked so if they had files that worked in Whisperer I made sure they'd work in MeerK40t.

> Is "Engrave" a vector or raster engrave?

Vector. I call the raster engrave Raster. 

> Does it matter?

They are largely different processes so I'd think it would.

> If it is vector, then a fill could not be engraved.

Well, sure it could. If you hit `F12` which loads up terminal. Select your closed shape and type `embroider` you'll have a perfectly engravable fill.

> It would seem that any fill, should be classified as a raster operation only, but maybe it could be image as well.

Image have an `Image` operation because there's some different needs with an image than with normal raster processes. For example images already have set pixels and doing a resample at the wrong time to fit the properties is an issue so the image operation has a passthrough step distance. They also raster one after another rather than combine into the same image. All the elements in a raster operation get turned into an image at go-time, but images will plot along one after the next.

> Regardless, it would still make sense that it can be classified similar to a stroked object or am I missing something?

I am missing something. The information as to whether that's something all users would sort of want. There was a lot of back and forth on the irc channel about how this stuff works. It's entirely reasonable for somebody to make a drawing in pure SVG size it up to what they want and use that. Say they make a heart shape, then put some text on that heart and color that text `white` and the heart a `dark gray` (and text to path the text since MeerK40t might mess it up). Then they get this nice rastering of a heart with a text relief since the white won't raster. But, if they fed into the different operations it would raster the entire gray heart and then it would raster the white text (would wouldn't do anything since it would find no pixels). 

Classification is a system put on top of the move-stuff-around-manually system. The colors themselves aren't the only methods either. I could certainly think of other schemes. Have the id of the specify the operation, based on the text. Have it take the svg groups into account and nest the elements better. And sure you could classify them based on fill color in addition to stroke color. But, that system is mostly so that whisperer's workflow works. I am not at all opposed to other workflows or methods. I settled on this as a compromise.

The supposed real system would be to fix the bugs that makes moving, deleting, and grouping, those elements in operations needlessly more hard than it needs to be and correct the loading and saving of already correctly arranged files. Defaulting to classification on load, only if there were not any elements registered inside those operations. Then more advanced users like yourself could more or less transcend the weird rules of classification.

You already figured most of it out but here's the classification source code:

```python
[... cut ...] nsole to change the behavior just for you. And maybe later give other people gui switches for how they would like their classifications to work. Though fixing the issues that make full manual classification harder than it needs to be would be my preference. Though I'd likely push it beyond 0.7.0 since it might be a lot of changes and I have to port all the minor dev changes into the 0.7.0 version, but really the individual fixes are would likely be pretty minor. I just have slacked off on those.

RuKeBo, 2020-11-25:
> Select the object and hit alt+r on your keyboard. Saves you the draggin.
Well, that makes it a bit easier.  This also separates each one into a separate operation.

> Fills of different colors are not used entirely just for classifications. They actually actively have that color. If you color something gray it will actually be gray. If you run a #888 grey it will [...]

Ok, so again, maybe my use case is not very good.  You are saying that the color red will raster differently than green?  I assume this is based on it's shade? It would make no sense to adjust the ppi value of each colored rectangle if the color would affect the outcome anyway.  Sorry, I know I could try this but want to get the most understanding of it before testing and am sort of used to a color doing one thing.  For my example, I was thinking I want to test burning the rectangles on some material and see the what affect the ppi value had.  This would give me some benchmark for settings.  I would use a different ppi value for each rectangle.  I guess what you are saying, and now might seem obvious, is that the only proper way to do this is the have all the rectangles be the same color, probably black, then make a raster operation for each one changing the ppi value for each.  Or, I could probably use different shade rectangles from white to say black with different shades of grey in between and leave the ppi value the same for all.  This would require just one raster operation for all the rectangles.  It would also allow the adjustment of the ppi value for another form of control if needed.  I guess it depends on the goal and honestly, I'd have to think about what I am trying to accomplish, or the best way to be consistent.  In either of these cases, it wouldn't make sense to classify the rasters, in fact, neither used color fills anyway.

I guess I can see why you would not classify rasters by color now, I don't see a case for doing that.

So from the last 3 comments there is no way to save the classification of elements.  So I have not used Lightburn, or any other software besides Whisperer, but what it looked like you could do in some software was classify operations like you do here.  Personally, the auto classify thing might be just an "extra" like you say.  However, it would seem counter-productive to have a classification system and not be able to save that for a project.  I assume you can't save this to the SVG file, and probably wouldn't want to.  Has there been discussion about a Meerk40T file?  Maybe you alluded to this.  When I have seen some other software, it seemed logical and preferable to have some order to events.  As you said and I am not sure the best way, but color, number, ID, letter, could be a way, not sure it matters.  Color is visual and you can even auto classify to it, but that can also be confusing at first (at least it was to me). But what you explained makes sense to me so far.  Again, not sure of the application for this but I assume you could have a project that you want to laser and it might have multiple operations.  If I want to run that more than once, I don't want to have to reclassify everything again.  With that said, and agreeing that it doesn't make too much sense to classify on color anyway, there might not need to be a need to save the classification.  If the auto classification works anyway then you really don't need to save it.

I think I need to put this into practice for it to be more clear and understand what I might or might not need.  But now that you explained it and I thought about it, it seems like it is a good way to do things.

tatarize, 2020-11-25:
> I assume this is based on it's shade?

Yes. Most of the time colors seem darker, blue is darker than pure green. It will get a different shade and will raster differently. If you took a sort premade raster character and you rastered that character with the different colors, it would raster out and be perfectly visible and not entirely black.

The raster process is basically take the raster objects and overlap them as such and make an image out of that and raster the picture.

>For my example, I was thinking I want to test burning the rectangles on some material and see the what affect the ppi value had.

PPI should be almost entirely linear. You would generally need black rectangles.

In console:
```
rect 1in 1in 1in 1in
fill black
grid 4 4 1.2in 1.2in
```
Click each of this and hit alt+r, then set the PPI value. That'd be a fair enough test. You could also likely color them gray based on the PPI. So in theory a rectangle with a #AAA gray color should be ppi 667 Since 170 / 255 = 0.66666... You could likewise do `fill #AAA` and see that this is true comparing it to a similar rectangle. You could dither the rectangles and they would pattern differently a bit.

This setup of a test case just isn't able to be correctly and losslessly saved. In reality that's the real bug here.

> So from the last 3 comments there is no way to save the classification of elements.

There is, I would just need to fix the defect there. Give the various export svgs a proper id, and reference those ids in the operation to add them in. It's been planned for a bit I could move it up if you wanted. 

> So I have not used Lightburn, or any other software besides Whisperer, but what it looked like you could do in some software was classify operations like you do here.

Not quite. The method here is intended to duplicate that a bit. In MeerK40t the operations are real objects, you put the elements in and they run. In Whisperer, there's a button to hit that runs all the red stroke vector objects as a vector cut. And another button you hit that runs all the blue stroke vector objects as a vector engrave. It's not so much classifying things as that's what that button does. Engraves the blue vectors. 

In others like Lightburn and RDWorks.  You get a different and maybe better system. You can download and run RDWorks for free (I'm pretty sure lightburn is based on the UI of RDworks). You get a work layer system. There is a list of like 20 colors. These are pretty much static and save settings and refer to different operations. Your objects added the scene either have one of these colors or they get put in one that might be a grab-all. You can then click the color button at the bottom with the items selected in the field. It changes the color of the item and puts that object in that layer color. So you'd have 20 pre-defined operations that you can quickly assign objects to with a single click. So you have your blue operation with Mode set to Scan with various speed and color settings. It's a manual classification system which actually tends to assign color/settings functions to particular vector objects. This seemed a bit 1:1. Each item could be assigned a single setting by applying a color/operation to that item. That felt like a much less open system, but it's also a lot easier. You'd have a color bar of buttons that you hit the color for, and it'll give it those settings you set previously and they save between uses. So your blue is always whatever you set blue to.

> Personally, the auto classify thing might be just an "extra" like you say. However, it would seem counter-productive to have a classification system and not be able to save that for a project. I [...]

You can save this information to a SVG file, and I already do. MeerK40t svgs save operation data already. They just rely rather wrongly on auto-classification, when I should define explicitly the links between the elements and operations. The Namespace is defined https://github.com/meerk40t/meerk40t/wiki/Namespace and it saves it correctly. This lets you save the SVG file, with the operations, and the notes you have for that file without any issue. You can then load these SVGs into other programs etc without trouble. The SVG spec has specific requirements for dealing with unknown objects, so it's fine to just extend the SVG format a bit and use that. I just lack the list of id tags I'd need to reference the contents of that particular object.

> With that said, and agreeing that it doesn't make too much sense to classify on color anyway, there might not need to be a need to save the classification. If the auto classification works anyway then you really don't need to save it.

Usually when you save the SVG in MeerK40t it saves the classifications you set. But, rather than put the objects where they belong it reclassifies them. And there's really no way to correctly classify several different raster operations.

Yeah, you should check out RDWorks, it has a great sort of UI for this, that is largely the same as it is in Lightburn. You can also check a wordpad of a MeerK40t svg file and see the operations. In fact, you can just define a bunch of operations and save those out as svg files anyway. Then loading them will force things to auto-classify into them. Letting you have different settings for oak or acrylic or whatever.

closed 2020-12-01 by RuKeBo