meerk40t/meerk40t #267: Cut option on CLI (no colour)

egasz, 2020-11-26:
Hello, as I found out through a post on Reddit, it's not possible to issue a cut command in the CLI if the colour of the SVG is not correct.
I don't know if this counts as an issue or a request, but here is my suggestion.
Meerk40t could have an argument in the CLI command so that it knew what was the operation, e.g
-c for cutting or -e for engraving
I don't know if this would work, but here's my reasoning for it.
At the moment when you input an SVG, Meerk40t looks at the "stroke" parameter and finds out what the operation is based on the colour (I assume it's blue for engraving and red for cutting. 
My proposal was that when receiving the '-c' argument in the CLI command it would cut every thing on the SVG regardless of the "stroke" value, and would bypass that "colour recognition to match operation" feature.
It would look something like: 
MeerK40t -zacS 10 drawingexample.svg
And regardless of the object colours in the svg (stroke values), it would cut everything.

Here's one of the svg files that I have:

<?xml version="1.0" encoding="UTF-16"?>
<!DOCTYPE svg PUBLIC "-//W3C//DTD SVG 1.1//EN" "http://www.w3.org/Graphics/SVG/1.1/DTD/svg11.dtd">

<svg xmlns="http://www.w3.org/2000/svg" xml:space="preserve" width="100mm" height="100mm" shape-rendering="geometricPrecision" text-rendering="geometricPrecision" image-rendering="optimizeQuality" fill-rule="evenodd" clip-rule="evenodd"
viewBox="0 0 10000 10000">
 <g id="Layer_x0020_1">
  <metadata id="CorelCorpID_0Corel-Layer"/>
  <rect fill="none" **stroke="#DA2724**" stroke-width="7.62" x="8004.9" y="562.3" width="1054.7" height="5910.3"/>
  <ellipse fill="none" **stroke="#DA2222**" stroke-width="7.62" cx="8582" cy="6383.05" rx="955.2" ry="527.35"/>
 </g>
</svg>

And it would be nice if it would cut both these item whether the stroke value is "DA2222"/"DA2724"/"black"/"red"/"blue"/etc...

Thank you guys, and keep up the good work

egasz, 2020-11-26:
Btw, at the moment I created a script to overcome the issue by receiving the file, replacing all stroke values and issuing the meerk40t command with the new file (new file has all stroke ="red") but the weird thing is that after the cutting is done the terminal still seems to be flooded with messages from MeerK40t. I'm running in verbose mode.

tatarize, 2020-11-26:
In theory this should all default to running a engrave without the ability to do the only operation that makes an engrave not an engrave. Namely it doesn't do the cut_inner optimization of the file to make sure the inner parts are cut first. The only actual difference beyond that is speed.

Cuts really are the same operation as Engrave except for them doing an optimization to ensure we do cut inner first. I don't think it cares what color it is but let me double check.

tatarize, 2020-11-26:
```
(venv) C:\Users\Tat\PycharmProjects\meerk40t-64>MeerK40t.py -zcS 10 ktest.svg
>operation
operation
----------
Operations:
0: Image 140mm/s =T2B 1000ppi ±20
1: Raster1 140mm/s =T2B 1000ppi ±20
2: Engrave 35mm/s 1000ppi
3: Cut 10mm/s 1000ppi
----------
>element
element
----------
Graphical Elements:
0: m 10871.5401555,13.8297716902 l 0,115.104239068 l ...
1: m 2357.58719106,5.41665664163 l 0,115.104239068 L ...
2: m 308.750215975,5.41665664163 l 0,115.104239068 l ...
3: m 9273.70971401,5.41665664163 l 0,115.104239068 l ...
----------
>stroke
stroke
----------
Stroke Values:
0: stroke = #ffff00 - m 10871.5401555,13.8297716902 l 0,115.104239068 l ...
1: stroke = #00ff00 - m 2357.58719106,5.41665664163 l 0,115.104239068 L ...
2: stroke = #0000ff - m 308.750215975,5.41665664163 l 0,115.104239068 l ...
3: stroke = #ff00ff - m 9273.70971401,5.41665664163 l 0,115.104239068 l ...
----------
>element *
element *
Selecting all elements.
>classify
classify
>operation
operation
----------
Operations:
0: Image 140mm/s =T2B 1000ppi ±20
1: Raster1 140mm/s =T2B 1000ppi ±20
2: Engrave 35mm/s 1000ppi
3: Cut 10mm/s 1000ppi
4: Engrave 35mm/s 1000ppi
5: Engrave 35mm/s 1000ppi
6: Engrave 35mm/s 1000ppi
----------
>shutdown
```

You appear to be right. I don't know why but this thing isn't actually classifying the operations you have. When they were loaded from the CLI they *should* have been classified automatically into their own little engrave and doing a -S should have set the speed.

tatarize, 2020-11-26:
Wait, it actually needs the -a flag.

```
----------
Operations:
0: Image 10mm/s =T2B 1000ppi ±20
1: Raster1 10mm/s =T2B 1000ppi ±20
2: Engrave 10mm/s 1000ppi
3: Cut 10mm/s 1000ppi
4: Engrave 10mm/s 1000ppi
5: Engrave 10mm/s 1000ppi
6: Engrave 10mm/s 1000ppi
----------
>stroke
stroke
----------
Stroke Values:
0: stroke = #ffff00 - m 10871.5401555,13.8297716902 l 0,115.104239068 l ...
1: stroke = #00ff00 - m 2357.58719106,5.41665664163 l 0,115.104239068 L ...
2: stroke = #0000ff - m 308.750215975,5.41665664163 l 0,115.104239068 l ...
3: stroke = #ff00ff - m 9273.70971401,5.41665664163 l 0,115.104239068 l ...
----------
>
```

The -a sets the speed of all the operations to that value given. This will result in everything running at 10mm/s. The operations being `Engrave` rather than `Cut` can't make a difference because I think in this mode they actually can't run the pre-processing stuff anyway. It would just run the `Cut` as `Engrave` anyway since it won't do the inner-first optimization and that's the only difference between the two. That and cut goes last. But, that assumes you didn't create other items first.

tatarize, 2020-11-26:
As for the other issue this should work in 0.7.x if all goes to plan. The plan would be rather than just give you a strict CLI environment you should be able to launch most things through the Console directly, and via a batch file. The option for -b <batchfile> already exists as does -c for `Console` mode. the difference here would be that you'd be given a bit more access so you could define your operation directly, classify it, and start it running in the console. And also have access to things like a evg-server thing where you could send data fairly directly to your laser. So you could do the work on your laptop and send the precompiled egv code to your laser over your network connection or whatever.

closed 2021-01-30 by tatarize