meerk40t/meerk40t #268: Add pause or wait operation

RuKeBo, 2020-12-01:
Not sure if anyone else has a use for this and could not find a way to do it.  It would be nice to be able to add a wait or pause Operation.  I see that a Wait can be added in the Job list but it adds it to the end and I don't see a way to move it, I am also not sure what this does.  It would be helpful, for me, to be able to pause or wait between specific operations.  For instance, if I want to do an engrave or raster without air assist on, then turn the air on before I do a cut operation(s), I would like it to pause before the cut so I can physically turn the air on.  Otherwise, I have to be right there next to it to catch it as it transitions, or just turn the air on before the raster is done and cut starts.

I did not try the Wait in the Job list, so not sure what it does and maybe that is unrelated to what I am asking here.  I assume you'd have to have some way to un-pause it though.  To me, it would make sense to add this as an Operation so it can be setup before hand and you can move it around in the stack of Operations to be performed. Maybe it would be added to the "Engrave, Cut, Raster, Image" list?  I haven't thought this through thoroughly and how that would affect the classifications.  I assume you remove the color or just don't classify a Pause operation.

tatarize, 2020-12-02:
You'd want Interrupt. And yeah, it would make sense to allow it into the operations lists. There's still a kinda open question as to whether Cutcode should get classified there. But, it probably would be, but it has some hybrid aspects of being elements too. Adding some other spoolable functions to the pre-jobs things makes a lot of sense.

renarsvilnis, 2021-04-10:
Just adding feedback.
Same as @RuKeBo would like to have the ability to set an indefinite pause to change power settings between engraving and cut operations. I imagine the job window showing a "resume job" button.
[image]

tatarize, 2021-04-10:
There is one, called `interrupt` but it's a little difficult to add since you can't really rearrange variosu objects extremely easily.

renarsvilnis, 2021-04-12:
@tatarize Thanks, didn't notice the "Add" on the macOS menu was for the job window. 😄  
Yeah, having it not be rearrangeable is a bit annoying now. Another thing is that it gets lost for each job/project.

tatarize, 2021-08-12:
This works correctly currently.

closed 2021-08-12 by tatarize