meerk40t/meerk40t #269: Question: Possibility for other software (

renarsvilnis, 2020-12-04:
Hi,
I'm interested in automating some stuff with k40, like shutting off-pump after a job is done, do something else when it starts etc. Is there a way without extending meerk40t to able to listen to start/stop/error/halt/.. events on the same system but from another process? 

The plan is to run meerk40t on a Pi which communicates with Arduino has custom software for LCD, relays, sensors etc.

joerlane, 2020-12-04:
While I can't speak directly to the job sharing inquiry; I do wonder, why the added complexity in this case? For example; the M2 Nano board itself has a switching port for things like the water pump. While that port is not enough to power an electromagnetic relay on its own; it can trigger a solid state relay just fine. I have a 40A rated SSR (probably overkill, but it was only $5) attached to mine that triggers the water pump, thermostat, exhaust fan, and air assist; all of which turn off automatically when the job is done.

The Raspberry Pi also has GPIO onboard; so it could directly drive those other things, like an LCD, relays, sensors, etc.; without needing an additional Arduino.

renarsvilnis, 2020-12-04:
@joerlane Yeah I know, tried raspberry only, got some limitations with my LCD and led strip variant. But I'm going over-engineering way to try different things and learn something.

joerlane, 2020-12-04:
That's fine. I just wanted to make sure you weren't trying to reinvent the wheel for no reason at all. 

I'm sure tatarize will stop in at some point and comment. I could use a refresher myself; as I've been meaning to interface a temperature sensor, and water flow sensor on my Pi's GPIO; as triggers to pause MeerK40t.

tatarize, 2020-12-06:
Sorry for the delay spent a while hacking on some old deterministic fractal noise algorithm I wrote.

--

This is certainly a desire. I'm not chiefly sure how to implement it. Obviously a direct hook into the script during an event would be the most obvious. In 0.7.0 there is a console server, so you could access meerk40t while it's running and change things if you start up the consoleserver. MeerK40t could also be coaxed into running other scripts or perhaps performing a socket connection itself.

I'm not sure what is best. There would also apparently be some plugin ability with pip. Someway to look through the entrypoints of installed items and finding items that are also installed and have the right entry points to sort of include these as fully fledged plugins. If MeerK40t then checked for those it could register the plugin and then make that functionality take effect. I am unsure how this interfaces with the precompiled binaries (I would assume it doesn't). 

In the current version I believe you can write the channels to disk, there might be some way to check and parse that information. Though I'd guess it's the least desirable way to do that.

The core elements within MeerK40t are built around the idea of having functionality that should easily hook into the rest of the system while not depending on it. It has a built in kernel with some powers behind it, which is the major change in 0.7.x that I'm putting the console in the kernel and changing how it works, so that it can grow in these directions. There's not a currently very api friendly way of doing this, but those signals of the meerk40t state are broadcast, and all it takes is a module to listen to those broadcasts and execute some code.

If you want a quick fix I could likely write that data to a file if you turned it on. Just add in a weird default module that registers a console command to write the current broadcast state to disk or something like that. (at least that's how it would work in 0.7.0, in 0.6.x I'd add a hardcoded console command and hide it just for you, which would use that module to just write some state information to disk).

It might also be fairly reasonable to add in triggers. Basically listen for a signal, and if that signal is triggered with the right state, execute some predefined command, the commands would likely be console commands rather than python commands, though adding one as `system_execute <>` would then be rightly powerful, and it could allow the establishment of system behaviors that could be redefined.

There's a lot of ideas but certainly no clear way forward at this point.

tatarize, 2021-01-18:
Some cursory work has been done with MeerK40t in 0.7.0 for allowing entry points and creating plugins. This would allow some rather strong interactions between meerk40t main and whatever code you'd write. It opens up listening to all the internal signals within MeerK40t and doing whatever sorts of things you need with that.

tatarize, 2021-08-12:
This is now pretty well established with plugins.

closed 2021-08-12 by tatarize