meerk40t/meerk40t #270: Numpy installation fails to pass a sanity check due to a bug in the windows runtime.

dietervansteenwegen, 2020-12-05:
Haven't used my laser for a while, but someone asked me to show her how to use hers. Thought I'd recommend her to use Meerk40t. However, after a GIT pull I can't start Meerk40t on my main machine. I get the following:

```
xxxxxxxxxxxxxxxxxxxxxxxxxxxx\meerk40t>pip install -r requirements.txt
Requirement already satisfied: Pillow in c:\users\dieterv\appdata\local\programs\python\python38\lib\site-packages (from -r requirements.txt (line 1)) (8.0.1)
Requirement already satisfied: pyusb in c:\users\dieterv\appdata\local\programs\python\python38\lib\site-packages (from -r requirements.txt (line 2)) (1.1.0)
Requirement already satisfied: wxPython in c:\users\dieterv\appdata\local\programs\python\python38\lib\site-packages (from -r requirements.txt (line 3)) (4.1.0)
Requirement already satisfied: numpy; python_version >= "3.0" in c:\users\dieterv\appdata\local\programs\python\python38\lib\site-packages (from wxPython->-r requirements.txt (line 3)) (1.19.4)
Requirement already satisfied: six in c:\users\dieterv\appdata\local\programs\python\python38\lib\site-packages (from wxPython->-r requirements.txt (line 3)) (1.15.0)
WARNING: You are using pip version 20.2.4; however, version 20.3.1 is available.
You should consider upgrading via the 'c:\users\dieterv\appdata\local\programs\python\python38\python.exe -m pip install --upgrade pip' command.

xxxxxxxxxxxxxxxxxxxxxxxxxxxx\meerk40t>python MeerK40t.py
 ** On entry to DGEBAL parameter number  3 had an illegal value
 ** On entry to DGEHRD  parameter number  2 had an illegal value
 ** On entry to DORGHR DORGQR parameter number  2 had an illegal value
 ** On entry to DHSEQR parameter number  4 had an illegal value
Traceback (most recent call last):
  File "MeerK40t.py", line 4, in <module>
    from Console import Console
  File "P:\Dieter\Projects\K40\meerk40t\Console.py", line 1, in <module>
    from CutPlanner import CutPlanner
  File "P:\Dieter\Projects\K40\meerk40t\CutPlanner.py", line 1, in <module>
    from svgelements import *
  File "P:\Dieter\Projects\K40\meerk40t\svgelements.py", line 20, in <module>
    import numpy as np
  File "C:\Users\dieterv\AppData\Local\Programs\Python\Python38\lib\site-packages\numpy\__init__.py", line 305, in <module>
    _win_os_check()
  File "C:\Users\dieterv\AppData\Local\Programs\Python\Python38\lib\site-packages\numpy\__init__.py", line 302, in _win_os_check
    raise RuntimeError(msg.format(__file__)) from None
RuntimeError: The current Numpy installation ('C:\\Users\\dieterv\\AppData\\Local\\Programs\\Python\\Python38\\lib\\site-packages\\numpy\\__init__.py') fails to pass a sanity check due to a bug in the windows runtime. See this issue for more information: https://tinyurl.com/y3dm3h86
```

Following the link on the last line leads me to think that the issue can't be fixed by the Meerk40t devs, but maybe a previous version of Numpy woulf work?
No txt log file is created by Meerk40t, as far as I can tell.

Downgrading by running
```pip install numpy==1.19.3```` seems to fix this, though I can't be sure as this leads me to another issue (will create ticket)

tatarize, 2020-12-06:
Hm. It means I need to change the svgelements import to seem much more runtime and less required. Since I switched to 1.3.x of svgelements I was somewhat worried about that. I need to more clearly make it import when only the unused numpy stuff is called for. Really numpy isn't a requirement. If that import fails it simply continues running. But, here apparently there's like two versions getting called or something weird.

tatarize, 2020-12-07:
This should be fixed in the main branch now.

closed 2020-12-11 by tatarize