meerk40t/meerk40t #271: Cutting a saved .svg file from CLI

davthomaspilot, 2020-12-08:
This isn't a bug report.  Maybe a feature request, but I'm guessing it's already possible and I just need some help.  Let me know if there's a better place for things like this.

I'd like to get the simplest possible workflow from CorelDraw to cut material using meerK40t (not CorelLaser).   I'm thinking that would involve exporting the design from CorelDraw to a .svg file, then  launching meerK40t and loading that file.

I just want to cut and I'm ok with cutting everything that was in the svg wrt color, line width, etc.  If I have to make the lines hairline width and a certain color, that's ok.

Can I use meerk40t CLI to automate the process of cutting what's in the SVG?  If so, I'd write a script to make it appear integrated to CorelDraw.  Similar to what's done in CorelLaser, but most options like  would (at least at first) be CLI commands hardcoded in the script.

I'd be happy to share what I come up with and enhance if there's sufficient interest.

tatarize, 2020-12-09:
Yeah, you load the file with the CLI `MeerK40t -zaS 10 <filename>` That might work. The z means no gui, the a means automatically start running this stuff, and S means set the speed to everything in the operation to 10mm/s which is about cutting speed.

If you could write a script from CorelDraw that would save the svg as a temp file and run that as a thing, in theory you'd never have to see MeerK40t at all. I know this *could* be done from Inkscape. I'm not sure how easy corel script it.

The line widths don't matter unless you're rastering them. Colors don't really matter either. In this case the file loaded with the -a does a classify on them and they'll all end up as an engrave that will go at 10mm/s which is cutting speed. You won't get the cut inner first optimization option though.

Mostly this is done with the -a flag though I think #267 has similar intent. Also, #226 tends to try to map this out a bit. Basically it *should* be there. -S and -a should be enough to start it and -z to keep the gui away. If using the straight windows compile there's an issue with #231 but that's a compile thing, that I'd need to work around. The options still work but the information feedback is lacking since it's a window mode thing. You could run MeerK40t from the pure python if that's a sticking point.

tatarize, 2020-12-09:
Also, strictly speaking the SVGs load up all the relevant data. So if you know CorelDraw scripting and well, you could actually, rather than just -S to set the speed, have as fine grain of control as you want over the particular operations. If they are defined in the svg file, they'd be loaded up and used. So you could if you wanted to get fancy have the CorelDraw able to set the speed of individual parts of what you're making and just sending that MeerK40t working as a backend.

This is however entirely within the realm of what MeerK40t should do so if you hit a snag ask, and if it's not possible, I'll make it possible.  I actually love the idea of just writing a front-end script for CorelDraw or Inkscape and defining your objects and letting you do all the designing there and sending it to the laser. I'm going to be doing much larger expansions of the Console in 0.7.x, so even if it's a sort of fancy procedure there's a CLI command called --batch which would, in theory, let you issue a bunch of console commands saved in a flat text file.  So there are a lot of options here if the current slate of options doesn't have you covered.

In theory I could see needing some of the features from Project Foresight (Completely unfinished and being truncated in 0.7.0 for the sake of finishing the version), to do a preview of thing your designing. There might be some issues just running the code blind. Since you're not entirely sure how it would turn out if you ran the job. In theory sometime in the future there should be preview the job with respect to the time/order of the project. But, that's a later problem.

davthomaspilot, 2020-12-09:
Thanks for the replies!
Oooh, finer control over the speed control sounds interesting.  Got some ideas about that. 
How can I find out what I can put in the SVG file and how meerk40t will use them?
I'll give it a try once I get the project I'm working on done--probably later this week.
Dave

tatarize, 2020-12-11:
The best method would be just to save a blank svg with some operations loaded. In theory it should be within the namespace information, so: https://github.com/meerk40t/meerk40t/wiki/Namespace

closed 2022-07-20 by tatarize