meerk40t/meerk40t #273: Centerize DXF files.

tatarize, 2020-12-09:
Apparently most programs loading dxf files center them. This isn't a very hard process but if that's what is expected that is what should be done. Unlike svg there's not clear specs on all aspects so going with what people tend to do seems the best plan.

Also, add an option to disable this, but opt for True to start.

Xgl0baltk, 2020-12-17:
just found out about MeerK40t, so don't know how it is handled with feature comparisons, but Whisperer also adds the option to use the Origin that is specified in the DXF file, can't say though how it is defined or other specs

tatarize, 2020-12-17:
I think part of the issue is that it uses the origin and in some cases it's really really far away from where you'd want it, which in a couple test files put the data really really far away. I'd have to do more research on DXF to make this stuff more consistent, so far I've just been fixing things as they are brought to my attention.

tatarize, 2021-02-25:
Added in 0.6.20. Windows -> Settings -> Centerize DXF.

closed 2021-02-25 by tatarize