meerk40t/meerk40t #280: Reorder

meeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee, 2020-12-24:
Not sure if this has been done but is there a feature for reordering a jobs individual components. EG. Tell it to engrave first then cut then do a raster... Also, can you merge 2 different engrave jobs, I simulated a design with an image to engrave and text to engrave (same settings but did the image completely then went back to the start to do the text)

Thanks

joerlane, 2020-12-24:
Yes. That feature already exists to reorder individual jobs or components. Just drag the operation (Raster, Image, Engrave, or Cut) up or down. Might make it easier if you collapse the drop-down triangles as you do those drag moves.

Not really sure what you mean about merging 2 different engrave jobs. You have have more than one engrave job in the same file. They can be made to run back to back which is exactly how it would function if you combined them. Alternatively; you could simply set those lines to the same color and they would open as a "combined" object when the file is loaded in MeerK40t.

Or did you mean load two files at the same time? That does work as well.

meeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee, 2020-12-24:
Hi, thanks for fast reply. I'm still getting used to this software, I've
used another piece for years so have some habits from that one... I've just
worked out the reordering. I was trying to do it from the job list pop up
box. Thanks
Also, I've just worked out what I was doing wrong with the jobs not doing
all of the details at once. Sorry for mithering. I learn best by messing
with stuff but after 2 hours gave up 😅😅 thanks for your help

tatarize, 2020-12-24:
It's not a bad idea though, if the settings between the two engrave operations are the same I don't see any trouble with moving everything from the one operation into the other operation. That stuff does, at some point, need an upgrade especially with regards to grouping objects together and moving multiple objects at the same time, or groups at the same time.  It wouldn't be too weird to do it that way. Or make trying to combine them like that create an arbitrary group with the operations as the leaf-nodes. I mean it would be into the 0.7.x series but certainly there's a few actions you could see with dragging operations into other operations that would make some logical sense.

meeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee, 2020-12-24:
😀 glad I've got you thinking 🤔 😂

closed 2021-05-13 by tatarize