meerk40t/meerk40t #281: step adjust

calderaro24, 2020-12-26:
hi, Is there the possibility to adjust the steps?
in Whisperer there is a "scale factor" setting.
Thanks

tatarize, 2020-12-26:
There's kinda two ways to do this, neither one is really solid. You can set the scaling in Rotary Settings and leave the settings on with the steps correctly adjusted. Then you can also hit alt+F3 and control+F3 which will adjust to the rotary scaling. (make sure to set the rotary off, with the correct scaling for that one).

tatarize, 2020-12-26:
Also see #253

mario4395, 2020-12-29:
Is there a way to set the steps per rotation value?
The textfield in the rotary settings is disabled.

calderaro24, 2020-12-29:
Hi, I tried to adjust the steps in the rotary menu, it doesn't work.
I checked, unchecked but nothing. I have a 1.8 degree stepper motor, whisperer works, meerk no ... I prefer meerk but I don't know how to use it ...help?

tatarize, 2020-12-29:
You can. Click rotary. Adjust those values. And turn rotary back off. The values still saved and are used for the rotaryview. It was the quick and dirty way to do that while 0.7.0 is still kinda screwy. 

Hitting Alt+F3 and Control+F3 will then trigger the view changes as set in there.

https://youtu.be/viCVItFcXi4?t=483

It uses the settings in rotary, even with rotary off. It still saves the scale_x and scale_y values.

calderaro24, 2020-12-29:
The rotary trick works if you have a job loaded. If you move the laser head with the "navigation" menu it does not accept the set values: if I move by 100mm, the head moves by 200mm ...
Do you think there is this change in the next versions?

tatarize, 2020-12-29:
Hm. That is a fantastic point. Yeah, moving to specific locations would absolutely fail because I'm not adjusting the tick-space correctly. It should properly and correctly warp a lot of those operations. I am trying to finish up 0.7.0 now. I think I have most things no longer crashing. And tweaks and corrections in 0.7.0 are a lot easier to make. Though I think correcting this might require a few different steps.

- [ ] Conversion away from Scale Y, Scale X in Rotary for something that is properly step-adjust. This should likely be added into Preferences.
- [ ] Change/Correct Navigation to move with regards to the scaling of the space. The `Jog distance 394 = 10.0075mm = 1.00076cm = 0.394in` information is wrong if we're saying a step in the y-direction is 1.1 mils rather than 1mil. The distance of 1 step is not 1mil but 1.1mil.
- [ ] Any returned tracking information is taken as units and would need to be converted to mils to be drawn in the scene.
- [ ] Shift home position, should give the position in units rather than mils.
- [ ] Most units should be given in steps rather than mils. Dot Length, Overscan, Raster Step. Since a mil is unit of actual distance and a step could be a completely different size accordingly.

Some objects like raster images where each pixel equals each one step would need to be automatically stretched with the deformation but viewed correctly. So if they were 10cm x 8cm but the y: 1step -> 2mils it would have 10cm  x 4cm worth of pixels and need to be stretched, accordingly.

As an alternative, the images could be adjusted correctly. 1 pixel = 1 step in images. If you have a deformed space it would still display the image as the image would actually exist. And give you options to transform it with the rotary scale accordingly. 

While I could easily modify the vectors to say that the vectorspace's final real units are not mils but steps and so those could be sent normally rasters would be treated differently with each pixel being equal to 1 step. Which unit-wise would be fairly correct. You would then see everything in the scene as displayed in the correct units but rasters would scrunch according to their pixels (which would then need correcting in a variety of ways).

Another oddity is the rotation of objects. If we have stretched views and scenes, we end up with objects that cannot rotate because of the non-uniform space. Rotating an object stored in steps would require stretching and scrunching it while it's rotating. But, if we treat the scene view as mils, we could have our uniform space, which would be non-uniform tickspace.

I dunno about next version. I'm kinda committed to a feature freeze and getting 0.7.0 up to snuff and working. It seems like a few choice changes could almost do this and then deal with some of the hairier details later. Basically define the scene native units as mils and keep the door open to the fact that the devices native units are steps, and then define the conversion.

tatarize, 2020-12-29:
I think the big issue is that rasters to work well natively have a pixel size of 1 step. This can be converted prior to actualization where we resample all the pixels. But, after resampling rather than viewed correctly they would be viewed scaled, but they would also be resampled to have more or fewer pixels according to the size of the step-adjust. This is kinda what's done with step-size, in fact it might be able to be folded into that kinda of stuff. Because declaring a raster-step to be 2 or 3 is really just declaring the size of the pixels to be 2 mils x 2 mils or 3 mils x 3 mils.

calderaro24, 2020-12-29:
I understood less than half of what you explained, but I think you are right .. :-)
I think the "scale factor" option should be integrated, perhaps with version .71 or later. It is also important to correct elongated belts, pulleys with different teeth and motors with different pitches.
Thank you for the time you spend improving the software.

closed 2022-07-20 by tatarize