meerk40t/meerk40t #283: New User Thoughts, issues and requests

inspectionsbybob, 2020-12-31:
Ok, I am a K40Whisper user, or Was... saw Meerk40T demo on YouTube and was stunned

Downloaded it and TRIED to use it...

NO documentation in the HELP section
No Tutorials, I re-re-re-reviewed the YouTube demo and workflow videos and got it running, sort-of.

(Setup is Win10 and a K40 with an analog power dial/meter)

The Drag in an Image, set the size, set it to dither, then laser was powerful... But incomplete.  What to set the K40 at?
Coming from the K40Whisper my power is set to barely leave a mark... so where do I set it so the dithering works?

You DESPERATELY need a "Quick setup guide"  that walks you through the common scenarios, like your workflow video, but you need to include the K40 setup as well... where to set the power is paramount.  Next is how to set the HOME location. I figured it out, but it would have been nice to have a "How to set the HOME location" tutorial.

Ok... CAMERA is a killer feature.  Something in Github mentioned a "Camera Wizard"  Where is it?  The version on my desktop doesn't even SEE my USB camera that I use multiple times a week for Zoom calls...  I couldn't play with it because it didn't see anything.     My Recommendation for setup is to Put a "DOT" of something on the top of the laser head, and then tell the head to move a "safe" distance, and track the dot.  The larger the movement, the more accurate. DONE.  Not having to create a special image and measure it.  A 200mm x 200mm movement is safe on almost every laser.  Start from home, find the dot, move to 200,200, find the dot. move to 0,200, 200,0, maybe 100,100... Find the dot each time... calibrate... you are DONE.  If you KNOW it works with certain USB cameras... please provide a list or URLs.  
The following camera is what LightBurn uses and can be found for UNDER $10
https://www.amazon.com/dp/B0869CB4DG/ref=cm_sw_em_r_mt_dp_FvK7FbYADDT45

The PPI is POWERFUL, but is VERY dependent on the power setting on the K40, as everything is a percentage of THAT.  Since it is easy to hurt the K40 by setting the power Too High, most of us are reluctant to set the power above what is needed for engraving at a good clip...  So you need a "quick setup guide" for JUST that or a wizard that creates a material profile.  Give the material a name, go to a location, and engrave a line with a scale, and slowly increase the PPI power until it marks, and then cuts.  Enter those numbers from the scale, keep it in the profile.  Then if you want to use that material again, set the K40 to the same as before, call up the material, and all the PPI stuff should be filled in  (ala Glowforge)

Sorry for the rant... I think your software has GREAT potential, but right now it is only for Hackers, not users...

Each MAJOR feature needs a mini-tutorial or a "Quick setup Guide" for people like me that do not play with things like this every day... OR are used to another program and have to UNLEARN certain behaviors. 

If there -ARE- help files, explanations, or Demos, somewhere online, please point me at them and disregard MOST of what I wrote.

tatarize, 2021-01-01:
> NO documentation in the HELP section

This is a fairly fair criticism. There's the couple video hit-or-miss answers that I make for specific responses. I recently wrote a rather weak manual:

https://github.com/meerk40t/meerk40t/wiki/MeerK40t-Manual

>No Tutorials, I re-re-re-reviewed the YouTube demo and workflow videos and got it running, sort-of.

Again fair. I spend most of my time writing software and very little of my time writing how the software works. So people ask some question and I can very often give a secret undocumented console command that will do that thing.

>The Drag in an Image, set the size, set it to dither, then laser was powerful... But incomplete. What to set the K40 at?
Coming from the K40Whisper my power is set to barely leave a mark... so where do I set it so the dithering works?

Much of this is that the step distance in MeerK40t is set to 1 since that seems like it's more proper. But, set to 2 in K40Whisperer. If you use RasterWizard like rather than dithering, you make it the right size then go to Apply Script -> Gravy. And it should generally just work.

> You DESPERATELY need a "Quick setup guide" that walks you through the common scenarios, like your workflow video, but you need to include the K40 setup as well... where to set the power is [...]

If you want to make a list of tutorials I'm missing I'd be happy to hammer them them all out. I don't really know where the missing sections of the knowledge base are. Everything seems perfectly reasonable to me and when something happens I know all the lines of code that made that happen so nothing seems surprising to me at all.

>Ok... CAMERA is a killer feature. Something in Github mentioned a "Camera Wizard" Where is it?

It's a idea going forward. The Github issues are things to address or discuss, not necessarily things that exist. Such a feature is needed because most of the stuff is kinda mysterious.

> The version on my desktop doesn't even SEE my USB camera that I use multiple times a week for Zoom calls... I couldn't play with it because it didn't see anything.

There's a beta version of 0.6.15 that tries to hammer the camera device. If that works for you and the normal version does not, please do report back. I have gotten a few such reports from Facebook and the materials I read suggested the issue is OpenCV may be returning no frame for the first several attempts while the camera is turning on. And MeerK40t was programed to give up if the first frame was blank. If that's the problem and the already implemented feature corrects it, I can fast track 0.6.15.

>My Recommendation for setup is to Put a "DOT" of something on the top of the laser head, and then tell the head to move a "safe" distance, and track the dot. The larger the movement, the more accurate. DONE.

The distance between camera and the laserhead vs. the distance between the material and the material being ablated is not the same. While the angle from one location could be determined with that method. But really that would give you the parallax and the distance there. It would also require that the camera be absolutely uniform, fisheye would be an issue, and that you could exactly extrapolate the location at the material. Also, this would at best give you the scale factors, It would not necessarily give you actual location. OpenCV does have the ability to track a green dot sort of thing, so you could somewhat tell where the laser currently is with the camera, but I already know that. I need to know, what the material that I will be shooting lasers at exactly is and have that lined up. Which isn't what that information actually solves for. Since it's unimportant to know the laser is shooting a location 2in over by 3in down, since I already know. What matters is getting the picture of the laser bed to exactly overlay the simulation of the laser bed  I have. And in that regard the laser head is just getting in the way.

It would however be entirely possible to solve the perspective if you could put a green dot at the exact corners of the bed. Then you get those locations and put that across your bed. But, it would *still* be needed to know the exact size of the bed. Which isn't something the angle between the dots could give you. 

>If you KNOW it works with certain USB cameras... please provide a list or URLs.

I know it works with a Logitech Camera that came out in 2000 that I've had kicking around for 20 years. And several other cameras in my house. It really *should* work with a lot of cameras but some folks have reported that they have cameras other applications see, but the software doesn't regard.

>So you need a "quick setup guide" for JUST that or a wizard that creates a material profile. Give the material a name, go to a location, and engrave a line with a scale, and slowly increase the PPI power until it marks, and then cuts.

Setting settings is certainly a thing I've looked into but a lot of lasers are different and asking people to do many hours of test cuts on their laser is something that should be made easier but not necessarily required. PPI also tends to be really really linear with regard to speed or power level of the laser. So if you set to 500 ppi, it's going to be half as powerful as 1000 ppi regardless of the other settings.

> Enter those numbers from the scale, keep it in the profile. Then if you want to use that material again, set the K40 to the same as before, call up the material, and all the PPI stuff should be fille [... cut ...]  spent more time time reverse engineering Moshiboard than telling folks how to move things around in the interface.

See issue #152. I'm happy to implement things suggested by people better at design or UX than I am, which is a lot of people.

I'm not great at communicating stuff to people. Like, you might be surprised to learn the WASD keys on your keyboard can totally control the laser really nicely. You'd maybe be surprised to learn that since I don't think I told anybody or said it anywhere.

inspectionsbybob, 2021-01-01:
Being an ex-coder (is there such a thing as an EX coder) I understand not
doing documentation...  I will try to come up with a list of things as I go
that need some form of explanation. Maybe you can pass that on to your
other contributors and get some help...

some just need a "path" to the proper page, like setting the "Home" location

"To set a new 'HOME' location it is : Windows>Preferences>Shift Home
Position.
   Note, both values are POSITIVE by default."

Might just want to create a Big (searchable) Note Pad with things like that
stuffed into it.

Another: "To run without being connected turn on 'Mock USB Connection'  at
Windows>Preferences>USB Settings : select 'Mock USB Connection Mode'.
MeerK40T will now behave as if it was connected to a laser and all the
Navigation commands will work among other things."

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
bob@inspectionsbybob.com
MD Lic#29666; ASHI Member #212016

tatarize, 2021-01-01:
I *DO* have a goodly amount of documentation, but it too is pushed towards hackers and such. A lot of internal documentation is trying to explain how MeerK40t works to somebody who might want to code for MeerK40t. And a lot of the external documentation is trying to explain how MeerK40t works towards somebody who might want to explain to other people how MeerK40t works in some instructional videos.

inspectionsbybob, 2021-01-04:
Closing to clean Open Issues of irrelevant or satisfactorily addressed items

closed 2021-01-04 by inspectionsbybob