meerk40t/meerk40t #284: No "Close"

inspectionsbybob, 2021-01-01:
Really simple omission.
There is no "CLOSE FILE" only EXIT...
If you want to abandon the current file, you have to EXIT the program, there is no "close" button or selection under FILE

Obviously, you may want to ask "save changes before exiting"

joerlane, 2021-01-01:
You can use "New" to work on something else without exiting the program. One thing to consider if Close were to be added; "what is closing"? You can have more than one file and/or image open at the same time in MeerK40t...

inspectionsbybob, 2021-01-01:
Hum... Thats works... doesn't open a NEW window as expected, it clears the workspace like close/open would...

That one was easy....

tatarize, 2021-01-01:
Hm. Yeah, I did opt for a weirder way of doing that.

It has some of the trappings of working on on a project, but doesn't really have any actual ability. You just have stuff in a window. You can add more stuff or delete the stuff or clear the stuff. But, it doesn't have the common CRUD interfacing stuff Create, Reload, Update, Delete, or proper finality to projects.

inspectionsbybob, 2021-01-01:
Happy new year...

I did notice that the initial "open" does not always start with a clean
workspace or stuff in the left side area... I now do a "new" as soon as I
open Meerk40t to make sure I am clean...

There is a lot of power in Meerk40t, but a lot of it seems to be for a
hacker not even the Hobby user. I would be happy to give up some of the
status & command line stuff for documentation and better error
management/messages.

I am a ex-coder pro-sumer... I know how I would do things if I had to.  My
forte' used to be proof of concept coding... I would make it work. It might
not scale, but I would prove it -was- possible...   So, I am impressed...

I need to work on my K40 more, air assist and a better bed this week... Usb
CAMERA is also on the list... and a hub and a and a ...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
bob@inspectionsbybob.com
MD Lic#29666; ASHI Member #212016

inspectionsbybob, 2021-01-01:
> No loss on the power settings and speeds when re-editing the .svg. Even
> the MeerK40t text notes stay in the svg through inkscape edits.

OK, I tried to send you a document, but Git Hub said no...

It included a Material Sample SVG and then a run on K40W vs MK... HUGE
Difference.

The following is the TEXT ONLY.

If you want the Photos and the sample file please tell me how to send them
to you ....

Here is -MY- attempt at the document I was talking about.
Spent the Night learning Inkscape and applying Gradients.
I know Gradients aren't implemented yet, but this is what I was looking for
in the future...

I ran the same SVG into K40W and into MK UNTOUCHED. Ie, load, and run
(2 steps on K40W, engrave and cut, 1 step on MK, execute)

The K40W run resulted in what I expected...

MK defaulted to CUTTING everything, so it will require a LOT of tweaking to
get it to Vector Engrave the Boxes, but it is a good start.

Also, not sure how, but the POWER from MK was a LOT higher than when using
K40W. I had set the current (via test button) to be very low, ie 2-3ma,
Meter barely moved while running, but when MK ran the current was VERY high
and the meter was moving into the 8-10ma range. Not anything like what I
expected. I did not touch anything between runs except to turn the material
to expose a clean location.

I thought the MAX current was set from the front panel and then the PPI was
a percentage of that?

This was an Interesting side-by-side.  MK may be powerful but it looks like
it needs a LOT more 'pre-press' work than K40W to take a supposedly
"press-ready" Inkscape SVG and run with it.  The run on K40W was exactly as
expected.

Bob Sisson

tatarize, 2021-01-01:
Max current is set from the front panel. PPI is per inch pulses. So 1000 is on always 500 is off every other pulse. Etc.

inspectionsbybob, 2021-01-04:
Solution/Explanation is acceptable

closed 2021-01-04 by inspectionsbybob