meerk40t/meerk40t #285: Document Request: MeerK40T specific Laser Material Template Sheet

inspectionsbybob, 2021-01-01:
I have been bouncing between Inkscape, K40Whisper and Meerk40T 

I, and I suspect many others need a VERSION of the Laser Material Template Sheet like
https://www.thingiverse.com/thing:2243854

BUT redone for the PPI... With circles or Boxes. It should be a relatively SMALL layout, so the sample can be rotated 90deg and re-run as necessary.
Start with 0% power and Finish with 100% power (of the maximum selected on the front panel)

So that with a given material, at a given selected power, a particular PPI will do nothing, engrave, or CUT something.

Usage:  Choose a material, dial in a safe "guestimate" of the Max power needed, and run the File.
The laser will begin drawing "nothing" and as the run progresses the text will appear, get darker, darker, and depending on the material and thickness, eventually burn through.

Depending on How fancy someone wants to get, you could Do Multiple other variables at the same time, as Meerk40T allows that.  Adjust Speed and Power on a Grid, speed on one axis, power on the other, One corner SHOULD be blank, while the other corner should be cut through. Where speed and power cross and what it looks like is what people are interested in. 

We All Know what 4,6, 8, 10... point fonts look like, and 2, 4, 6, 8mm circles and boxes as well.

What we want to know is if I choose 1/4" White Birch, and Set my Power to 10ma (or some repeatable setting), what PPI do I need for ENGRAVE, and what PPI will CUT.  

Something Like
300mm/sec (or start higher)
10% 20% 30% 40% 50% 60% 70% 80% 90% 100%  (each number at a different PPI)
200mm/sec 
10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
100mm/sec
10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
50mm/sec
10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
and so on....

300mm/sec at 10% should be invisible, somewhere in the 70%-80% range might look good for engraving....
As the speed goes down, engraving should appear at lower and lower power until it starts to CUT...

THIS Laser Material test run would be REALLY useful and would also highlight the power and flexibility of MeerK40T

Someone could have a lot of fun building this template... but it is beyond my Inkscape and Meerk40t skills at this time.

joerlane, 2021-01-01:
That is kind of done for PPI in that template. Those % greys are generated with PPI.

I personally use a slightly different material test. I cut a sort of barcode in the edge of my sheet with tuned speeds. Based on a specific power level that I set with a volt meter (and save in the MeerK40t note); I cut small rectangles off the edge of my stock in small intervals like 8.0mm/s, 8.1mm/s, 8.25mm/s between about 7-14mm/s. This all depends on the power I'm using to cut a specific material though. Some materials won't cut with a single high power pass. Others I run the lowest power I can cut through single pass with at a reasonable speed.

It is possible to setup a template like that with multiple colors and could be set to cut vector shapes at different PPI settings. It's just not something I personally do though. And that's just how I use MeerK40t to perform those actions. There are certainly other approaches to each of those tests, and settings choices.

inspectionsbybob, 2021-01-01:
Unfortunately, when I bring THAT template into either K40W or Meerk40T it comes in as a big black BLOB...  When I try to edit it in  Inkscape it behaves like it was imported as an Image, I 'select something to get rid of the Fill...
That's what I was trying to do today, run a sample thin chipboard and see what power/speed would engrave and what would cut reliably.  With something like what I described I could follow the diagonal line along the speed/power curve  so I could run faster.  Currenly that takes (too) many aborted runs to dial in a (new) material.  To me everything is new as we just got this machine and are trying to get running so we don't have to go across town to run 1 sample on the community GlowForge every time. Trying to learn Meerk40T at the same time is, uhm, interesting.

joerlane, 2021-01-01:
Ok. I'll try to fix that file up for MeerK40t. It's not complicated; and one I have seen in the past. 

I tried it raw; and sure enough it crashed my head before I realized there was a giant black rectangle in the middle of nowhere I needed to delete.

inspectionsbybob, 2021-01-01:
I just looked at Inkscape, again, open the online manual and looked at gradiants.
If I knew what I was doing more, I could probably create a document that had text in a gradient from invisible to a full color
I could choose an outline font rather than a raster font 
But then I need to assign each line a different speed... and THAT I don't know how to do from Inkscape, I always did that globally in K40Whisper as I was setting the run up.  Black was engraving at one speed, Red was cutting at another speed. No option for changing speeds, or power within a run.

If there were more example files, videos, or Help files I could probably sus it out... but....

Can you tell me how I would change speeds vs power from Inkscape?
As I said, I figured out how to do a simple gradient from Red to Black, so I can probably figure out how to fade from white to red, or any other color, and that would be the PPI control... (I just need to know what to fade from and to)
What do I need to enter/change from Inkscape to adjust the Speed?  If I have to do that in Meerk40T, do I lose those edits when I take the plot back into Inkscape for edits?

joerlane, 2021-01-01:
The quick way to use that file is to open it in MeerK40t and zoom out if you need to. I found the single black rectangle up above the work area. Just select it; and press the delete key.

I tried to save a "fixed" version from both Inkscape and MeerK40t and both are broken. I'll pass this on as there appears to be at least a text bug that might be MeerK40t related. 

It came out fine on my test cardboard at very low power after deleting that one rectangle.

joerlane, 2021-01-01:
So; in regards to changing speeds and powers in a single file. I simply use inkscape and change the colors of the objects. Red is my regular cut, blue is my regular engrave; but any other color of the rainbow will show up in MeerK40t as a different operation. I then adjust settings for speed, PPI, etc. for each of those other colors.

Then you can save the file again from MeerK40t as an SVG and it will retain those laser related settings.

joerlane, 2021-01-01:
Gradients aren't working last I checked. You can convert them to bitmaps in inkscape, which will work in MeerK40t... That's a known missing feature.

joerlane, 2021-01-01:
No loss on the power settings and speeds when re-editing the .svg. Even the MeerK40t text notes stay in the svg through inkscape edits.

inspectionsbybob, 2021-01-01:
Wow, its New Years Eve and you are still at it, well its New Years now...
I am impressed, the Camera started working when I went in from another
Direction, grabbed an image of me, and played with the Wizard... Nice.

No Gradients, for now, but I can fool that by using a myriad of colors to
set PPI and Speed, that's going to be a pain to build, but doable

VERY happy that MeerK40t edits are saved through Inkscape....  I will build
a Table of Speeds and PPI and put a Symbol in each one...

I will only have to build that table ONCE (hopefully) and then run it
against different materials and build a Library.

I will edit tonight, and run the laser tomorrow some more....

Happy New Year
Please don't wait for me to respond when booking an inspection. I turn off
my devices when on appointments and someone else could get your spot.
Please CALL our service or book yourself online to save the spot as soon as
you know the time you want.

Bob Sisson ACI, BVI
Inspections by Bob
ASHI Member # 212016
MD Home inspectors License #29666*
*MAC-ASHI Chapter President 2010-2012
ASHI Mid-Atlantic Group Leader 2009-2012
ASHI National Board Of Directors 2012-2015, 2017,2018-2020
Bob@Inspectionsbybob.com
www.inspectionsbybob.com
(301) 208-8289 Scheduling/Pricing

inspectionsbybob, 2021-01-01:
Here is -MY- attempt at the document I was talking about.
[Boxes for test.zip](https://github.com/meerk40t/meerk40t/files/5759243/Boxes.for.test.zip)

Spent the Night learning Inkscape and applying Gradients.
I know Gradients aren't implemented yet, but this is what I was looking for in the future...

I ran the same SVG into K40W and into MK UNTOUCHED. Ie, load, and run
(2 steps on K40W, engrave and cut, 1 step on MK, execute)

IMG_6527.JPG
The K40W run resulted in what I expected...
IMG_6528.JPG

MK defaulted to CUTTING everything, so it will require a LOT of tweaking to get it to Vector Engrave the Boxes, but it is a good start.

Also, not sure how, but the POWER from MK was a LOT higher than when using K40W. I had set the current (via test button) to be very low, ie 2-3ma, Meter barely moved while running, but when MK ran the current was VERY high and the meter was moving into the 8-10ma range. Not anything like what I expected. I did not touch anything between runs except to turn the material to expose a clean location.

I thought the MAX current was set from the front panel and then the PPI was a percentage of that? 

This was an Interesting side-by-side.  MK may be powerful but it looks like it needs a LOT more 'pre-press' work than K40W to take a supposedly "press-ready" Inkscape SVG and run with it.  The run on K40W was exactly as expected.  

[image]
[image]

inspectionsbybob, 2021-01-01:
PS... I didn't let the MK run finish as it was clear what it was doing WASN'T what was expected.
The Two images above were reversed.  The One where "Material Test Grid" is readable was from the K40W

tatarize, 2021-01-01:
Max current is set from the front panel. PPI is per inch pulses. So 1000 is on always 500 is off every other pulse. Etc.

Hm. It seems like a default even built in meerk40t document thing to test materials is quite doable. It would just need a series of tests and then rather than get fancy I'd add a command to load it up. Though some test files do really interesting things like testing how close you'd need to be to make a living hinge sorts of things. Also, it seems like this sort of highly specific work would be best compiled largely as cutcode with basic primitives, since I sort of wouldn't want changes to affect how the template works.

Obviously it would need to run at different speeds to see the difference in cut depth. And some different PPIs. There's some new functionality in 0.7.0 to basically do dash lines. I could see needing such a thing to properly cut leather lace holes or something. Maybe you'd want some tests for that. Though a special SVG with the each item being a different color and getting classed into a set of operations would be the easiest and likely the best method there. Since building it into the program itself might be a bit meh. Though it would also be really nice if the operations themselves could be turned off for the output. So you'd have each test classed in an operation and could turn off the hinge test if you're running acrylic or whatever.

inspectionsbybob, 2021-01-01:
That was my intent, was to create a sample document that would help you TUNE your laser for a particular material.

Once the Outline is created at some relatively "safe" values of engraving, Each line was designed to print a "star" at various speed and power levels, starting a FAST, typically used for engraving, but how much power do you need to engrave nicely at 400mm/sec... 10%, 40% or 60%?  
Moving down the chart, it slows down incrementally...
At 20mm/Sec how much power do I need to cut that star cleanly without burning?

Once this SVG  is built, and tweaked, take ANY material, dial in a Guestinate on front panel power, and Execute.

In about 8min you will have your numbers for engraving and cutting that you can use to optimize your job.

If Even the 400mm/sec line is toasted even at 30% you know you need to cut your power a lot and re-run
if 20mm/sec  line doesn't cut through cleanly by 60%, you know you need to add more power (or possibly another pass)

Put ANYTHING on the bed, make sure the file is aligned with the material (camera), and execute...  Leather, Chipboard, Lexan, rubber....

The Job needs to be re-ordered, so the Cut-out is last, the Material Test Grid Text and it's Bounding box should be first.  If you can't see that, or it is BURNT, you know you need to adjust up/down and re-run within a minute.

inspectionsbybob, 2021-01-02:
OK, I got -A- version of the demo file running.
Because of reasons, it was done on K40whisper, sorry, but it makes the point.
The Material test grid, Material, Power, and Software boxes were done at the 400mm rate.
Each next line was done progressively slower until it made no sense to continue, the 20mm/sec and 10mm/sec were useless
This was done at 2-3ma, really low power, so the outline CUT took multiple passes.
But this tells you that 100-200mm/sec at minimum power looks good. 400mm needs more power
Unfortunately, I couldn't change the power within a line... that is what I need help with... (RASTER did some gradient)

[image]

I would love to see this file optimized for MeerK40T.
On K40Whipter it is 8-files and up to 9 runs, ouch...

inspectionsbybob, 2021-01-04:
Updated file with "Software" Box added.
Not a Developer issue, more a user submission.
Closing to clean Open Issues of irrelevant or satisfactorily addressed items

[Boxes for test.zip](https://github.com/meerk40t/meerk40t/files/5765887/Boxes.for.test.zip)

inspectionsbybob, 2021-01-04:
Updated file with "Software" Box added.
Not a Developer issue, more a user submission.
Closing to clean Open Issues of irrelevant or satisfactorily addressed items

[Boxes for test.zip](https://github.com/meerk40t/meerk40t/files/5765890/Boxes.for.test.zip)

closed 2021-01-04 by inspectionsbybob