meerk40t/meerk40t #286: Documentation Needed for CAMERA section

inspectionsbybob, 2021-01-01:
While I wait for my Laser to finish...

Camera section Documentation Wanted/Needed.  (Stream of consciousness questions)

The Camera assumes what?
Locally connected USB OTG UVC compliant type camera? Examples of known to work?
I have the following on order and will be mounting it on the Lid somehow....
https://www.amazon.com/dp/B08MTBYGP5/ref=cm_sw_em_r_mt_dp_YX47FbFY2M3A6?_encoding=UTF8&psc=1
Where do you find the 6x9 Grid Document page?
How do you "calibrate" the camera  (steps, please)
How do you connect to a camera if it says "Camera not found" ( 'Windows>Camera' seems to work **MUCH** better than 'Camera' for startup but is still not 100%)
"Update Image" does what? Copies image to the workspace, How do you CLEAR the image? NEW doesn't. The image copied onto the workspace doesn't show up on anything.  Is this typically the Image of the laser Workspace that will be used for alignment?  If so, please say that.  I still need a way to "remove" it... (Hide Background?? if so, please say "to remove...")
"Export Snapshot"  Takes the currently selected camera and copies the snapshot onto the workspace.
"Correct Fisheye"    How, what does it need to do that (the 6x9 image placed on the Laser Bed?)
"Correct Perspective"  How, what does it need to do that (the 6x9 image placed on the Laser Bed?)
"Slider"  What does it do?  What does it need to do that?
"Detect Distortions/Calibrations"   What does it do? What does it need to do that (the 6x9 image placed on the Laser Bed?)
Top "Camera" menu:
       Reset Perspective - Why would you use this?
       Reset Fisheye - Why would you use this?
       **NEED** " Remove Workspace Image"
       **NEED** "Flip Image V, Flip Image H"
       Set IP camera 1 - Why would I use this.  Needs to be what type of camera. What about camera logins?
       Set Camera 0 - Why would I use this.  Needs to be what type of camera (locally connected USB??)
           (Why does one start at 0 and the other at 1?)
           (Would it be better to Say **Network** Camera #1 and **Local** Camera #2, etc...)
       Is "Calibrate" another word for "align"
       Are there other "tricks" that would be helpful in calibrating/aligning the camera?  

If the answers to these questions are put on a "Help" page it would -help- a LOT...

Hope this Helps...

tatarize, 2021-01-03:
> Locally connected USB OTG UVC compliant type camera? Examples of known to work?

Also, webcam rtsp and http connections work.

> I have the following on order and will be mounting it on the Lid somehow....

There's no documentation recommending any mounting location or how to mount it.

> Where do you find the 6x9 Grid Document page?

[image]

> How do you "calibrate" the camera (steps, please)

* Load if your camera has fisheye, then print the checkerboard pattern.
* Use checkerboard pattern firmly glued to a flat surface. The paper can't be bent or curved.
* Use the detect checkerboard button to teach it to get rid of the fisheye.
* Use a standard piece of A4 or any standard sized object in your bed.
* Set your bed dimensions to the size of that object.
* In CameraInterface, move the perspective marks, zooming and panning in the camera view as needed, to mark the corners of the standard sized object you have in there.
* When you've marked the corners you have correct perspective and correct fish eye checked. You're done. It will be pretty close.

> How do you connect to a camera if it says "Camera not found" ( 'Windows>Camera' seems to work MUCH better than 'Camera' for startup but is still not 100%)

Try the beta. 0.6.15 beta-4 or so. This will try hammering the camera. Some windows cameras don't produce a frame at the very start and are wrongly truncated.

> "Update Image" does what? Copies image to the workspace,

It pastes the currently marked image to be the marked bed location.

> How do you CLEAR the image?

You can't.

> NEW doesn't.

New doesn't.

> The image copied onto the workspace doesn't show up on anything.

It's supposed to be below everything and is drawn as such. So you can position things on to whatever material you have placed in your bed.

> Is this typically the Image of the laser Workspace that will be used for alignment? If so, please say that. I still need a way to "remove" it... (Hide Background?? if so, please say "to remove...")

This is actually kind of typical. But, it's not used for alignment this is used for positioning an already aligned object.

> "Export Snapshot" Takes the currently selected camera and copies the snapshot onto the workspace.

That is correct.

> "Correct Fisheye" How, what does it need to do that (the 6x9 image placed on the Laser Bed?)

In several different spots, without any the paper being curved or bent. It trains the screen to undo the fisheye perspective.

> "Correct Perspective" How, what does it need to do that (the 6x9 image placed on the Laser Bed?)

It doesn't care about the checkboard at all. If you zoom out there are perspective markers. These are set at the corners of the bed. Though a standard candle would be preferrable. Though you could also laser holes in a postit note in the corner and mark those positions.

> "Slider" What does it do? What does it need to do that?

The slider sets the frames per second. It does that because sometimes you might not care if the camera updates every few second or is realtime.

> "Detect Distortions/Calibrations" What does it do? What does it need to do that (the 6x9 image placed on the Laser Bed?)

Yes. That requires the 6x9 checkerboard.  In view of the camera, and flat. It doesn't really need to be a particular size and it's for undoing fisheye perspective.

> Reset Perspective - Why would you use this?

Because you moved your perspective markers somewhere weird and figure starting over is better.

[... 16 lines cut ...]
> Set Camera 0 - Why would I use this. Needs to be what type of camera (locally connected USB??)

Locally connected USB camera, or built in laptop camera.

> (Why does one start at 0 and the other at 1?)

Because the 0 at the camera is the local OpenCV code whereas I chose the indexing for the different RTSP/HTTP stream IP cameras and started counting it at 1.

> (Would it be better to Say Network Camera #1 and Local Camera #2, etc...)

Yes. That would make better sense with regard to UI and interfacing. Though I will be likely shifting these slightly in the 0.7.x camera upgrade but those are much more cogent and better methods of addressing that.

> Is "Calibrate" another word for "align"

No. Calibrate refers to adjusting the fisheye correction whereas align means actually getting all aspects correct, bed width, perspective, and fisheye.

> Are there other "tricks" that would be helpful in calibrating/aligning the camera?

Yes. You can use a standard sized object and set your bed size to the size of that object. Then set the corners to that object. Or if you want your entire bed working and in view. You would place post it notes in your corners. Then add tiny circles or a full rectangle to outline your bed. In console these commands would be `circle 100% 0% 1mm` to make a circle at the upper right hand corner. Then go ahead and cut that in a postit note. Or `move 100% 0%` would take you to the same corner and just hit your test button with a post-it note in that corner. Then mark each of your corners with perspective markers. Then you'll have it correctly lined up. 

> If the answers to these questions are put on a "Help" page it would -help- a LOT...

Hm. I suppose.

https://github.com/meerk40t/meerk40t/wiki/HELP:-Setting-Up-Your-Camera.

inspectionsbybob, 2021-01-04:
OK, I give up, how do I download the Beta versions?  I can't find a windows
EXE only source code (lots of code)

I was going to play with Beta4 to see if the Camera came online any
faster/better...

Please don't wait for me to respond when booking an inspection. I turn off
my devices when on appointments and someone else could get your spot.
Please CALL our service or book yourself online to save the spot as soon as
you know the time you want.

Bob Sisson ACI, BVI
Inspections by Bob
ASHI Member # 212016
MD Home inspectors License #29666*
*MAC-ASHI Chapter President 2010-2012
ASHI Mid-Atlantic Group Leader 2009-2012
ASHI National Board Of Directors 2012-2015, 2017,2018-2020
Bob@Inspectionsbybob.com
www.inspectionsbybob.com
(301) 208-8289 Scheduling/Pricing

tatarize, 2021-01-04:
It's under releases. It's not a major real release so it shows up a bit differently:

https://github.com/meerk40t/meerk40t/releases/tag/0.6.15-beta4

MiBu-dev, 2022-03-04:
Hi 
I bought camera module  similar (in dimensions ) to [inspectionsbybob](https://github.com/inspectionsbybob
although I had choose wider lens :)  not FOV60 but FOV160 degree 
https://pl.aliexpress.com/item/1005003279752689.html?gatewayAdapt=glo2pol&spm=a2g0o.order_detail.0.0.3b1143ceMuH8Dk

there is slight  squeak - default camera res is 640x480 or so - in Cheese  I can set up up to 2Mpix - lsusb -s Id:Dev also list all avaialble modes - but in MeerK40t  GUI (linux) there is no  way to set res :(
I have to make also pivoted mount  as camera glued  to  lid cannot see all workspace (near edge) 
[image]
)

tatarize, 2022-03-10:
There actually may be a way to set res. I didn't fiddle much with it but there is a `camera set` command and `camera list` command.

```
[17:55:23] camera list
[17:55:23]     0: CAP_PROP_POS_MSEC -- -1.0
[17:55:23]     1: CAP_PROP_POS_FRAMES -- -1.0
[17:55:23]     2: CAP_PROP_POS_AVI_RATIO -- -1.0
[17:55:23]     3: CAP_PROP_FRAME_WIDTH -- 640.0
[17:55:23]     4: CAP_PROP_FRAME_HEIGHT -- 480.0
[17:55:23]     5: CAP_PROP_FPS -- 0.0
[17:55:23]     6: CAP_PROP_FOURCC -- 844715353.0
[17:55:23]     7: CAP_PROP_FRAME_COUNT -- -1.0
[17:55:24]     8: CAP_PROP_FORMAT -- -1.0
[17:55:24]     9: CAP_PROP_MODE -- -1.0
[17:55:24]     10: CAP_PROP_BRIGHTNESS -- 128.0
[17:55:24]     11: CAP_PROP_CONTRAST -- -1.0
[17:55:24]     12: CAP_PROP_SATURATION -- -1.0
[17:55:24]     13: CAP_PROP_HUE -- -1.0
[17:55:24]     14: CAP_PROP_GAIN -- -1.0
[17:55:24]     15: CAP_PROP_EXPOSURE -- -4.0
[17:55:24]     16: CAP_PROP_CONVERT_RGB -- 1.0
[17:55:24]     17: CAP_PROP_WHITE_BALANCE_BLUE_U -- -1.0
[17:55:24]     18: CAP_PROP_RECTIFICATION -- -1.0
[17:55:24]     19: CAP_PROP_MONOCHROME -- -1.0
[17:55:24]     20: CAP_PROP_SHARPNESS -- -1.0
[17:55:24]     21: CAP_PROP_AUTO_EXPOSURE -- -1.0
[17:55:24]     22: CAP_PROP_GAMMA -- -1.0
[17:55:24]     23: CAP_PROP_TEMPERATURE -- -1.0
[17:55:24]     24: CAP_PROP_TRIGGER -- -1.0
[17:55:24]     25: CAP_PROP_TRIGGER_DELAY -- -1.0
[17:55:24]     26: CAP_PROP_WHITE_BALANCE_RED_V -- -1.0
[17:55:24]     27: CAP_PROP_ZOOM -- -1.0
[17:55:24]     28: CAP_PROP_FOCUS -- -1.0
[17:55:24]     29: CAP_PROP_GUID -- -1.0
[17:55:24]     30: CAP_PROP_ISO_SPEED -- -1.0
[17:55:24]     31: CAP_PROP_DC1394_MAX -- -1.0
[17:55:24]     32: CAP_PROP_BACKLIGHT -- -1.0
[17:55:24]     33: CAP_PROP_PAN -- -1.0
[17:55:24]     34: CAP_PROP_TILT -- -1.0
[17:55:24]     35: CAP_PROP_ROLL -- -1.0
[17:55:24]     36: CAP_PROP_IRIS -- -1.0
[17:55:24]     37: CAP_PROP_SETTINGS -- 0.0
[17:55:24]     38: CAP_PROP_BUFFERSIZE -- -1.0
[17:55:24]     39: CAP_PROP_AUTOFOCUS -- -1.0
[17:55:24]     40: CAP_PROP_SAR_NUM -- -1.0
[17:55:24]     41: CAP_PROP_SAR_DEN -- -1.0
[17:55:24]     42: CAP_PROP_BACKEND -- 700.0
[17:55:24]     43: CAP_PROP_CHANNEL -- 0.0
[17:55:24]     44: CAP_PROP_AUTO_WB -- -1.0
[17:55:24]     45: CAP_PROP_WB_TEMPERATURE -- -1.0
[17:55:24]     46: CAP_PROP_CODEC_PIXEL_FORMAT -- -1.0
[17:55:24]     47: CAP_PROP_BITRATE -- -1.0
[17:55:24]     48: CAP_PROP_ORIENTATION_META -- -1.0
[17:55:24]     49: CAP_PROP_ORIENTATION_AUTO -- -1.0
[17:55:24]     50: CAP_PROP_HW_ACCELERATION -- -1.0
[17:55:24]     51: CAP_PROP_HW_DEVICE -- -1.0
[17:55:24]     52: CAP_PROP_HW_ACCELERATION_USE_OPENCL -- -1.0
[17:55:24]     53: CAP_PROP_OPEN_TIMEOUT_MSEC -- -1.0
[17:55:24]     54: CAP_PROP_READ_TIMEOUT_MSEC -- -1.0
[17:55:24]     55: CAP_PROP_STREAM_OPEN_TIME_USEC -- -1.0
[17:55:24]     56: CAP_PROP_VIDEO_TOTAL_CHANNELS -- -1.0
[17:55:24]     57: CAP_PROP_VIDEO_STREAM -- -1.0
[17:55:24]     58: CAP_PROP_AUDIO_STREAM -- -1.0
[17:55:24]     59: CAP_PROP_AUDIO_POS -- -1.0
[17:55:24]     60: CAP_PROP_AUDIO_SHIFT_NSEC -- -1.0
[17:55:24]     61: CAP_PROP_AUDIO_DATA_DEPTH -- -1.0
[17:55:25]     62: CAP_PROP_AUDIO_SAMPLES_PER_SECOND -- -1.0
[17:55:25]     63: CAP_PROP_AUDIO_BASE_INDEX -- -1.0
[17:55:25]     64: CAP_PROP_AUDIO_TOTAL_CHANNELS -- -1.0
[17:55:25]     65: CAP_PROP_AUDIO_TOTAL_STREAMS -- -1.0
[17:55:25]     66: CAP_PROP_AUDIO_SYNCHRONIZE -- -1.0
[17:55:25]     67: CAP_PROP_LRF_HAS_KEY_FRAME -- -1.0
[17:55:25]     68: CAP_PROP_CODEC_EXTRADATA_INDEX -- -1.0
```

Camera list gives all the different capture properties and lets you set them. `camera set 4 2048` or whatever might actually work. Or whatever the different width things are.

tatarize, 2022-03-10:
https://www.learnpythonwithrune.org/find-all-possible-webcam-resolutions-with-opencv-in-python/

While you get some raw access to these things your odds aren't great unless you know the right resolution anyway. I tried and couldn't change it but I think I did mess with it with my old camera and got some other resolutions to exist.

Ah, I did get one to kick in.
```
[18:11:06] camera set 4 720
[18:11:06]     Attempt camera setting (CAP_PROP_FRAME_HEIGHT) to 720.000000. 480.000000->480.000000
[18:11:11] camera set 3 1280
[18:11:11]     Attempt camera setting (CAP_PROP_FRAME_WIDTH) to 1280.000000. 640.000000->1280.000000
```

I had to set width and height but it worked.

MiBu-dev, 2022-03-10:
` lsusb -s Id:Dev -v | egrep "Width|Height"  `
shows all supported  modes - maybe it is possible read, get list  to let user choose  res himself ? verification  if it works - user decides (same as with 2 webcams I can see 4 devices in camera control panel (and only 2 works)

I've seen another thread - with cam control done via  command line parameters  - as such would it be  possible to put cam config  in file  and GUI meerk40t would read that file ?  

in 640x480  mode  perspective corrections  make image so blurry that it is totally useless - I just use  standard preview for monitoring progress - currently  just in standard Cheese app  on another display

inspectionsbybob, 2022-03-10:
My desire is for brightness/contrast/focus controls. Don't need the rest, I
just want to be able to get better images that aren't blown out by a bright
light or so dark I can't see.

Manual focus would also be good as it tries to focus on the back wall, or
bottom and the target is out of focus...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

MiBu-dev, 2022-03-14:
Wide lens is MF only - so you set it once rotating lens - so even macro shot is possible ;)
lightburn camera is mounted on the edge of cover  - geometry  of such placement is a main cause of blurry image  after applying perspective correction - that's why lighburn camera is 5mpix - it is almost twice  more  res on  axis 

I think that i will try  to find by trial  good spot for camera - somewhere on acrylic  window  - it has to be in place where perspective is not skewed too much , and image not constantly obstructed by... gantry

closed 2024-10-04 by jpirnay