meerk40t/meerk40t #287: Rendering Issues

inspectionsbybob, 2021-01-04:
The image Bellow is a screengrab showing 
Inkscape K40Whisper MeerK40T side by side

Obviously, Inkscape is our design and what we WANT and expect to see on screen

K40 is Spot on, even got the Gradient of the stars

Meerk40T Has the text in the wrong spots, at least on Screen
The Gradient is missing (Known issue)

[image]

inspectionsbybob, 2021-01-04:
Here is the SVG for the above Image.
[Boxes for test.zip](https://github.com/meerk40t/meerk40t/files/5765866/Boxes.for.test.zip)

tatarize, 2021-01-05:
YEah, the text is a bit screwy like that. My text routine is weaker than it should be. Inkscape is going to do fine there since it runs literally everything back through inkscape and doesn't do much parsing there at all. 

Do text-to-path on the file and save that out to load. Yeah, the font doesn't preserve very well. I might need to do a bunch of work to get that in. That or run the text through Inkscape to properly figure out the glyphs.

tatarize, 2021-01-05:
#183 #61 #247

I kind of want some future ability to flow text and edit the text within MeerK40t, and that requires having them as text object.

closed 2022-07-20 by tatarize